RobertCNelson/omap-image-builder #47: temporary directory: target is busy

crunchex, 2015-01-08:
Sometimes if a rootfs build doesn't go well, a subseqeuent run will end with something like the following line:

umount: /home/crunchex/work/omap-image-builder4/ignore/tmp.3e5NsX4RjV/proc: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)

I suspect that this happens because a previous build might have gotten interrupted before the script unmounted the subdirectories while in chroot - but I could be wrong. Anyway, this leaves me with a tmp directory with some messed up permissions that I can't remove, even as root (via sudo). Any tips on how to resolve this issue?

RobertCNelson, 2015-01-08:
Yeah, if the run fails you'll have to unmount these 3 folders in the tmp directory:

```
sudo umount -f ${tempdir}/dev/pts
sudo umount -f ${tempdir}/proc
sudo umount -f ${tempdir}/sys
```

crunchex, 2015-01-10:
The umount does work in most cases where the run fails.

Although I've run into a particular case where sudo umount -f doesn't work on ${tempdir}/proc and it still says "target is busy". This case seems to happen when I try to use repo_external, and specifically when I add packages to repo_external_pkg_list. This will cause the run to fail and I won't be able to unmount ${tempdir}/proc.

RobertCNelson, 2015-01-10:
Sounds like it's failing from the external repo. First do not enable the debian.beagleboard.org repo, the key expired, thus we can't update it at the moment.

Next check the run log.

crunchex, 2015-01-11:
Sorry, where is the run log?

I'm not using the debian.beagleboard.org repo, but one I added myself (along with its key): packages.namniart.com/repos/ros

The repository details can be found here: http://wiki.ros.org/indigo/Installation/UbuntuARM

But anyway, I don't think it's an issue with the repo as I can manually install the same packages once I boot up the board using the rootfs generated with the repo enabled (just with the package list empty).

RobertCNelson, 2015-01-12:
Hi Mike,
to check this i setup a quick 'ros-ubuntu' example and pushed it to the repo. Right now it's working in both qemu (via debian jessie) and native (i.mx6q)... Feel free to take ownership of that target.

```
./RootStock-NG.sh -c ros_ubuntu
```

https://github.com/RobertCNelson/omap-image-builder/commit/b47faea5c0f7c59d049f932242f91a592fe301b4

https://github.com/RobertCNelson/omap-image-builder/commit/1bb0fe5dc38eff6d5c1718fdf9d9f5e39261177c

Regards,

closed 2017-07-18 by RobertCNelson