RobertCNelson/omap-image-builder #54: emmc-flasher does not set custom dtb in uEnv on eMMC?

robert-budde, 2015-03-28:
Hi!

I added an own config to the build system (https://github.com/robert-budde/omap-image-builder/blob/master/configs/ibb_gateway_debian_testing_armhf.conf) which sets chroot_uenv_txt="ibb_uEnv.txt". The file states "dtb=am335x-boneblack-ibb.dtb". Writing non-flasher sd-cards using setup_sdcard.sh works like a charm and makes the resulting card also use the referenced dtb, though using a flasher-image to flash the eMMC seems it does not migrate the dtb-setting?

RobertCNelson, 2015-03-30:
That's very odd, as the eMMC flasher just modifies /boot/uEnv.txt it doesn't generate it from scratch:

https://github.com/RobertCNelson/boot-scripts/blob/master/tools/eMMC/init-eMMC-flasher-v3.sh#L360

So you custom dtb= define should stay their.

Regards,

robert-budde, 2015-04-09:
I could not find the reason for this but did a workaround by creating my own machine configs which include the specific dtb. Maybe this is also the more correct way of doing it.

closed 2015-04-09 by robert-budde