RobertCNelson/omap-image-builder #56: temporary directory: random tempdir, need to re-download again if download fail.

zymxjtu, 2015-06-18:
Hi, one thing I encountered is that in the "RootStock-NG.sh", it will make temp directory which is random generated, for example, "tempdir=$(mktemp -d -p ${DIR}/ignore)" http://pastebin.ubuntu.com/11734252/

The problem is my network sometimes is unstable, and when it failed to download (and it always will :( ), everything that have been downloaded need to re-download again (Since another new different random folder will be generated). So I fixed the temp directory location instead of letting it random generate each time (I replace all 3 occurrence in the script to be something like "tempdir=$(mkdir -d -p ${DIR}/ignore/tempdir)").

Are there any design consideration for random generating the temp folder to store data each time? Any issue if fix the temp directory as what I did to overcome network issue?

Thanks. :)

RobertCNelson, 2015-06-18:
It's mostly to ensure a clean working directory, also useful for a batch build where one of the builds fail.

To save on network bandwidth and make the script more reliable with a questionable network, install "apt-cacher-ng" on either your development host or a local server in your network, then just pass:
```
export apt_proxy=serverip:3142/
```
before 
```
/RootStock-NG.sh -c <xyz>
```
Regards,

zymxjtu, 2015-06-18:
Hi Robert,

Thanks. :+1: I believe I have the confidence that I can successfully build debian image for BBB using BBB now, will test the generated image tomorrow if everything goes well :)

Thanks for the suggestions, will try "apt-cacher-ng".
One error I encountered during my attempt to build debian image is "Hash Sum mismatch" error (when I tried to build debian image using my Ubuntu 14.04 X64 laptop), after some searching, I found this article:
http://www.zyxware.com/articles/2008/07/27/how-to-fix-hash-sum-mismatch-and-size-mismatch-errors-from-apt-get-apt-cacher
talking about hash sum mismatch situation using apt-cacher.
In the end I get rid of the error by disabling one package source (but finally the attempt to cross build debian image failed due to qemu problem... =.= ). After cross build fail, following your comments in the log message, I tried to build the image using my BBB board (running BBB debian image. I have chicken, so my chicken is producing eggs now :D). The hash sum mismatch didn't occur when build using BBB. After solving some other problems I believe I can see the sunlight now. :)  

I have encountered several problems when I tried to build the image. Maybe I will summarize and share it in case some other people trying the same thing will have a reference.

Thanks~

Regards,

RobertCNelson, 2015-06-18:
The Hash Sum mismatch seems to happen when apt-cacher-ng hits teh ubuntu/debian mirror network.  I just disable that, as i'm always on testing/experimental so the mirrors see lots of updates. (for stable releases the mirrors are usually in sync..)

```
#Remap-debrep: file:deb_mirror*.gz /debian ; file:backends_debian # Debian Archives
#Remap-uburep: file:ubuntu_mirrors /ubuntu ; file:backends_ubuntu # Ubuntu Archives
#Remap-debvol: file:debvol_mirror*.gz /debian-volatile ; file:backends_debvol # Debian Volatile Archives
```

PS, while building it on x86 thru qemu 'sometimes' work, lots of random things can happen. ;)
Regards,

zymxjtu, 2015-06-18:
I see. :D

I'm always out of luck for the qemu error thing... 0.0
I believe I at least tried 5 to 6 times and none of them pass... 0.0
So in the end it was like OK, seems I have no luck here, time to move on. And I started to use BBB to build (painful thing is it is slow to do this kind of task and I'm building it inside Micro SD card which again is slow).

Good thing is I believe I can get my fresh and tasty debian image tomorrow and anything could be better than that? :D

zymxjtu, 2015-06-18:
May I ask mainly which hardware board you are using to build the image?
It is tough for BBB to do this kind of task. :)

RobertCNelson, 2015-06-18:
For official beagleboard.org images i use an omap5 with sata drive running Debian Stretch. http://www.svtronics.com/5432
The Wandboard-quad + sata drive is a cheaper solution and is my backup for when my omap5's are busy..
Regards,

zymxjtu, 2015-06-19:
Thanks Robert, I have successfully build the image for BBB. Settle some issues today and finally. Tested the generated image just now and it is working well :D.  The building process takes hours on BBB 0.0 :D.

Regards,
[image]
[image]

closed 2015-06-19 by zymxjtu

zymxjtu, 2015-07-25:
Forgot to update here.

Not sure if it is appropriate, but I have created one Wiki page last month after I successfully build the image:
http://elinux.org/BeagleBoneBlack_Rebuild_Debian_Image_Using_image-builder

Just in case if some other people try the same thing and they will have a reference.