RobertCNelson/omap-image-builder #58: Failed to fetch cdrom://[Debian GNU/Linux 8 _Jessie_ ?

g40, 2015-07-05:
Hi Robert, 

Using a freshly cloned copy of this repo and attempting to build a image for Debian Jessie

```
me@debi7:~/bonebuild/omap-image-builder$ uname -a
Linux debi7 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt11-1 (2015-05-24) x86_64 GNU/Linux
```

```
me@debi7:~/bonebuild/omap-image-builder$ ./RootStock-NG.sh -c eewiki_minfs_debian_jessie_armhf
...
Fetched 18.1 MB in 1min 27s (207 kB/s)                                                                                        
W: Failed to fetch cdrom://[Debian GNU/Linux 8 _Jessie_ - Official Snapshot amd64 LIVE/INSTALL Binary 20150606-14:41]/dists/jes
sie/main/binary-i386/Packages  Please use apt-cdrom to make this CD-ROM recognized by APT. apt-get update cannot be used to add
 new CD-ROMs

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Is this a local configuration problem or something else?

TAIA.

RobertCNelson, 2015-07-05:
Get rid of your bad apt line in /etc/apt/sources.list

g40, 2015-07-06:
Done. Thanks. Appears to have been left behind by Debian Live installer.

closed 2015-07-06 by g40