RobertCNelson/omap-image-builder #60: Beagleboard Rev C + 14.04 image

SplashX11, 2015-08-17:
I tried your ubuntu-14.04.2-console-armhf-2015-07-08.tar.xz image with my beagleboard rev C2. I found two issues. The MLO downloaded with setup_sdcard.sh does not work. After using an older version (v2015.04-r1 or v2015.01-r6) the board boots. Furthermore, the serial port in omap3-beagle.conf is set to ttyS2 but it has to be ttyO2. Otherwise a login from the serial port is not possible.

Nice to see up to date software for such an old board. Good work!

RobertCNelson, 2015-08-17:
So the goal to eventually get to ttyS2, we had a little misstep with v4.0.x where ttyS2 was causing issues so we have to revert to ttyO2, i think in my v4.1.x tree it's still ttyO2.. This should be ttyO2, sorry i didn't catch this, as debian/systemd fixes this up for us automaticly..
Regards,

RobertCNelson, 2015-08-17:
fixed in: https://github.com/RobertCNelson/omap-image-builder/commit/ac93d1f254ce1976cdbc4066c74de1ca235e7ce2

closed 2017-07-18 by RobertCNelson