RobertCNelson/omap-image-builder #61: ref@1.1.0 isn't broken, it's the node version being distributed.

fivdi, 2015-08-29:
This commit [ref@1.1.0 is broken, make sure to install ref@1.0.2 before bonescript](https://github.com/RobertCNelson/omap-image-builder/commit/3521136146aaf27985ba0dc102f6e9249a586c64) appears to assume that ref@1.1.0 is broken and works around the issue for BoneScript. I guess that there was a compile error similar to the following installing ref@1.1.0
```
../node_modules/nan/nan.h:316:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
cc1plus: warning: unrecognized command line option "-Wno-unused-local-typedefs" [enabled by default]
```
If the is the case, then it's not ref@1.1.0 that's broken, it's the version of node v0.10.29 that's is being distributed with Operating Systems like Debian. The Node.js v0.10.29 version being distributed doesn't carry the REPLACE_INVALID_UTF8 patch.

I had a similar issue installing the epoll module. See [here](https://github.com/nodejs/nan/issues/394#issuecomment-135793288) for additional information.

RobertCNelson, 2015-08-29:
@fivdi 

Testing it locally, ref@1.1.0 was failing too, so i down version-ed to ref@1.0.2..  i had thought npm would have used the local version of ref, but i guess "-g" bypasses that version..

but we have:
```
debug: node: [v0.10.38]
debug: npm: [1.4.21]
```
node was backported from stretch. ;)

I've also pinged @jadonk 

As ref is a dependicy of iff:
https://github.com/jadonk/bonescript/blob/master/package.json#L33

i'm not sure how we define sub-dependices..

Regards,

RobertCNelson, 2015-08-29:
@fivdi 
but yet the same error as what you saw...

Wonder if it's one of the v8 lib's node is built-against?

```
voodoo@a3-imx6q-wandboard-2gb:~$ node --version
v0.10.38
voodoo@a3-imx6q-wandboard-2gb:~$ npm install epoll
-
> epoll@0.1.15 install /home/voodoo/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/voodoo/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../src/epoll.cc:15:0:
../node_modules/nan/nan.h:316:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
epoll.target.mk:84: recipe for target 'Release/obj.target/epoll/src/epoll.o' failed
make: *** [Release/obj.target/epoll/src/epoll.o] Error 1
make: Leaving directory '/home/voodoo/node_modules/epoll/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Linux 4.1.6-armv7-x2
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/voodoo/node_modules/epoll
gyp ERR! node -v v0.10.38
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! epoll@0.1.15 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the epoll@0.1.15 install script.
npm ERR! This is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.1.6-armv7-x2
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "epoll"
npm ERR! cwd /home/voodoo
npm ERR! node -v v0.10.38
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/voodoo/npm-debug.log
npm ERR! not ok code 0
```

RobertCNelson, 2015-08-29:
Interesting...

```
voodoo@a3-imx6q-wandboard-2gb:~/node_modules/npm/bin$ node npm-cli.js --version
2.14.1
voodoo@a3-imx6q-wandboard-2gb:~/node_modules/npm/bin$ node npm-cli.js install epoll
-
> epoll@0.1.15 install /home/voodoo/node_modules/npm/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/voodoo/node_modules/npm/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
  SOLINK_MODULE(target) Release/obj.target/epoll.node
  COPY Release/epoll.node
make: Leaving directory '/home/voodoo/node_modules/npm/node_modules/epoll/build'
epoll@0.1.15 ../node_modules/epoll
├── bindings@1.2.1
└── nan@2.0.8
```

I'm going to  try:
```
npm install -g npm
```

fivdi, 2015-08-29:
> I've also pinged @jadonk
>
> As ref is a dependicy of iff:
> https://github.com/jadonk/bonescript/blob/master/package.json#L33
>
> i'm not sure how we define sub-dependices..

The sub-dependices of ffi are defined by ffi itself. BoneScript [depends on ffi@1.2.6](https://github.com/jadonk/bonescript/blob/master/package.json#L33) and ffi@1.2.6 [depends on the latest version of ref](https://github.com/node-ffi/node-ffi/blob/v1.2.6/package.json#L26). The latest version of ref, which is currently ref@1.1.0, [depends on the latest version of nan@2](https://github.com/TooTallNate/ref/blob/1.1.0/package.json#L35). The latest version of nan, which is currently nan@2.0.8, depends on [v8::String::REPLACE_INVALID_UTF8](https://github.com/nodejs/nan/blob/v2.0.8/nan.h#L316).

The problem is that `v8::String::REPLACE_INVALID_UTF8` doesn't exist in the `v8.h` that's being distributed with Debian 2015-03-01. It probably doesn't exist in any of the Debian version that are being made for the BeagleBone Black, even the versions with Node.js v0.10.38 backported from stretch.

> Wonder if it's one of the v8 lib's node is built-against?

The Node.js developers patch V8. For example, they patched the version of V8 that they distribute with Node.js v0.10.29 [here](https://github.com/joyent/node/commit/881ac26f27f4ac9585d66c8d8a67d5b246a23d1b). This is the patch that added `v8::String::REPLACE_INVALID_UTF8` to `v8.h`. The Debian 2015-03-01 distribution doesn't contain this patch for V8.

Debian 2015-03-01 comes with Node.js v0.10.29 and the corresponding `v8.h` contains the following enum:
```cpp
  enum WriteOptions {
    NO_OPTIONS = 0,
    HINT_MANY_WRITES_EXPECTED = 1,
    NO_NULL_TERMINATION = 2,
    PRESERVE_ASCII_NULL = 4
  };
```
However, `v8.h` in the Node.js v0.10.29 version distributed by the Node.js developers looks like [this](https://github.com/joyent/node/blob/v0.10.29/deps/v8/include/v8.h#L1075-L1084) because it's patched in V8:
```
  enum WriteOptions {
    NO_OPTIONS = 0,
    HINT_MANY_WRITES_EXPECTED = 1,
    NO_NULL_TERMINATION = 2,
    PRESERVE_ONE_BYTE_NULL = 4,
    // Used by WriteUtf8 to replace orphan surrogate code units with the
    // unicode replacement character. Needs to be set to guarantee valid UTF-8
    // output.
    REPLACE_INVALID_UTF8 = 8
  };
```
I'd imagine that the problem still exists with the versions Node.js and V8 that have be backported from stretch.

fivdi, 2015-08-29:
This issue is probably inappropriately titled. The title "ref@1.1.0 isn't broken, it's the **V8** version being distributed" would be better than "ref@1.1.0 isn't broken, it's the **node** version being distributed."

fivdi, 2015-08-30:
> Interesting...
>
> ```
> voodoo@a3-imx6q-wandboard-2gb:~/node_modules/npm/bin$ node npm-cli.js --version
> 2.14.1
> voodoo@a3-imx6q-wandboard-2gb:~/node_modules/npm/bin$ node npm-cli.js install epoll
> - [...]

I just tried `npm install -g npm` and then both `npm install bonescript` and `npm install epoll` and everything installed without errors. The epoll tests also ran successfully. However, I don't think this is actually a sane thing to do.

Before `npm install -g npm`, the `v8.h` header file used to compile native addons is `/usr/include/vh.h`. This doesn't contain the REPLACE_INVALID_UTF8 patch and there are compile errors when installing native addons.

After `npm install -g npm`, the `v8.h` header file used to compile native addons is `/root/.node-gyp/0.10.29/deps/v8/include/v8.h`. This does contain the REPLACE_INVALID_UTF8 patch and native addons can be installed successfully. However, the `v8.h` file being used here doesn't correspond to the installed binaries so there are inconsistencies. This isn't a good idea.

RobertCNelson, 2015-08-31:
Found it:
https://sources.debian.net/src/libv8-3.14/3.14.5.8-10/debian/patches/nodejsREPLACE_INVALID_UTF8.patch/

Currently rebuilding libv8-3.14 for wheezy/jessie..
Regards,

RobertCNelson, 2015-08-31:
3.14.5.8-10 is now migrated to wheezy/jessie, and we don't need the npm install -g npm hack...

npm install bonescript@2.5.0

fivdi, 2015-09-01:
Tried it out with [bone-debian-8.1-lxqt-4gb-armhf-2015-08-31-4gb.img.xz](https://rcn-ee.com/rootfs/bb.org/testing/2015-08-31/lxqt-4gb/bone-debian-8.1-lxqt-4gb-armhf-2015-08-31-4gb.img.xz) and was able to install and test `epoll` successfully. Not sure why `./root/.node-gyp` is there, so I threw it away first. I guess this issue can be close so. The [recommend Debian image](http://beagleboard.org/latest-images) for the BBB (Debian 2015-03-01) is now busted, but that's a different issue.

closed 2015-09-01 by fivdi

RobertCNelson, 2015-09-01:
```
sudo apt-get update ; sudo apt-get upgrade
```
Should be enough to fix the 2015-03-01 release would work again. (it also needs that to fix cloud9, as the version in that release is busted in chrome. )

fivdi, 2015-09-01:
Thanks, I'll give that a try.

RobertCNelson, 2015-09-01:
./root/.node-gyp is there from the npm packages needed for cloud9... (underneath cloud9 is a mess)

Here's what we do:
https://github.com/rcn-ee/repos/blob/master/c9-core-installer/suite/jessie/debian/c9-core-installer.postinst#L136

Based on their "install" script:
https://github.com/c9/install

Regards,

fivdi, 2015-09-01:
>```
>sudo apt-get update ; sudo apt-get upgrade
>```
>Should be enough to fix the 2015-03-01 release. (it also needs that to fix cloud9, as the version in that release is busted in chrome. )

Yes, it is enough to fix the 2015-03-01 release :+1: