RobertCNelson/omap-image-builder #66: Inconsistent naming of `Adafruit_BBIO` module

Mausy5043, 2015-12-08:
See this issue here: https://github.com/adafruit/adafruit-beaglebone-io-python/issues/86

Could the image builder be affected too?
Searching in this repository returns results for both flavors of the module name.
But I only find commented code.  It's late. Maybe I'm missing something. :sleeping:

RobertCNelson, 2015-12-08:
Well we install via:
```
pip install Adafruit_BBIO
```

```
Downloading/unpacking Adafruit-BBIO
  Downloading Adafruit_BBIO-0.0.30.tar.gz
  Running setup.py egg_info for package Adafruit-BBIO
    
Installing collected packages: Adafruit-BBIO
  Running setup.py install for Adafruit-BBIO
    building 'Adafruit_BBIO.GPIO' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/py_gpio.c -o build/temp.linux-armv7l-2.7/source/py_gpio.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/event_gpio.c -o build/temp.linux-armv7l-2.7/source/event_gpio.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/constants.c -o build/temp.linux-armv7l-2.7/source/constants.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/common.c -o build/temp.linux-armv7l-2.7/source/common.o -Wno-format-security
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-2.7/source/py_gpio.o build/temp.linux-armv7l-2.7/source/event_gpio.o build/temp.linux-armv7l-2.7/source/constants.o build/temp.linux-armv7l-2.7/source/common.o -o build/lib.linux-armv7l-2.7/Adafruit_BBIO/GPIO.so
    building 'Adafruit_BBIO.PWM' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/py_pwm.c -o build/temp.linux-armv7l-2.7/source/py_pwm.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/c_pwm.c -o build/temp.linux-armv7l-2.7/source/c_pwm.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/constants.c -o build/temp.linux-armv7l-2.7/source/constants.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/common.c -o build/temp.linux-armv7l-2.7/source/common.o -Wno-format-security
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-2.7/source/py_pwm.o build/temp.linux-armv7l-2.7/source/c_pwm.o build/temp.linux-armv7l-2.7/source/constants.o build/temp.linux-armv7l-2.7/source/common.o -o build/lib.linux-armv7l-2.7/Adafruit_BBIO/PWM.so
    building 'Adafruit_BBIO.ADC' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/py_adc.c -o build/temp.linux-armv7l-2.7/source/py_adc.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/c_adc.c -o build/temp.linux-armv7l-2.7/source/c_adc.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/constants.c -o build/temp.linux-armv7l-2.7/source/constants.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/common.c -o build/temp.linux-armv7l-2.7/source/common.o -Wno-format-security
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-2.7/source/py_adc.o build/temp.linux-armv7l-2.7/source/c_adc.o build/temp.linux-armv7l-2.7/source/constants.o build/temp.linux-armv7l-2.7/source/common.o -o build/lib.linux-armv7l-2.7/Adafruit_BBIO/ADC.so
    building 'Adafruit_BBIO.SPI' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/spimodule.c -o build/temp.linux-armv7l-2.7/source/spimodule.o -Wno-format-security
    source/spimodule.c:39:0: warning: "PyLong_Check" redefined [enabled by default]
    /usr/include/python2.7/longobject.h:14:0: note: this is the location of the previous definition
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/constants.c -o build/temp.linux-armv7l-2.7/source/constants.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/common.c -o build/temp.linux-armv7l-2.7/source/common.o -Wno-format-security
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-2.7/source/spimodule.o build/temp.linux-armv7l-2.7/source/constants.o build/temp.linux-armv7l-2.7/source/common.o -o build/lib.linux-armv7l-2.7/Adafruit_BBIO/SPI.so
    building 'Adafruit_BBIO.UART' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/py_uart.c -o build/temp.linux-armv7l-2.7/source/py_uart.o -Wno-format-security
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/c_uart.c -o build/temp.linux-armv7l-2.7/source/c_uart.o -Wno-format-security
[... cut ...]
On jessie:

```
debian@beaglebone:~$ python
Python 2.7.9 (default, Mar  1 2015, 13:48:22) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO
>>> 
```

```
debian@beaglebone:~$ cat /etc/dogtag 
BeagleBoard.org Debian Image 2015-12-07
debian@beaglebone:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 8.2 (jessie)
Release:	8.2
Codename:	jessie
```

It seems to be working out of the box...

RobertCNelson, 2015-12-08:
python3 isn't happy, thatll be an issue in stretch:

```
debian@beaglebone:~$ python3
Python 3.4.2 (default, Oct  8 2014, 14:38:51) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'Adafruit_BBIO'
>>> quit()
```

RobertCNelson, 2015-12-08:
and your example works in jessie to::
```
debian@beaglebone:~$ python
Python 2.7.9 (default, Mar  1 2015, 13:48:22) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO.GPIO as GPIO
>>> quit()
```

Mausy5043, 2015-12-09:
I'm on:
```
debian@bbone:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 7.9 (wheezy)
Release:	7.9
Codename:	wheezy
debian@bbone:~ $ sudo cat /etc/dogtag
BeagleBoard.org Debian Image 2015-11-12
debian@bbone:~ $ uname -a
Linux bbone 3.8.13-bone79 #1 SMP Tue Oct 13 20:44:55 UTC 2015 armv7l GNU/Linux
```

Just to be sure: I'm using the wheezy distribution I got from here: http://beagleboard.org/latest-images .
I tried the Debian 8.2 image there but that seemed broken at the time. Are those the distros you're referring to?

RobertCNelson, 2015-12-09:
http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Image_Testing_Snapshots

Mausy5043, 2015-12-09:
Testing Snapshots... :fearful:

RobertCNelson, 2015-12-09:
Yeap... "Testing Snapshots" means i'm actively testing them and pushing out weekly builds...

This should give a hint on how much "wheezy" actually gets tested by me.. (hint = none)..

Regards,

Mausy5043, 2015-12-09:
Are you saying "jessie" is a better choice at the moment?

RobertCNelson, 2015-12-09:
Right now, only two things don't work as well in jessie's 4.1.x then wheezy's 3.8..

audio: bone-audio-cape-2
rs485: still works best in 3.8..

Otherwise, jessie is a 99% fill in for the old 3.8/wheezy image..

Regards,

RobertCNelson, 2015-12-09:
and one of the only reasons 'wheezy' is still defaut... there's many books written about it..

when the x15 actually launches, the jessie 4.1.x image will be default... (same kernel for both boards)

Mausy5043, 2015-12-09:
OK. Thanks for the info. I'm expecting a new SD-card later this week. I'll try out `jessie` on the BBB when I get the new card. I already have it running everywhere else (RPi's and NAS), so that won't be a problem.

To make sure I understand the status of the various images lying around:
* http://beagleboard.org/latest-images has the latest official release.
* http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Image_Testing_Snapshots has weekly builds, where the Jessie-build marked as "*microSD/Standalone: (console) (BeagleBone/BeagleBone Black/BeagleBone Green)*" is considered reasonably stable and seems most interesting (i.e. headless purposes) for me.

And finally regarding the OT: This needs to be fixed by Adafruit, but seems to be posing no problems on Jessie.

Mausy5043, 2015-12-12:
@RobertCNelson Sorry, but still not clear on this. I just got the snapshot https://rcn-ee.com/rootfs/bb.org/testing/2015-12-07/console/bone-debian-8.2-console-armhf-2015-12-07-2gb.img.xz and un-`xz`-ed it onto an SD-card.
Booting from the SD-card I get an empty system.

RobertCNelson, 2015-12-12:
Well that's the console image, if you want a full desktop use the image marked by lxqt...

closed 2015-12-12 by Mausy5043