RobertCNelson/omap-image-builder #68: Removing xscreensaver from autostart causes incorrect permissions

hlapp, 2015-12-28:
It looks like most image generation script (specifically, all except machinekit-wheezy) remove xscreensaver from being auto-started by writing a filtered version of `/etc/xdg/lxsession/LXDE/autostart` to a temporary file which is then copied back. Unfortunately, this can, and at least for the latest Nov 12 debian wheezy image for BBB has resulted in incorrect permissions for this file, namely readable only by root. As a result, lxpanel won't start up by default for a user, and so the X display lacks the task bar, desktops, and everything else that comes with lxpanel.

(This has confounded me for quite a while - there are no error messages from the lightdm startup about this logged anywhere that would point to the culprit.)

RobertCNelson, 2015-12-28:
thanks @hlapp i had never noticed.. i copied the machinekit version to all the other versions.

closed 2015-12-28 by RobertCNelson

hlapp, 2015-12-28:
Thanks @RobertCNelson for the prompt fix - was just getting started on a PR with exact same fix :smile: