RobertCNelson/omap-image-builder #69: Getting g_serial in console image

adilinden, 2016-02-15:
Hi,

I have been trying the prebuilt images on a BBB.  I am most interested in a very basic image, which I think the *-concole-* images will provide.  However, I cannot get the USB serial gadget to work (I do not need the USB ethernet gadget).  Looking at the /opt/scripts/boot/am335x_evm.sh script it sees the the console build only has a single ext4 partition and does not use g_multi but uses g_ether instead.  If I connect to the device over the network I cannot load g_serial, it throws an error of 'no such device' (IIRC).

Questions:
- Can g_serial not be loaded when g_ether is loaded?
- Can one create an image that would forego g_ether and use g_serial instead?
- Can one create a *-console-* image with FAT partition, which would allow g_multi and therefore all USB gadget support?

Thanks,
Adi

RobertCNelson, 2016-02-15:
Just remove udhcpd:

```
sudo apt-get remove udhcpd --purge
```
It'll default to g_serial on bootup:

RobertCNelson, 2016-02-15:
btw, you can also cheat and just do:

```
sudo touch /var/local/usb_mass_storage.img
```

Regards,

adilinden, 2016-02-16:
I tried

>` sudo touch /var/local/usb_mass_storage.img`

But on next boot I get:

> `[   78.354450] g_multi musb-hdrc.0.auto: failed to start g_multi: -525`

It loads neither g_multi nor any other USB gadget kernel modules.

RobertCNelson, 2016-02-16:
darn i guess g_multi is smarter then that. .;)

```
sudo rm /var/local/usb_mass_storage.img
sudo apt-get update
sudo apt-get install doc-beaglebone-getting-started
sudo reboot
```

adilinden, 2016-02-16:
Awesome, that did it! So g_multi is looking for a valid FAT image then?

closed 2017-07-18 by RobertCNelson