RobertCNelson/omap-image-builder #70: Failure at bb-npm-installer.postinst step

spiderkeys, 2016-02-24:
Trying to build an image that installs the experimental node v4.x. Our build was successful a couple weeks ago, but something about the bb-npm-installer involving an initial install of node v0.10 seems to be breaking the build now. We installed node v0.12, as shown in the other config files and there doesn't seem to be a package for nodejs-v0.10.x or nodejs-v0.10.x-legacy. Tried to match the other Jessie config files as best as possible and made sure we were working from the omap-image-builder's HEAD.

Build log:
[omap-build-log.txt](https://github.com/RobertCNelson/omap-image-builder/files/145297/omap-build-log.txt)

Config file being used:

https://github.com/spiderkeys/omap-image-builder/blob/openrov-dev-image/configs/OpenROV-debian-jessie-development.conf

Any insight is appreciated. Thanks!

RobertCNelson, 2016-02-24:
Sorry @spiderkeys, for about the nodejs v4.x option..

https://github.com/rcn-ee/repos/commit/91a9214d2ceabd02e5d0f223ba166bcb90190aaa

(i need to push a 4.x update to include a /usr/bin/nodejs-v4.x option..)

Regards,

spiderkeys, 2016-02-24:
No problem; definitely appreciate your work and responsiveness (and willingness to support our Node shenanigans). Will keep my eye out for the update.

RobertCNelson, 2016-02-26:
Hey @spiderkeys , i'm thinking about just adding this repo:

https://github.com/nodesource/distributions

```
wget -qO- https://deb.nodesource.com/gpgkey/nodesource.gpg.key | apt-key add -
```

```
apt-get update
apt-get install apt-transport-https

echo 'deb https://deb.nodesource.com/node_4.x jessie main' > /etc/apt/sources.list.d/nodesource.list
echo 'deb-src https://deb.nodesource.com/node_4.x jessie main' >> /etc/apt/sources.list.d/nodesource.list

apt-get update
apt-get install nodejs
```

Have you tried their builds before?

spiderkeys, 2016-02-26:
@RobertCNelson no, I don't believe we've installed any of their builds before, but reading through the open and closed issues, and their support version/platform list, I don't see any indicators that it wouldn't work for us. If that is the easiest way to proceed, I'd say go for it and we can try it out. It also seems like it would give us regular access to future builds with less work on your part, so that's a win-win.

spiderkeys, 2016-03-02:
Marking this as closed. Updated to your recent commits and we now have a working node 4.3.1 via the nodesource packages. Thanks Robert!

closed 2016-03-02 by spiderkeys