RobertCNelson/omap-image-builder #71: Element14 Rev C ID

spiderkeys, 2016-03-16:
There may be a typo or error with the last two bytes in the Rev C identifier under your EEPROM database section in the readme:

```
C: [aa 55 33 ee 41 33 33 35  42 4e 4c 54 30 30 30 43 |.U3.A335BNLT000C|]
```

On our Element14 Rev C we read (not byteswapped):

```
[ 55aa ee33 3341 3533 4e42 544c 3030 3043  | .U3.A335BNLT00C0 ]
```

RobertCNelson, 2016-03-16:
So, 'some' clones ended up being quite liberal on their choices, would you please confirm via:

```
wget https://raw.githubusercontent.com/RobertCNelson/boot-scripts/master/device/bone/tester/show-eeprom.sh
sudo /bin/bash show-eeprom.sh
```

Example:
```
debian@test-bbb-2:~$ sudo /bin/bash show-eeprom.sh 
eeprom: [�U3�A335BNLT0A5B]
eeprom raw: [00000000  aa 55 33 ee 41 33 33 35  42 4e 4c 54 30 41 35 42  |.U3.A335BNLT0A5B|]
```

spiderkeys, 2016-03-16:
```
eeprom: [�U3�A335BNLT00C0]
eeprom raw: [00000000  aa 55 33 ee 41 33 33 35  42 4e 4c 54 30 30 43 30  |.U3.A335BNLT00C0|]
```

RobertCNelson, 2016-03-16:
@jadonk can you ping them, is this a newer rev?

RobertCNelson, 2016-03-16:
Thanks @spiderkeys i've added it to the list, not sure what changed... Hopefully Jason can find out. ;)

RobertCNelson, 2016-03-17:
@spiderkeys , can you run the script again. (just pushed a change to also get the board serial number) we are trying to track what happened, but the ID number is also needed

```
wget https://raw.githubusercontent.com/RobertCNelson/boot-scripts/master/device/bone/tester/show-eeprom.sh
sudo /bin/bash show-eeprom.sh
```
```
sudo /bin/bash show-eeprom.sh 
eeprom: [�U3�A335BONE00A5]
eeprom raw: [00000000  aa 55 33 ee 41 33 33 35  42 4f 4e 45 30 30 41 35  |.U3.A335BONE00A5|]
eeprom raw: [00000010  31 30 31 32 42 42 30 31  32 33 39 31 00 00 00 00  |1012BB012391....|]
```

Thanks!

spiderkeys, 2016-03-18:
```
eeprom: [�U3�A335BNLT00C0]
eeprom raw: [00000000  aa 55 33 ee 41 33 33 35  42 4e 4c 54 30 30 43 30  |.U3.A335BNLT00C0|]
eeprom raw: [00000010  33 36 31 35 42 42 42 4b  31 35 37 32 ff ff ff ff  |3615BBBK1572....|]
```

closed 2016-07-22 by spiderkeys

jadonk, 2016-07-24:
The report from E14 is they took this number "00C0" from the SRM.