RobertCNelson/omap-image-builder #72: DE0-Nano-SoC kit - suboptimal partition layout

mhaberler, 2016-03-29:
hi Robert, not sure this is the place to discuss https://www.eewiki.net/display/linuxonarm/DE0-Nano-SoC+Kit#DE0-Nano-SoCKit-SetupmicroSDcard (pls advise otherwise)

the [partition layout as per GSRD recommendations](https://rocketboards.org/foswiki/view/Documentation/GSRD151SdCard) is a tad silly an be improved:

- the vfat partition is not really necessary - the [/boot in ext4 rootfs](http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0) works just fine and gives less problems on upgrading (vfat symlink snafus during kernel upgrade etc)
- the A2 partition for the u-boot-with-spl.img should be before the root partition so the root partition can be easily extended to match the SD card

not sure how to proceed on this, are you working on this?
Michael

RobertCNelson, 2016-03-29:
Yeah, right now it's sub-optimal, i tried your setup yesterday, but i couldn't get u-boot to load /zImage or the dtb from the /boot/xyz location.. (u-boot 2016.03) what was really weird the response from ls:

```
ls mmc 0:2 /boot/
```

there seems to be a major mmc bug in the current u-boot code for this device...

mhaberler, 2016-03-29:
bizarre. Just noting I compiled u-boot natively on a cubox-i, gcc version 4.9.2 (Debian 4.9.2-10).

Can you point me to your u-boot binary so I can try?

RobertCNelson, 2016-03-29:
So it's just mainline no patches:

https://rcn-ee.net/repos/bootloader/socfpga_de0_nano_soc/u-boot-socfpga_de0_nano_soc-v2016.03-r6.sfp

```
=> version          

U-Boot 2016.03 (Mar 28 2016 - 16:53:19 -0500)
arm-linux-gnueabihf-gcc (Linaro GCC 5.3-2016.02) 5.3.1 20160113
GNU ld (GNU Binutils) 2.25.0 Linaro 2015_10
=> ls mmc 0:1 /     
  3786920   zimage 
    19446   socfpga.dtb 

2 file(s), 0 dir(s)

=> ls mmc 0:2 /     
<DIR>       4096 .
<DIR>       4096 ..
<DIR>      16384 lost+found
<DIR>       4096 boot
<DIR>       4096 opt
<DIR>       4096 mnt
<DIR>       4096 run
<DIR>       4096 dev
<DIR>       4096 usr
<DIR>          0 media
<DIR>          0 lib
<DIR>          0 etc
<DIR>       2808 bin
<DIR>          6 home
<DIR>       2434 proc
<DIR>       1922 srv
<DIR>       2743 var
<DIR>       2814 root
<DIR>       2124 tmp
<DIR>       2070 sbin
<DIR>        159 sys
=> ls mmc 0:2 /boot/
<DIR>       4096 .
<DIR>       4096 ..
<DIR>       4096 uboot
=> ls mmc 0:2 /etc/ 
=> 
```

/etc/ should be full....

mhaberler, 2016-03-29:
ok, rebuilt mainline v2016.03, installed that:

`````
=> ls mmc 0:2 /etc/
<DIR>       4096 .
<DIR>       4096 ..
<DIR>       4096 modules-load.d
<DIR>       4096 rc2.d
<DIR>       4096 rc6.d
<DIR>       4096 logcheck
<DIR>       4096 rc4.d
            1822 passwd-
<DIR>       4096 pulse
<DIR>       4096 emacs
            1046 shadow-
<DIR>       4096 ppp
<DIR>       4096 modprobe.d
<DIR>       4096 dictionaries-common
             103 shells
            3054 udhcpd.conf
<DIR>       4096 chatscripts
              37 fstab
<DIR>       4096 systemd
<DIR>       4096 wicd
              34 e2fsck.conf
<DIR>       4096 hostapd
<DIR>       4096 apparmor.d
              66 wvdial.conf
              90 rcn-ee.conf
             771 staff-group-for-usr-local
             343 subuid
<DIR>       4096 dkms
            8997 locale.gen
<DIR>       4096 dbus-1
             211 hosts
<DIR>       4096 udisks2
<DIR>       4096 tmpfiles.d
             859 insserv.conf
<DIR>       4096 linuxcnc
<DIR>       4096 init.d
<DIR>       4096 profile.d
              35 dogtag
<DIR>       4096 rc5.d
            2492 locale.alias
               9 host.conf
<DIR>       4096 python3
             722 crontab
             111 magic.mime
<DIR>       4096 rc1.d
            2969 debconf.conf
<DIR>       4096 udev
               8 timezone
             604 deluser.conf
            1054 group
<DIR>       4096 bash_completion.d
<DIR>       4096 vim
<DIR>       4096 python2.7
             411 hosts.allow
               0 .pwd.lock
            2987 adduser.conf
              33 machine-id
<DIR>       4096 security
           19605 services
<DIR>       4096 initramfs-tools
            4812 wgetrc
<DIR>       4096 menu-methods
             903 gshadow
<DIR>       4096 ca-certificates
            1822 passwd
             691 sudoers
             367 bindresvport.blacklist
<DIR>       4096 avahi
<DIR>       4096 kernel
<DIR>       4096 cron.hourly
<DIR>       4096 gtk-2.0
             118 localtime
           10852 gnome-vfs-mime-magic
<DIR>       4096 X11
<DIR>       4096 resolvconf
<DIR>       4096 usb_modeswitch.d
<DIR>       4096 fonts
<DIR>       4096 terminfo
<DIR>       4096 bonobo-activation
             286 motd
<DIR>       4096 apache2
             150 issue.net
<DIR>       4096 python3.4
<DIR>       4096 dhcp
            2584 gai.conf
<DIR>       4096 gtk-3.0
<DIR>       4096 insserv.conf.d
<DIR>       4096 gss
<DIR>       4096 sgml
             887 rpc
<DIR>       4096 newt
<DIR>       4096 libnl-3
<DIR>       4096 xdg
             888 usb_modeswitch.conf
<DIR>       4096 cron.d
<DIR>       4096 rc0.d
            7820 ca-certificates.conf
<DIR>       4096 ssh
<DIR>       4096 pam.d
<DIR>       4096 xrdp
<DIR>       4096 rc3.d
               4 debian_version
           24501 fb.modes
            2932 protocols
<DIR>       4096 perl
<DIR>       4096 ufw
<DIR>       4096 network
<DIR>       4096 logrotate.d
               0 environment
            5173 manpath.config
<DIR>       4096 alternatives
<DIR>       4096 skel
              51 vdpau_wrapper.cfg
<DIR>       4096 init
              60 networks
             156 issue
<DIR>       4096 cron.daily
             323 subuid-
<DIR>       4096 opt
<DIR>       4096 selinux
<DIR>       4096 polkit-1
             191 libaudit.conf
            2632 rsyslog.conf
            4324 mailcap
             268 rmt
<DIR>       4096 gdb
             100 resolv.conf
            4038 securetty
<DIR>       4096 rsyslog.d
            1863 bash.bashrc
<DIR>       4096 insserv
<DIR>       4096 groff
            8453 nanorc
<DIR>       4096 at-spi2
             711 hosts.deny
             529 nsswitch.conf
             280 fuse.conf
             449 mailcap.order
<DIR>       4096 iproute2
            3663 screenrc
            2084 sysctl.conf
             552 pam.conf
             956 mke2fs.conf
<DIR>       4096 bluetooth
<DIR>       4096 lightdm
<DIR>       4096 sane.d
<DIR>       4096 sound
<DIR>       4096 dpkg
<DIR>       4096 ifplugd
<DIR>       4096 lighttpd
           10478 login.defs
<DIR>       4096 rcS.d
             599 logrotate.conf
<DIR>       4096 apt
<DIR>       4096 ld.so.conf.d
            3980 drirc
             142 uniconf.conf
             855 group-
           24146 mime.types
            1748 inputrc
           73175 ld.so.cache
<DIR>       4096 default
<DIR>       4096 xml
<DIR>       4096 sudoers.d
<DIR>       4096 python
<SYM>         21 os-release
             195 modules
<DIR>       4096 sysctl.d
              13 timestamp
<DIR>       4096 calendar
              34 ld.so.conf
             306 rc.local
<DIR>       4096 ldap
             111 magic
<DIR>       4096 pcmcia
             761 profile
<DIR>       4096 cron.monthly
<DIR>       4096 binfmt.d
<DIR>       4096 gconf
<DIR>       4096 gnome-vfs-2.0
             323 subgid-
<DIR>       4096 ssl
             708 gshadow-
              10 hostname
            1260 ucf.conf
<DIR>       4096 acpi
<DIR>       4096 cron.weekly
<DIR>       4096 wpa_supplicant
            1046 shadow
             343 subgid
<SYM>         12 mtab
             206 idmapd.conf
             767 netconfig
<DIR>       4096 request-key.d
=> 
````

mhaberler, 2016-03-29:
`````
=> version
 
U-Boot 2016.03-00001-ge3c66fe (Mar 29 2016 - 18:08:52 +0200)
gcc (Debian 4.9.2-10) 4.9.2
GNU ld (GNU Binutils for Debian) 2.25
=> 
````

mhaberler, 2016-03-29:
gut feeling - give gcc-4.9 a try

mhaberler, 2016-03-29:
nope, it isnt. Installed your binary on my SD:

`````
 
 
U-Boot 2016.03 (Mar 17 2016 - 16:49:42 -0500), Build: jenkins-github_Bootloader-Builder-351
 
CPU:   Altera SoCFPGA Platform
FPGA:  Altera Cyclone V, SE/A4 or SX/C4, version 0x0
BOOT:  SD/MMC Internal Transceiver (3.0V)
       Watchdog enabled
I2C:   ready
DRAM:  1 GiB
MMC:   dwmmc0@ff704000: 0
In:    serial
Out:   serial
Err:   serial
Model: Terasic DE0-Nano(Atlas)
Net:   
Error: ethernet@ff702000 address not set.
No ethernet found.
Hit any key to stop autoboot:  0 
=> version
 
U-Boot 2016.03 (Mar 17 2016 - 16:49:42 -0500)
arm-linux-gnueabihf-gcc (Linaro GCC 5.2-2015.11-2) 5.2.1 20151005
GNU ld (GNU Binutils) 2.25.0 Linaro 2015_10
=> ls mmc 0:2 /
<DIR>       4096 .
<DIR>       4096 ..
<DIR>      16384 lost+found
<DIR>       4096 boot
<DIR>       4096 sys
<DIR>       4096 usr
<DIR>       4096 srv
<DIR>       4096 proc
<DIR>       4096 sbin
<DIR>       4096 lib
<DIR>       4096 var
<DIR>       4096 root
<DIR>       4096 run
<DIR>       4096 media
<DIR>       4096 tmp
<DIR>       4096 opt
<DIR>       4096 mnt
<DIR>       4096 dev
<DIR>       4096 etc
<DIR>       4096 bin
<DIR>       4096 home
=> ls mmc 0:2 /etc
<DIR>       4096 .
<DIR>       4096 ..
<DIR>       4096 modules-load.d
<DIR>       4096 rc2.d
<DIR>       4096 rc6.d
<DIR>       4096 logcheck
<DIR>       4096 rc4.d
            1822 passwd-
<DIR>       4096 pulse
<DIR>       4096 emacs
            1046 shadow-
<DIR>       4096 ppp
<DIR>       4096 modprobe.d
<DIR>       4096 dictionaries-common
             103 shells
            3054 udhcpd.conf
<DIR>       4096 chatscripts
              37 fstab
<DIR>       4096 systemd
<DIR>       4096 wicd
[... 153 lines cut ...]
             111 magic
<DIR>       4096 pcmcia
             761 profile
<DIR>       4096 cron.monthly
<DIR>       4096 binfmt.d
<DIR>       4096 gconf
<DIR>       4096 gnome-vfs-2.0
             323 subgid-
<DIR>       4096 ssl
             708 gshadow-
              10 hostname
            1260 ucf.conf
<DIR>       4096 acpi
<DIR>       4096 cron.weekly
<DIR>       4096 wpa_supplicant
            1046 shadow
             343 subgid
<SYM>         12 mtab
             206 idmapd.conf
             767 netconfig
<DIR>       4096 request-key.d
=> 
`````

must be something else. let me think.. bbl.

mhaberler, 2016-03-29:
hm, hardware revision? switches? crap on SD card  at the beginning?

mhaberler, 2016-03-29:
btw are you _positive_ there is something in /etc ? ;)

RobertCNelson, 2016-03-29:
the only id on this board: SoC-003-04034

Jumpers:
```
1=on
2=off
3=on
4=off
5=on
6=on
```

RobertCNelson, 2016-03-29:
Yeap, /etc/ is populated. ;)

mhaberler, 2016-03-29:
look at your / listing: 

`````
=> ls mmc 0:2 /     
<DIR>       4096 .
<DIR>       4096 ..
<DIR>      16384 lost+found
<DIR>       4096 boot
<DIR>       4096 opt
<DIR>       4096 mnt
<DIR>       4096 run
<DIR>       4096 dev
<DIR>       4096 usr
<DIR>          0 media
<DIR>          0 lib
<DIR>          0 etc  <--- notice size 0!
````

mine says:

`````
<DIR>       4096 etc
````

RobertCNelson, 2016-03-29:
Oh it's there, when i take a look at the card on another system..

```
voodoo@hades:/tmp/drive$ lsblk
NAME   MAJ:MIN RM    SIZE RO TYPE MOUNTPOINT
sda      8:0    0  465.8G  0 disk 
└─sda1   8:1    0  465.8G  0 part /
sdd      8:48   1    3.7G  0 disk 
├─sdd1   8:49   1     48M  0 part 
├─sdd2   8:50   1    3.7G  0 part /tmp/drive
└─sdd3   8:51   1 1023.5K  0 part 
voodoo@hades:/tmp/drive$ ls ./etc/
Display all 156 possibilities? (y or n)
voodoo@hades:/tmp/drive$ ls ./etc/
adduser.conf            dbus-1          host.conf        ld.so.conf.d    modules-load.d  python2.7     security                   timestamp
alternatives            debconf.conf    hostname         libaudit.conf   motd            python3       selinux                    timezone
apache2                 debian_version  hosts            libnl-3         nanorc          python3.4     services                   tmpfiles.d
apm                     default         hosts.allow      locale.alias    network         rc0.d         shadow                     udev
apt                     deluser.conf    hosts.deny       locale.gen      networks        rc1.d         shadow-                    udhcpd.conf
avahi                   dhcp            ifplugd          localtime       newt            rc2.d         shells                     ufw
bash.bashrc             dpkg            init             logcheck        nsswitch.conf   rc3.d         skel                       uniconf.conf
bash_completion.d       e2fsck.conf     init.d           login.defs      opt             rc4.d         ssh                        usb_modeswitch.conf
bindresvport.blacklist  environment     initramfs-tools  logrotate.conf  os-release      rc5.d         ssl                        usb_modeswitch.d
binfmt.d                fb.modes        inputrc          logrotate.d     pam.conf        rc6.d         staff-group-for-usr-local  vim
ca-certificates         fstab           insserv          machine-id      pam.d           rc.local      subgid                     wgetrc
ca-certificates.conf    gai.conf        insserv.conf     magic           passwd          rcn-ee.conf   subgid-                    wpa_supplicant
calendar                groff           insserv.conf.d   magic.mime      passwd-         rcS.d         subuid                     wvdial.conf
chatscripts             group           iproute2         mailcap         pcmcia          resolvconf    subuid-                    X11
cron.d                  group-          issue            mailcap.order   perl            resolv.conf   sudoers                    xdg
cron.daily              gshadow         issue.net        manpath.config  ppp             rmt           sudoers.d
cron.hourly             gshadow-        kernel           mime.types      profile         rpc           sysctl.conf
cron.monthly            gss             ldap             mke2fs.conf     profile.d       rsyslog.conf  sysctl.d
crontab                 hdparm.conf     ld.so.cache      modprobe.d      protocols       rsyslog.d     systemd
cron.weekly             hostapd         ld.so.conf       modules         python          securetty     terminfo
```

the-snowwhite, 2016-03-29:
Sorry for barging in will just say something about the jumpers:

ON  = 0
OFF = 1

So this  looks like this on the circuit board:

1=on = 0
2=off = 1
3=on = 0
4=off = 1
5=on = 1
6=on = 1

Visually

mhaberler, 2016-03-29:
switches same here, dogtag: http://snag.gy/OImRQ.jpg

mhaberler, 2016-03-29:
Robert, can you upload your image? I'll poke around, want to know now ;)

the-snowwhite, 2016-03-29:
Good its just because the manuals are very confusing about this akwardness ON = 0

the-snowwhite, 2016-03-29:
I have very strong recommendations for sticking to

gcc 4.9
uboot 2016.01
altera socfpga 3.10-rt-ltst
Quartus generated .dtb  + rbf.
Based upon the DE0_NANO_SOC GHRD release from Terasics NANO 1.0 release
Debian Jessie armhf.

the-snowwhite, 2016-03-29:
This is the combination that has been manifesting successful results

mhaberler, 2016-03-29:
btw I used fdisk for partitioning the SD

mhaberler, 2016-03-29:
I rebuilt an SD manually according to your website instructions, your u-boot, using sfdisk, layout as proposed, extracted my rootfs tar - I cannot reproduce the problem

please upload your image so I can check on my boards (hey, DigiKey customer ;), then at least we can narrow it down to hardware vs image/sd card

`````
=> ls mmc 0:2     
<DIR>       4096 .
<DIR>       4096 ..
<DIR>      16384 lost+found
<DIR>       4096 boot
<DIR>       4096 sys
<DIR>       4096 usr
<DIR>       4096 srv
<DIR>       4096 proc
<DIR>       4096 sbin
<DIR>       4096 lib
<DIR>       4096 var
<DIR>       4096 root
<DIR>       4096 run
<DIR>       4096 media
<DIR>       4096 tmp
<DIR>       4096 opt
<DIR>       4096 mnt
<DIR>       4096 dev
<DIR>       4096 etc
<DIR>       4096 bin
<DIR>       4096 home
=> 

=> ls mmc 0:2 /etc
<DIR>       4096 .
<DIR>       4096 ..
<DIR>       4096 modules-load.d
<DIR>       4096 rc2.d
<DIR>       4096 rc6.d
<DIR>       4096 logcheck
<DIR>       4096 rc4.d
            1714 passwd-
<DIR>       4096 pulse
<DIR>       4096 emacs
             878 shadow-
<DIR>       4096 ppp
<DIR>       4096 modprobe.d
<DIR>       4096 dictionaries-common
             103 shells
            3054 udhcpd.conf
<DIR>       4096 chatscripts
             100 fstab
<DIR>       4096 systemd
<DIR>       4096 wicd
              34 e2fsck.conf
<DIR>       4096 hostapd
<DIR>       4096 apparmor.d
              66 wvdial.conf
              90 rcn-ee.conf
             771 staff-group-for-usr-local
             323 subuid
<DIR>       4096 dkms
            8997 locale.gen
<DIR>       4096 dbus-1
             211 hosts
<DIR>       4096 udisks2
<DIR>       4096 tmpfiles.d
             859 insserv.conf
<DIR>       4096 linuxcnc
<DIR>       4096 init.d
<DIR>       4096 profile.d
              35 dogtag
<DIR>       4096 rc5.d
            2492 locale.alias
               9 host.conf
<DIR>       4096 python3
             722 crontab
             111 magic.mime
<DIR>       4096 rc1.d
            2969 debconf.conf
<DIR>       4096 udev
               8 timezone
             604 deluser.conf
            1064 group
<DIR>       4096 bash_completion.d
<DIR>       4096 vim
<DIR>       4096 python2.7
             411 hosts.allow
               0 .pwd.lock
            2987 adduser.conf
              33 machine-id
<DIR>       4096 security
           19605 services
<DIR>       4096 initramfs-tools
            4812 wgetrc
<DIR>       4096 menu-methods
             913 gshadow
<DIR>       4096 ca-certificates
            1779 passwd
             691 sudoers
             367 bindresvport.blacklist
<DIR>       4096 avahi
<DIR>       4096 kernel
<DIR>       4096 cron.hourly
<DIR>       4096 gtk-2.0
             118 localtime
           10852 gnome-vfs-mime-magic
<DIR>       4096 X11
<DIR>       4096 resolvconf
<DIR>       4096 usb_modeswitch.d
<DIR>       4096 fonts
<DIR>       4096 terminfo
<DIR>       4096 bonobo-activation
             286 motd
<DIR>       4096 apache2
             150 issue.net
<DIR>       4096 python3.4
<DIR>       4096 dhcp
            2584 gai.conf
<DIR>       4096 gtk-3.0
<DIR>       4096 insserv.conf.d
<DIR>       4096 gss
<DIR>       4096 sgml
             887 rpc
<DIR>       4096 newt
<DIR>       4096 libnl-3
<DIR>       4096 xdg
             888 usb_modeswitch.conf
<DIR>       4096 cron.d
<DIR>       4096 rc0.d
            7820 ca-certificates.conf
<DIR>       4096 ssh
<DIR>       4096 pam.d
<DIR>       4096 xrdp
<DIR>       4096 rc3.d
               4 debian_version
           24501 fb.modes
            2932 protocols
<DIR>       4096 perl
<DIR>       4096 ufw
<DIR>       4096 network
<DIR>       4096 logrotate.d
               0 environment
            5173 manpath.config
<DIR>       4096 alternatives
<DIR>       4096 skel
              51 vdpau_wrapper.cfg
<DIR>       4096 init
              60 networks
             156 issue
<DIR>       4096 cron.daily
             299 subuid-
<DIR>       4096 opt
<DIR>       4096 selinux
<DIR>       4096 polkit-1
             191 libaudit.conf
            2632 rsyslog.conf
            4324 mailcap
             268 rmt
<DIR>       4096 gdb
             100 resolv.conf
            4038 securetty
<DIR>       4096 rsyslog.d
            1863 bash.bashrc
<DIR>       4096 insserv
<DIR>       4096 groff
            8453 nanorc
<DIR>       4096 at-spi2
             711 hosts.deny
             529 nsswitch.conf
             280 fuse.conf
             449 mailcap.order
<DIR>       4096 iproute2
            3663 screenrc
            2084 sysctl.conf
             552 pam.conf
             956 mke2fs.conf
<DIR>       4096 bluetooth
<DIR>       4096 lightdm
<DIR>       4096 sane.d
<DIR>       4096 sound
<DIR>       4096 dpkg
<DIR>       4096 ifplugd
<DIR>       4096 lighttpd
           10478 login.defs
<DIR>       4096 rcS.d
             599 logrotate.conf
<DIR>       4096 apt
<DIR>       4096 ld.so.conf.d
            3980 drirc
             142 uniconf.conf
            1054 group-
           24146 mime.types
            1748 inputrc
           72991 ld.so.cache
<DIR>       4096 default
<DIR>       4096 xml
<DIR>       4096 sudoers.d
<DIR>       4096 python
<SYM>         21 os-release
             195 modules
<DIR>       4096 sysctl.d
              13 timestamp
<DIR>       4096 calendar
              34 ld.so.conf
             306 rc.local
<DIR>       4096 ldap
             111 magic
<DIR>       4096 pcmcia
             761 profile
<DIR>       4096 cron.monthly
<DIR>       4096 binfmt.d
<DIR>       4096 gconf
<DIR>       4096 gnome-vfs-2.0
             299 subgid-
<DIR>       4096 ssl
             903 gshadow-
              10 hostname
            1260 ucf.conf
<DIR>       4096 acpi
<DIR>       4096 cron.weekly
<DIR>       4096 wpa_supplicant
            1019 shadow
             323 subgid
<SYM>         12 mtab
=> 
`````

mhaberler, 2016-03-29:
btw - if you get the image running, could you check if it works reliably with 1GB/s ethernet?

we had issues (worked fine at 100MB/s) and it turned out to be a device tree issue

the r/txc-skew-ps params here must be commented out when using a recent u-boot: https://github.com/torvalds/linux/blob/v4.4/arch/arm/boot/dts/socfpga_cyclone5_de0_sockit.dts#L68-L70

symptom is - incoming packets are fine, tcpdump shows them
it looks like packets go out (no tx errors, dmsg entries or such) but the other side cant hear us

comment out - rebuild dtb - reboot, works; would be good to know this wasnt a mirage

mhaberler, 2016-03-30:
@RobertCNelson - could you check if your ext4 root file system has the `huge_file` flag set? (mine does)

`tune2fs -l  blockdevice  |grep huge`

it seems u-boot ext4 support requires this; as a consequence the kernel should have it too (CONFIG_LBDAF=y)

you can toggle that flag with tune2fs 

makes the difference in uboot?

mhaberler, 2016-03-30:
I'm onto a different trail

I used fdisk and that leaves 2048 sectors before the first partition - I note sfdisk does not do this 

Note the uboot [socfpga default config for mmc](https://github.com/RobertCNelson/u-boot/blob/master/include/configs/socfpga_common.h#L291-L294) stores the persistent uboot environment **right after the MBR**, this obviously assumes the fdisk behavior of starting the first partition at offset 2048 (see https://rocketboards.org/foswiki/view/Documentation/GSRD151SdCard)

Maybe a environment save operation overwrites something here, like the 'invalid CRC' which likely zeroes it? one would think if so,  that would damage the DOS partition, not the rootfs 

Other than that - fighting sfdisk 2.25.2  ;-/  the [sfdisk script for <= 2.25](https://www.eewiki.net/display/linuxonarm/DE0-Nano-SoC+Kit#DE0-Nano-SoCKit-LinuxKernel) breaks for me - trying to add the A2 partition fails because sfdisk wont allocate space for unknown ID's (?)

`````
# trying manually:
root@cubox-i:/home/mah/omap-image-builder/deploy# sfdisk --unit M /dev/sdb
sfdisk: Checking that no-one is using this disk right now ...
sfdisk: OK

Disk /dev/sdb: 1017 cylinders, 123 heads, 62 sectors/track
sfdisk:  /dev/sdb: unrecognized partition table type
Old situation:
sfdisk: No partitions found
sfdisk: Input in the following format; absent fields get a default value.
<start> <size> <type [E,S,L,X,hex]> <bootable [-,*]> <c,h,s> <c,h,s>
Usually you only need to specify <start> and <size> (and perhaps <type>).

sfdisk: /dev/sdb1 :
2,48,0xE,
/dev/sdb1         0+    48-    49-     49568+   e  W95 FAT16 (LBA)
sfdisk: /dev/sdb2 :
50,,,
/dev/sdb2        48+  3786-  3739-   3828252   83  Linux
sfdisk: /dev/sdb3 :
1,1,0xA2,
sfdisk: Warning: given size (4) exceeds max allowable size (0)

#--------- ^^^^^^^^  out of my depth understanding this one

sfdisk: /dev/sdb3 :
New situation:
Units: 1MiB = 1024*1024 bytes, blocks of 1024 bytes, counting from 0

   Device Boot Start   End    MiB    #blocks   Id  System
/dev/sdb1         0+    48-    49-     49568+   e  W95 FAT16 (LBA)
/dev/sdb2        48+  3786-  3739-   3828252   83  Linux
/dev/sdb3         0      -      0          0    0  Empty        <----- not allocated
/dev/sdb4         0      -      0          0    0  Empty
sfdisk: Warning: no primary partition is marked bootable (active)
This does not matter for LILO, but the DOS MBR will not boot this disk.
Do you want to write this to disk? [ynq] y
Successfully wrote the new partition table
`````

mhaberler, 2016-03-30:
ok, after installing sfdisk 2.28 the partitioning issues went away immediately - after playing with sfdisk 2.25 (jessie) for too many hours I'd recommend against using it - too many issues

@RobertCNelson - which one did you use?

RobertCNelson, 2016-03-30:
@mhaberler I'm on Debian Stretch, so 2.27.1, there's a lot oddities with sfdisk <= 2.25, once ubuntu 16.04 get's released, i'm going to nuke all 2.25 references...

* we had moved to sfdisk over fdisk a few years back, as fdisk was having other issues..

RobertCNelson, 2016-04-01:
@mhaberler @the-snowwhite found it...

For the last couple days I was tasked at fixing an old 3.8.x image, which ended up having the exact same problem.. e2fsprogs...  On Debian Stretch/Testing, e2fsprogs 1.43 migrated over 1.42 on 2016-03-28..

One of the changes in 1.43, was to enable for ext4 "metadata_csum & 64bit" by default..

```
sudo mkfs.ext4 -O ^metadata_csum,^64bit -L rootfs ${DISK}2
```

Now it works perfectly fine..  I haven't tested any other boards since this change went, but i bet they are broken too..

Regards,

mhaberler, 2016-04-01:
congratulations - happy to hear you nailed it! now it makes all sense that my install worked and yours did not, since using straight jessie here 

I'm almost done adding the changes for the separate a2 partition and should have something for this repo early next week (on the assumption we're not both working on it, right?)

RobertCNelson, 2016-04-01:
@mhaberler yeah, i'm leaving for the day shortly, so i won't be tweaking stuff till sunday. ;)

Now that we know why, i'll finish my u-boot patch, so we can do:

partition 1: A2
partition 2: ext4 boot/rootfs..

with /boot/uEnv.txt (uname_r) lookup..

Regards,

mhaberler, 2016-08-22:
Hi @RobertCNelson 

we're readying the machinekit SoC/FPGA images for both the CycloneV and Zynq platforms, starting from a debian-8.5-console config, and I am switching to your master

on the `partition 1: A2`  u-boot-with-spl.sfp install - is that finished? how would I go about it? or do I need to carry forward my setup_sdcard.sh cludges ;-?

closed 2018-07-04 by RobertCNelson