RobertCNelson/omap-image-builder #73: Correct usage of local --bootloader argument in create-sdcard.sh?

desflynn, 2016-03-30:
Hey,

I'm trying to insert a custom built bootloader (from RobertCNelson/Bootloader-Builder) locally during the image creation of a jessie distro.

I put the built  MLO-am335x_evm-v2016.03-r6 and u-boot-am335x_evm-v2016.03-r6.img into a folder under omap-image-builder and added --bootloader ../../mydir/ to the setup-sdcard.sh command, but the script exits inside local_bootloader () with an error copying.  ${boot_name} does not seem to be set - not sure if it is needed?

I tried using * at the end of the dir and an absolute path but neither worked for me.

Any suggestions welcome - Maybe my syntax is off in the switch or I am missing something with another flag.   Also - I presume I need to have a local copy of bootloader-ng also?

Many thanks!

Des

RobertCNelson, 2016-03-30:
use
```
--spl /direct/path/MLO-am335x_evm-v2016.03-r6
--bootloader /direct/path/u-boot-am335x_evm-v2016.03-r6.img
```
I don't think i had relative path's working 100%...

desflynn, 2016-03-30:
Awesome, it's building as we speak!

Thanks!

closed 2016-03-30 by desflynn