RobertCNelson/omap-image-builder #77: debian-8.4-iot image lacks ntpdate

hongquan, 2016-04-23:
This image lacks `ntpdate` and `ping` command requires root permission to run.

RobertCNelson, 2016-04-23:
@hongquan don't use ntpdate, systemd-timesyncd is enabled by default...

https://www.freedesktop.org/software/systemd/man/systemd-timesyncd.service.html

(ntpdate has no clue if the network is 'up' so it errors out on startup, whereas systemd-timesyncd knows when the network is connected..)

ping, it should be enabled...

https://github.com/RobertCNelson/omap-image-builder/blob/master/target/chroot/beagleboard.org-jessie.sh#L179-L193

Regards,

closed 2016-04-23 by hongquan

hankchan, 2016-07-05:
when the network is not 'up' on startup,ntpdate may return error date.I've had this kind of problem.