RobertCNelson/omap-image-builder #78: Is this the official build for Debian on BBB?

marcosscriven, 2016-05-20:
I'm not quite sure if the scripts here are the ones used to produce the images at https://beagleboard.org/latest-images ?

Have you thought of setting up Travis on here (since it's open source), then GitHub would be hosting all the latest images automatically.

RobertCNelson, 2016-05-20:
Yes they are..

Travis has a 50 minute timeout, since they are x86, this script would use qemu, which pretty much doubles the build time.

We already have a Jenkins server where regression testing on each commit occurs.

closed 2017-07-18 by RobertCNelson

sinback, 2018-10-11:
This is an old thread, but I'm curious about this comment that using qemu makes the build "longer". Is this still true? Is the build time using qemu on a build server really double that of running the build on a BBB itself?? or are you using a more powerful ARM build server than the BBB?