RobertCNelson/omap-image-builder #79: Explanation of the different configs

marcosscriven, 2016-05-20:
Please can you clarify what each config is for? There seems to be various components in the name:

- There's ```distro-release```, like ```debian-wheezy``` or ```ubuntu-xenial```, which is clear enough
- There's ```bb.org```, ```eewiki```, ```rcn-ee```, ```machinekit```, &```seeed-debian```. I've no idea what each of those are. I've heard of ```machinekit``` (LinuxCNC successor), but I thought that was a platform, not a board type?
- The ```bb.org``` releases seem to have what I assume are various desktop envs - but other releases don't. I'm guessing that means this is the BeagleBone releases (which have a HDMI port)
- The ```eewiki```, & ```rcn-ee``` releases have EL and HF variations

Also - I'm unclear if at this stage whether kernel changes for BeagleBone Black are in the mainline? I note your scripts use your own kernel repo```http://repos.rcn-ee.com/```, does that mean it'll always be a fork that's never integrated upstream?

RobertCNelson, 2016-05-20:
Read the readme.md for the top 4.

On the last, getting a large subsystem like dt-overlays take a lot of time and work. Since v4.1.x a lot of things are in-flight to mainline..

marcosscriven, 2016-05-22:
Thanks for response - I did of course check the readme

However, the first half appears to be hex fingerprints for board revisions, and the second half seems to just have code examples of calling scripts with various ```-c``` options, which even once you realise is the config option, doesn't really seem to explain what ```bb.org```, ```seed-debian``` etc are.

closed 2016-05-22 by marcosscriven

RobertCNelson, 2016-05-23:
bb.org -> beagleboard.org -> https://rcn-ee.net/rootfs/bb.org/testing/ -> http://beagleboard.org/latest-images

seeed -> http://www.seeedstudio.com/

marcosscriven, 2016-05-23:
Thanks for the responses @RobertCNelson