RobertCNelson/omap-image-builder #85: Xenomai kernel issue on Arrow BeagleBone Black

subspclr4, 2016-07-22:
Hello,

I am having an issue booting a Xenomai kernel on an [industrial temperature rated BeagleBone Black](https://www.arrow.com/en/products/bb-bblk-000-itemp/arrow-development-tools) I purchased from Arrow electronics.

I’m not sure where’s the best place to post, but I am posting to the [omap-image-builder](https://github.com/RobertCNelson/omap-image-builder) project because the issue applies to both the eMMC image detailed below and @machinekit’s bone-debian-8.5-machinekit-armhf-2016-06-19-4gb.img.xz image on an SD card.

The following is a description of the problem.  I can flash a minimal Debian image to the eMMC with the following steps:

* wget https://rcn-ee.com/rootfs/2016-06-09/flasher/BBB-eMMC-flasher-debian-8.5-console-armhf-2016-06-09-2gb.img.xz
* xzcat BBB-eMMC-flasher-debian-8.5-console-armhf-2016-06-09-2gb.img.xz | sudo dd of=/dev/sdX
* Hold down the button near SD card until "Cylon" lights start.
* Once the lights go out, power off the unit, remove the SD card, and power it back on.

After power cycling, the unit boots as expected.  However, I cannot boot a pre-packaged Xenomai Linux kernel.  The system hangs on boot using the following steps:

* sudo apt-get update
* sudo apt-get install -y linux-image-3.8.13-xenomai-r79 linux-headers-3.8.13-xenomai-r79 linux-firmware-image-3.8.13-xenomai-r79
* sudo reboot
* The system hangs with all four user LEDs lit and nothing shown over HDMI.

[image]

As mentioned earlier, the same issue happens if I try the recent @machinekit bone-debian-8.5-machinekit-armhf-2016-06-19-4gb.img.xz image on an SD card.  On a regular BeagleBone Black the above steps work flawlessly and Xenomai functions great in all the benchmarks I’ve thrown at it.

**Does anyone have any suggestions for how to fix this?  Also, once fixed, what tests would you recommend to be reasonably assured the Xenomai image is stable on this Arrow board?**

I don’t have a TTL UART adapter handy, but I can try to get one tomorrow to see if more debug information can be gleaned from the boot loader.

Thanks.

-Adam

P.S. Thank you so much @RobertCNelson for providing a packaged Xenomai kernel.  It really makes Xenomai so much more accessible by avoiding the dance with multiple patches, “well aged” kernel commits, and reading up on old blog posts just to get it going.  I greatly appreciate it.

RobertCNelson, 2016-07-22:
@subspclr4 So the problem with the Arrow board, they swapped out the NXP based HDMI framer for an Analog Devices unit.  The Analog Devices patches where written for v4.1.x:

https://github.com/RobertCNelson/ti-linux-kernel-dev/tree/ti-linux-4.1.y/patches/beaglebone/abbbi

With some work, these could be back-ported to our 3.8.13 based kernel.

Regards,

subspclr4, 2016-07-22:
@RobertCNelson Thank you for the quick reply.

That is unfortunate to hear about the HDMI framer.  Is it possible adjust the Device Tree to disable / workaround this HDMI conflict?  For my application I believe (based on [this post](https://nathanielrlewis.com/archives/author/nathaniel)) that I need to disable HDMI anyway since it's muxed with the extra encoder pins I'm looking to use.

I agree those patches could be back ported, but I unfortunately don't have the bandwidth to do that right now given other project deliverables.  I'd also greatly prefer to to stick with a well-tested kernel like the packaged one you kindly offer.

jadonk, 2016-09-28:
@subspclr4 It is indeed possible to cut it out of the device tree and run with a 3.8 kernel. Did you figure it out?

subspclr4, 2016-10-02:
@jadonk No, not yet, but that's mainly because I haven't made it a priority given other project deliverables.  My first pass at adding `cape_disable=capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN` to uEnv.txt worked as expected on an ordinary BeagleBone Black with the 3.8.13 Xenomai kernel, but didn't solve the problem on the Arrow board with the same kernel.

My next thoughts were to try recompiling:
- The base Device Tree blob without HDMI instead of using a capemgr overlay.
- The Xenomai kernel without HDMI support.

Is there anything else you'd recommend I try?

jadonk, 2017-03-04:
Matt Porter at Konsulko might have a good suggestion.

razed11, 2018-02-22:
We just ran into this. Have a deadline to ship units and couldn't source the Element14 board. We're running `3.8.13`. I'd prefer not to have to have to two SD cards for production. Do we think that removing HDMI from the device tree would work on both boards?

RobertCNelson, 2018-02-22:
@razed11 the hdmi encoder is the only difference, if your not using video, just remove it.

razed11, 2018-02-22:
Sorry @RobertCNelson: Remove it in what way? Do I find the source for one of these and rebuild?

RobertCNelson, 2018-02-22:
@razed11 use https://github.com/RobertCNelson/yakbuild

use the "recipe.sh.v3.8.x.sample" sample

then change kernel_tag="3.8.13-bone86" -> kernel_tag="3.8.13-bone62"

You'll find the hdmi cape under /firmware/capes/ or similar and nuke the file and the reference in teh same dir's makefile..

Regards,

razed11, 2018-02-22:
Is found an included dtsi. Remove these and rebuild?

RobertCNelson, 2018-02-22:
yeap

razed11, 2018-02-23:
@RobertCNelson Not much luck so far. I used yakbuild (which I used successfully in the past to increase the SPI buffer size). I removed the HDMI dts files from `KERNEL/firmware/capes`. There is no makefile in that directory but I removed the reference in `KERNEL/firmware/Makefile`. Using `dtc` to convert it back to dts I get:

closed 2018-07-04 by RobertCNelson