RobertCNelson/omap-image-builder #86: fixing ping broken as long as tar is used

mhaberler, 2016-08-23:
in [this patch](https://github.com/RobertCNelson/omap-image-builder/commit/d40a4efdaa41cd54cc2ed7863c5691c3e9f43064)  I tried to fix the ping permissions issue "properly" by using setcap if available

that works within the chroot:
````
mah@cubox-slave:~/omap-image-builder/ignore/tmp.s1acWkkRtu$ sudo getcap  bin/ping
bin/ping = cap_net_raw+ep
````
but as soon as the chroot file system is wrapped into a tar archive the capabilities are either gone, or cannot be restored by any combination of `--xattrs` or `--acls`.

After wasting too much time on this, I propose we rate tar capability save/restore as hopeless and revert to brute-force setuid root for `bin/ping` and `bin/ping6`.

@RobertCNelson - if ok with you I'll propose a patch.

RobertCNelson, 2016-08-23:
@mhaberler sure, we could add it to the list of rootfs finalization tweaks before here:

https://github.com/RobertCNelson/omap-image-builder/blob/master/tools/setup_sdcard.sh#L1454

Regards,

RobertCNelson, 2016-08-23:
Or... we could patch the "ping" debian package and revert the port 80 ping limitation...

closed 2018-07-04 by RobertCNelson