RobertCNelson/omap-image-builder #87: Seperate boot partition on BBB

sylencecc, 2016-08-24:
Hello,

is the classical layout with a separate boot partition still supported on the Beaglebone Black? The official hwpack files don't use it anymore and if I change bootloader_location to 'fatfs_boot' and conf_boot_fstype to 'fat', the flasher fails with 'Failed to mount /' after copying all files over to eMMC.

RobertCNelson, 2016-08-24:
Hi @sylencecc the separate boot partition is still supported, on boards based on omap35xx/dm37xx silicon, since the bootrom wasn't updated till the omap4/am335x generation.  In the beaglebone.conf file under hwpack change:

```
#Bootloader Partition:
bootloader_location="dd_spl_uboot_boot"
#bootrom_gpt="enable"

spl_uboot_name="MLO"
dd_spl_uboot_count="1"
dd_spl_uboot_seek="1"
dd_spl_uboot_conf=""
dd_spl_uboot_bs="128k"
spl_name="MLO"

uboot_name="u-boot.img"
dd_uboot_count="2"
dd_uboot_seek="1"
dd_uboot_conf=""
dd_uboot_bs="384k"
boot_name="u-boot.img"
```

to:
```
#Bootloader Partition:
conf_boot_fstype="fat"
bootloader_location="fatfs_boot"
spl_name="MLO"
boot_name="u-boot.img"
```

To specify the fat partition size (default is 12Mb (smallest valid fat16)), the former size was "96", so add this to beaglebone.conf

```
conf_boot_endmb=96
```

Regards,

RobertCNelson, 2016-08-24:
@sylencecc oh and what flasher? (file name and value of version_message in script)..

sylencecc, 2016-08-24:
Thanks so far, I'm trying it out right now!
I'm using the current flasher (as far as I understood the setup_sdcard.sh script always fetches the newest one from the repo anyway):
init-eMMC-flasher-v3.sh
1.20160718

The last time I built images without issues was back in February, but my omap-image-builder from that time doesn't generate working flashers anymore, too. I guess that is either because something changed in Debian itself (debootstrap) or in the dependencies that are pulled in during the creation process.

sylencecc, 2016-08-24:
Sorry for the inconvenience, but I just noticed that I can't even get the regular flasher (single-partition) without any modifications on my side to work.
All I did was
```
RootStock-NG.sh -c bb.org-debian-jessie-console-v4.1
setup_sdcard.sh --img-2gb "flasher" --dtb beaglebone --hostname beaglebone --emmc-flasher --bbb-old-bootloader-in-emmc
```
Then after flashing the resulting to a SD card and inserting it everything looks alright. This process ends with
```
[  154.995137] reboot: Power down
[  154.998457] System will go to power_off state in approx. 1 second
```
After restarting the device it either starts flashing again if I leave the SD card inserted (this wasn't the case back in February) or it just can't boot from eMMC if I put the SD card out:
```
U-Boot 2016.09-rc2-00001-g1f14af4 (Aug 23 2016 - 15:12:35 -0500), Build: jenkins-github_Bootloader-Builder-440

DRAM:  512 MiB
Reset Source: Power-on reset has occurred.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Using default environment

Net:   
Warning: ethernet@4a100000 using MAC address from ROM
eth0: ethernet@4a100000
Press SPACE to abort autoboot in 2 seconds
MMC: no card present
MMC: no card present
MMC: no card present
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
MMC: no card present
MMC: no card present
** First descriptor is NOT a primary desc on 1:1 **
switch to partitions #0, OK
mmc1(part 0) is current device
** No partition table - mmc 1 **
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
** First descriptor is NOT a primary desc on 1:1 **
switch to partitions #0, OK
mmc1(part 0) is current device
** First descriptor is NOT a primary desc on 1:1 **
gpio: pin 54 (gpio 54) value is 1
** No partition table - mmc 1 **
Checking for: /uEnv.txt ...
** No partition table - mmc 1 **
Checking for: /boot.scr ...
** No partition table - mmc 1 **
Checking for: /boot/boot.scr ...
** No partition table - mmc 1 **
Checking for: /boot/uEnv.txt ...
** No partition table - mmc 1 **
** No partition table - mmc 1 **
** No partition table - mmc 1 **
** No partition table - mmc 1 **
** No partition table - mmc 1 **
** No partition table - mmc 1 **
** No partition table - mmc 1 **
## Error: "bootcmd_nand0" not defined
starting USB...
USB0:   Port not available.
link up on port 0, speed 100, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.1.169 (13 ms)
Using ethernet@4a100000 device
TFTP from server 192.168.1.1; our IP address is 192.168.1.169
Filename 'zImage'.
Load address: 0x82000000
Loading: T T 
U-Boot SPL 2016.09-rc2-00001-g1f14af4 (Aug 23 2016 - 15:12:35)
Trying to boot from MMC2
Expected Linux image is not found. Trying to start U-boot
```

RobertCNelson, 2016-08-24:
Just ran the un-modified:

ARM:
```
./RootStock-NG.sh -c bb.org-debian-jessie-console-v4.1
```

x86:
```
wget http://qcom.local/ssd/git/omap-image-builder/deploy/debian-8.5-console-armhf-2016-08-24.tar
tar xf debian-8.5-console-armhf-2016-08-24.tar 
cd debian-8.5-console-armhf-2016-08-24/
sudo ./setup_sdcard.sh --img-2gb "flasher" --dtb beaglebone --hostname beaglebone --emmc-flasher --bbb-old-bootloader-in-emmc
bmaptool create -o flasher-2gb.img.bmap flasher-2gb.img 
sudo bmaptool copy flasher-2gb.img /dev/sde
```

flashed and booted fine..

```
Debian GNU/Linux 8 beaglebone ttyS0

BeagleBoard.org Debian Image 2016-08-24

Support/FAQ: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian

default username:password is [debian:temppwd]

The IP Address for usb0 is: 192.168.7.2
beaglebone login: debian
Password: 
Linux beaglebone 4.1.30-ti-r69 #1 SMP Sun Aug 14 11:23:09 UTC 2016 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
debian@beaglebone:~$ uname -r
4.1.30-ti-r69
```

Next i'll add the beaglebone.conf modifications and rerun...

Regards,

RobertCNelson, 2016-08-24:
```
-----------------------------
Writing bootloader to [/dev/mmcblk1]
dd if=/opt/backup/uboot/ of=/dev/mmcblk1 conv=notrunc 
-----------------------------
dd: failed to open '/opt/backup/uboot/'[   30.886179] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
[   30.886179] 
[   30.899136] CPU: 0 PID: 1 Comm: init-eMMC-flash Not tainted 4.1.30-ti-r69 #1
[   30.906211] Hardware name: Generic AM33XX (Flattened Device Tree)
[   30.912369] [<c001521d>] (unwind_backtrace) from [<c0011541>] (show_stack+0x11/0x14)
[   30.920153] [<c0011541>] (show_stack) from [<c06aeec7>] (dump_stack+0x73/0x80)
[   30.927423] [<c06aeec7>] (dump_stack) from [<c06ad149>] (panic+0x79/0x19c)
[   30.934340] [<c06ad149>] (panic) from [<c0034f6d>] (complete_and_exit+0x1/0x18)
[   30.941685] [<c0034f6d>] (complete_and_exit) from [<00000002>] (0x2)
[   30.948080] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
[   30.948080] 
: No such file or directory
```

sylencecc, 2016-08-24:
I'm also in the process of building clean images from the unmodified repository. I'm not running Rootstock on a native ARM machine, but on qemu. Not sure if this is an issue, it wasn't in the past. Is `bmaptool` optional? I'm using `dd` to write the img files directly to SD.

sylencecc, 2016-08-24:
Building the clean default image (using dd instead of bmaptool and qemu-arm-static) still lead to an unbootable system. I made a gist with the serial console output of the whole process (inserting flasher SD and removed it for the first manual power cycle):
https://gist.github.com/sylencecc/85c17c6da9b5473a410ac7d87f80f75c

RobertCNelson, 2016-08-24:
This get's closer:
```
#!/bin/bash
#        --dtb beaglebone (BeagleBone & BeagleBone Black:supported)

#Bootloader:
conf_board="am335x_evm"
conf_bl_http="https://rcn-ee.com/repos/bootloader/latest"
conf_bl_listfile="bootloader-ng"

#Bootloader Partition:
bootloader_location="dd_spl_uboot_boot"
#bootrom_gpt="enable"

spl_uboot_name="MLO"
dd_spl_uboot_count="1"
dd_spl_uboot_seek="1"
dd_spl_uboot_conf=""
dd_spl_uboot_bs="128k"
spl_name="MLO"

uboot_name="u-boot.img"
dd_uboot_count="2"
dd_uboot_seek="1"
dd_uboot_conf=""
dd_uboot_bs="384k"
boot_name="u-boot.img"

conf_boot_fstype="fat"
conf_boot_endmb="96"

#Kernel:
conf_kernel="bone"
usbnet_mem=
dtb=
SERIAL="ttyO0"
drm_device_identifier="HDMI-A-1"
```

just add teh 
```
conf_boot_fstype="fat"
conf_boot_endmb="96"
```

It's failing here:

fixed: https://github.com/RobertCNelson/boot-scripts/commit/cadd682ed5c308524e2764733cfbfe0a5388b427

```
rsync: /boot/uboot/ -> /tmp/boot/
umount: /boot/uboot: not mounted
umount: /boot/uboot: not mounted
[   42.181608] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00002000
[   42.181608] 
[   42.191475] CPU: 0 PID: 1 Comm: init-eMMC-flash Not tainted 4.1.30-ti-r69 #1
[   42.198550] Hardware name: Generic AM33XX (Flattened Device Tree)
[   42.204708] [<c001521d>] (unwind_backtrace) from [<c0011541>] (show_stack+0x11/0x14)
[   42.212492] [<c0011541>] (show_stack) from [<c06aeec7>] (dump_stack+0x73/0x80)
[   42.219761] [<c06aeec7>] (dump_stack) from [<c06ad149>] (panic+0x79/0x19c)
[   42.226678] [<c06ad149>] (panic) from [<c0034f6d>] (complete_and_exit+0x1/0x18)
[   42.234022] [<c0034f6d>] (complete_and_exit) from [<00000002>] (0x2)
[   42.240415] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00002000
[   42.240415] 
```

bmaptools is just way faster then dd. ;)
qemu-arm-static works fine, native on a quad core arm is just faster...

Regards

RobertCNelson, 2016-08-24:
Okay, so here's where i broke it:

https://github.com/RobertCNelson/omap-image-builder/commit/0260050539f98174bce73824e347fa851fe45d10

Most of that was okay, except for this hack:

```
 		echo "Using sfdisk to create partition layout"
  		echo "Version: `LC_ALL=C sfdisk --version`"
  		echo "-----------------------------"
 -		if [ "x${bborg_production}" = "xenable" ] ; then
 -			conf_boot_endmb="96"
 -		fi
  		sfdisk_partition_layout
  		;;
  	dd_uboot_boot)
 @@ -593,15 +590,8 @@ create_partitions () {
  		dd_spl_uboot_boot
  		dd_uboot_boot
  		bootloader_installed=1
 -		if [ "x${bborg_production}" = "xenable" ] ; then
 -			conf_boot_endmb="96"
 -			conf_boot_fstype="fat"
 -			sfdisk_fstype="0xE"
 -			sfdisk_partition_layout
 -		else
 -			sfdisk_single_partition_layout
 -			media_rootfs_partition=1
 -		fi
 +		sfdisk_single_partition_layout
 +		media_rootfs_partition=1
  		;;
  	*)
  		echo "Using sfdisk to create partition layout"
```

i need to add "bborg_production" or similar flag, and overide that partition creation section..

Regards,

sylencecc, 2016-08-24:
I figured that this commit broke something, but didn't know the under-the-hood details to build a workaround.

I'm still struggling to build the default single-partition flasher, though. Do you see anything unusual in the gist log I posted above? I'm currently saving the output of the qemu build process to browse it for errors later. If that doesn't lead to anything, I might try to run Rootstock on an ARM system and exactly replicate your process above.

RobertCNelson, 2016-08-24:
Give this change to "setup_sdcard.sh" a shot:

https://github.com/RobertCNelson/omap-image-builder/commit/f24e65c7b7b63a13c41216f893cde05529101722

wget https://github.com/RobertCNelson/omap-image-builder/raw/master/tools/setup_sdcard.sh

untested, but i think it'll work, heading to lunch, so be back in a bit..

Regards,

RobertCNelson, 2016-08-24:
okay everything looks good here:

```
Debian GNU/Linux 8 beaglebone ttyS0

BeagleBoard.org Debian Image 2016-08-24

Support/FAQ: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian

default username:password is [debian:temppwd]

The IP Address for usb0 is: 192.168.7.2
beaglebone login: debian
Password: 
Linux beaglebone 4.1.30-ti-r69 #1 SMP Sun Aug 14 11:23:09 UTC 2016 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
debian@beaglebone:~$ lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0boot0 179:8    0    2M  1 disk 
mmcblk0boot1 179:16   0    2M  1 disk 
mmcblk0      179:0    0  3.7G  0 disk 
|-mmcblk0p1  179:1    0   96M  0 part 
`-mmcblk0p2  179:2    0  3.6G  0 part /
debian@beaglebone:~$ uname -r
4.1.30-ti-r69
```

sylencecc, 2016-08-25:
Alright, I'm eager to try out that layout, since my use case heavily relies on that single FAT partition. However, I'm still unable to build a single flasher that actually works. I just ran Rootstock on a Beaglebone, like you described above. Everything seems to build without any issues, the flasher also does what it's supposed to, but after a restart I'm back at square one: The board can't find a valid partition and doesn't boot [1].
`** First descriptor is NOT a primary desc on 1:1 **`
`** No partition table - mmc 1 **`
I'm completely out of ideas right now. The official flasher images are working fine, though.

[1] https://gist.github.com/sylencecc/85c17c6da9b5473a410ac7d87f80f75c

schallee, 2016-08-30:
Following along as my use case has a strong need for a separate boot partition (eg: unstable power). Tell me if you'd prefer this in a separate issue as this one has diverged towards flashing issues.

When I change the options to include --enable-fat-partition I do get two partitions but the boot partition is actual ext4 (which is actually preferred by me and my use case but not what --enable-fat-partition implies;).

Additionally, the boot partition is not added to fstab so /boot is actually on the root file system. While the boot loader is included on the boot partition, the kernels and such get added to /boot on the root partition. 

I'm happy to implement a fix and  put together a pull request but figured I would chime in before doing so.

Thanks!

RobertCNelson, 2016-08-30:
@schallee for un-stable power, use:

```
--ro
```
It's hardcoded to 2048:
https://github.com/RobertCNelson/omap-image-builder/blob/master/tools/setup_sdcard.sh#L1705

So you'll have to play with that variable depending on the rootfs size..

Regards,

schallee, 2016-08-30:
Yeah, I'm headed for --ro but even with that I need to be able to do OS upgrades even on flaky power. Of course you have to mount -oremount,rw /boot to to upgrade kernels but it is still a smaller window then apt-get upgrade. Being able to get an initramfs off of boot that can do a fsck is still of value.

RobertCNelson, 2016-08-30:
@schallee well what you really want is /boot/<zImage/initrd.img/dtbs> on ext2, then the rootfs split in two, ro and var..

Regards,

schallee, 2016-08-30:
Exactly. Sorry I didn't explain it too well. Something with a fstab like

/dev/mmcblkXp2 / ext2 ro 
/dev/mmcblkXp1 /boot ext2 ro
/dev/mmcblkXp3 /var ext4 rw

I'm thinking to put the sensor feeds, the volume of which is yet to be determined, on a sdcard to reduce the wear on the emmc. 

Thanks

closed 2018-07-04 by RobertCNelson