RobertCNelson/omap-image-builder #90: Disabling all USB Device behavior except for chosen HID class

ElSordo, 2016-10-26:
I'm trying to define an hid device using ConfigFs.  The problem I run into is that when I try to set up the UDC attribute of the device that I have defined, I get a device busy error.

In looking at the output of lsusb, I see a lot of modules that depend on libcomposite.  So, what I would like to do is to remove all of them, e.g., usb_f_acm, g_serial, u_serial, ...  Basically, I only want libcomposite to be loaded.

So how do I tell the boot process not to load these beasties?

-David Glaser

RobertCNelson, 2016-10-27:
They get loaded by an early ran boot script under: /opt/scripts/boot/

Remove this block:

https://github.com/RobertCNelson/boot-scripts/blob/master/boot/generic-startup.sh#L72-L75

Regards,

ElSordo, 2016-10-27:
The fixed worked like a charm.
I'm up and running with my hid function.
Thanks

closed 2016-10-27 by ElSordo

ElSordo, 2016-10-27:
Worked like a charm

Thanks