RobertCNelson/omap-image-builder #97: beaglebone-universal-io

bigguiness, 2017-03-03:
Is this package available as a .deb or only as the git repo?

It appears to be installed on the wheezy lxde and jessie lxgt images but it is missing on both the console images.

Thanks

RobertCNelson, 2017-03-03:
i need to package this..

```
git clone https://github.com/cdsteinkuehler/beaglebone-universal-io.git
sudo ln -s `pwd`/beaglebone-universal-io/config-pin /usr/local/bin/
```

bigguiness, 2017-03-03:
Doesn't it actually need to be built and installed in order for the overlays to be in /lib/firmware?

RobertCNelson, 2017-03-03:
no it's just a shell script to help interpret /sys/...

bigguiness, 2017-03-03:
I understand that config-pin is a shell script. But doesn't it need the overlays from the repo?

RobertCNelson, 2017-03-03:
the overlays are provided by the bb-cape-overlays package.

bigguiness, 2017-03-03:
Isn't that package only available on the jessie images?

RobertCNelson, 2017-03-03:
Jessie/Stretch and Xenial

closed 2018-07-04 by RobertCNelson