RobertCNelson/omap-image-builder #98: Some helpful packages for Blue

jadonk, 2017-03-04:
I'd really like to have mjpg-streamer and avahi-utils installed by default. While it is generally easier to get on a network in a classroom setting now, having avahi-utils really helps find other boards on the network and mjpg-streamer gives a nice quick demo.

RobertCNelson, 2017-03-04:
mjpg-streamer is already installed in the iot and lxqt-4g, i'll add avahi-utils now..

closed 2017-05-10 by jadonk

JochiPochi, 2017-05-11:
@RobertCNelson, mjpg-streamer and avahi-utils are definitely nice to have but there is something uncomfortable about adding packages out of the blue just because we like them.
What if some users don't want this?
What worries me is that each time more additions like this happen, the images built with the omap-image-builder deviate more and more from a standard Ubuntu and Debian installation but they are branded as such.

I encountered this problem not too long ago when conman was added to the image. It seriously messed up the network configuration we were used to using and it took a couple of days to find out that conman was the cause.

My point is, @RobertCNelson do you think it would be a good idea to offer a virgin Ubuntu installation along side the spiced Ubuntu and Debian you currently offer?

It would really help those that expect those distros to be as Debian or Canonical intended instead of being surprised by these extra additions when no one is expecting them. 

I can help make a PR that allows to do this if you think it would be a worthy contribution. 

Thanks,

JochiPochi, 2017-05-11:
I can see that conman was just removed 1 hour before I made the previous comment. I'm happy about that, but it's also a surprise. I just know that because I happen to read this issue. And that's kind of my point. Now that the beaglebone is being used by so many people, for education, industry and hobbyist surprises can be huge sinks of time.
@RobertCNelson @jadonk thoughts?

RobertCNelson, 2017-05-11:
@JochiPochi for a more virgin install, just use the netinstall

```
https://github.com/RobertCNelson/netinstall
```

Regards,

JochiPochi, 2017-05-11:
@RobertCNelson, Thank you. This clarifies a lot. I think I got a bit confused because all the images on elinux.org and beaglebone.org have very similar names so I didn't realize 
https://github.com/RobertCNelson/netinstall
produced what I was looking for.

I'm going to start playing with it right away.