RobertCNelson/omap-image-builder #100: Another BeagleBone Black revision number in the wild

BrianAdams, 2017-03-15:
aa5533ee41333335424e4c5430303042

May  have been from an Element14 run

RobertCNelson, 2017-03-16:
@jadonk another one snuck by..

@BrianAdams run this with the wp-grounded on the eeprom  (TP4)

```
wget https://raw.githubusercontent.com/RobertCNelson/boot-scripts/master/device/bone/bbb-eeprom.dump
dd if=bbb-eeprom.dump of=/sys/devices/platform/ocp/44e0b000.i2c/i2c-0/0-0050/eeprom
```

jadonk, 2017-05-10:
2GB eMMC? Rev B is listed on the wiki: http://elinux.org/Beagleboard:BeagleBoneBlack#Revision_B

jadonk, 2017-05-22:
@RobertCNelson will support for Rev B be added, since it is valid?

RobertCNelson, 2017-05-22:
@jadonk @BrianAdams can we get the hexdump version:

```
debian@test-bbb-1:/opt/scripts/tools$ sudo ./version.sh | grep eeprom
eeprom:[A335BNLT0A5A1813BBBK3053]
```
Regards,

closed 2018-07-04 by RobertCNelson