RobertCNelson/omap-image-builder #107: Proxy override/disabling Issues

DuaneEllis-TI, 2017-07-05:
These overrides should not exist.

* https://github.com/beagleboard/image-builder/blob/master/scripts/chroot.sh#L326
* https://github.com/beagleboard/image-builder/blob/master/scripts/chroot.sh#L330

They are most likely unique to someones test or development environment, in contrast when others are truely behind a proxy and must use a proxy these two settings cause nothing but problems.

RobertCNelson, 2017-07-05:
Testing this:

https://github.com/RobertCNelson/omap-image-builder/commit/c90e7368b3e905a579951f9177a47bac31561730

Right now:
http://gfnd.rcn-ee.org:8080/job/bb.org-debian-jessie-lxqt-4gb-v4.4/82/console

I don't remember if it was wheezy apt-cacher-ng issue or what, so let's see what the builder does. (running debian stretch)

Regards,

DuaneEllis-TI, 2017-07-05:
Looking at the output of Jenkins it does fail.  That said, I think this proxy configuration is 100% unique to your Jenkins build environment.

For example, if I was to setup a proxy environment why would I choose these two sites and not two other sites to require a direct connection.

RobertCNelson, 2017-07-06:
Yeap, it failed, so i can't just remove it, instead i need to work around it..

Regards,

closed 2017-07-06 by RobertCNelson