RobertCNelson/omap-image-builder #108: stretch: couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)

Pillar1989, 2017-09-04:
Hi 
I used ./RootStock-NG.sh -c bb.org-debian-stretch-lxqt-v4.9.conf build stretch image.   Get apt warnning.
```
Setting up git-core (1:2.11.0-3+deb9u1) ...
Setting up libdns162:armhf (1:9.10.3.dfsg.P4-12.3+deb9u2) ...
Setting up apache2 (2.4.25-3+deb9u2) ...
Running in chroot, ignoring request.
invoke-rc.d: policy-rc.d denied execution of start.
Running in chroot, ignoring request.
invoke-rc.d: policy-rc.d denied execution of restart.
Setting up libisccfg140:armhf (1:9.10.3.dfsg.P4-12.3+deb9u2) ...
Setting up libbind9-140:armhf (1:9.10.3.dfsg.P4-12.3+deb9u2) ...
Setting up bind9-host (1:9.10.3.dfsg.P4-12.3+deb9u2) ...
Processing triggers for initramfs-tools (0.130) ...
Processing triggers for libc-bin (2.24-11+deb9u1) ...
W: Download is performed unsandboxed as root as file '/var/cache/apt/archives/partial/libxml2_2.9.4+dfsg1-2.2+deb9u1_armhf.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Log: (chroot) Installing: chromium lightdm lxqt-about lxqt-admin lxqt-common lxqt-config lxqt-panel lxqt-qtplugin lxqt-session lxqt-sudo lxqt-runner mesa-utils-extra pcmanfm-qt qterminal tightvncserver xinput xserver-xorg-video-fbdev xserver-xorg-video-omap alsa-utils at apache2 automake avahi-utils bash-completion bc bluetooth build-essential ca-certificates can-utils connman cpufrequtils curl device-tree-compiler dosfstools dnsmasq fbset firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-libertas firmware-misc-nonfree firmware-realtek firmware-ti-connectivity firmware-zd1211 git-core haveged hdparm hexedit hostapd htop i2c-tools initramfs-tools iperf iw less libiio-utils libncurses5-dev libnss-mdns libtool libdbus-1-dev libusb-1.0-0-dev libudev-dev linux-base linux-cpupower locales lshw lsof lzma lzop memtester net-tools openssh-server pastebinit pkg-config python-dev python3 python3-dev python3-numpy python3-setuptools rfkill rsync screen ssl-cert sudo systemd tio tmux u-boot-tools udhcpd usb-modeswitch usbutils v4l-utils vim wget wireless-tools wpasupplicant xz-utils ardupilot-rover-blue ardupilot-rover-bbbmini ardupilot-copter-blue ardupilot-copter-bbbmini ardupilot-plane-blue ardupilot-plane-bbbmini bb-cape-overlays bb-customizations bb-node-red-installer bb-wl18xx-firmware bone101 bonescript c9-core-installer doc-beaglebone-getting-started doc-beaglebonegreen-getting-started doc-seeed-bbgw-getting-started firmware-am57xx-opencl-monitor ipumm-dra7xx-installer linux-image-4.4.84-ti-r120 mjpg-streamer nodejs pru-software-support-package rcn-ee-archive-keyring roboticscape ti-pru-cgt-installer tiomapconf vpdma-dra7xx-installer xserver-xorg-video-armada-etnaviv
Hit:1 http://repos.rcn-ee.com/debian stretch InRelease
Ign:2 http://deb.debian.org/debian stretch InRelease
Hit:3 http://deb.debian.org/debian stretch-updates InRelease
Hit:4 http://deb.debian.org/debian-security stretch/updates InRelease
Hit:5 http://deb.debian.org/debian stretch Release
Reading package lists... Done                     
W: Download is performed unsandboxed as root as file '/var/lib/apt/lists/partial/deb.debian.org_debian_dists_stretch_InRelease' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
alsa-utils is already the newest version (1.1.3-1).
at is already the newest version (3.1.20-3).
automake is already the newest version (1:1.15-6).
avahi-utils is already the newest version (0.6.32-2).
bash-completion is already the newest version (1:2.1-4.3).
bc is already the newest version (1.06.95-9+b3).
bluetooth is already the newest version (5.43-2).
build-essential is already the newest version (12.3).
```

RobertCNelson, 2017-09-04:
this isn't a bug, it's an annoying feature of apt in stretch. Just ignore all "Download is performed unsandboxed as root as file" msg's..

Pillar1989, 2017-09-05:
Really ? I didnt search any official  explanation .

RobertCNelson, 2017-09-05:
Early on in the Stretch cycle this error printed out:

```
No sandbox user '_apt' on the system, can not drop privileges
```

It was later changed to:
```
Download is performed unsandboxed as root as file
```

Essentially, "as" root we used "apt" and download a *.deb file (or files) thru the network.  "apt" has a new feature to download the *.deb file as "_apt" user (non-root, non-login, sandboxed/etc..).

However at the time of this error in our script, we still do not have a non-root account available for use.

Search for both,

Pillar1989, 2017-09-05:
Thank you,  Get it.

closed 2017-09-05 by Pillar1989

d78ui98, 2018-06-25:
So this is just a warning that I can ignore?

RobertCNelson, 2018-06-25:
@d78ui98 not our bug, talk to upstream developers of "apt" in debian..

d78ui98, 2018-06-25:
@RobertCNelson I am getting this warning only when I am using `vagrant cachier` plugin. Should I report to developer ?

RobertCNelson, 2018-06-25:
@d78ui98 do whatever you want, it's not my package/bug... It's just a warning from apt..

RobertCNelson, 2018-06-25:
renamed, hopefully users googling doesn't find this out of topic issue..