RobertCNelson/omap-image-builder #109: PWM permission problem with new unrooted environment

ulitiy, 2017-09-27:
```
debian@beaglebone:~$ ls -la /sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip0/pwm0/enable
-rw-r--r-- 1 root root 4096 Sep 27 19:53 /sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip0/pwm0/enable
```

I get `Error: EACCES: permission denied, open '/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip0/pwm0/enable'` when trying to `analogWrite` so I've added rules

```
SUBSYSTEM=="pwm", ACTION=="add", PROGRAM="/bin/sh -c '/bin/chown -R root:pwm /sys/devices/platform/ocp/*.epwmss'"
# Change user permissions to ensure user and group have read/write permissions
SUBSYSTEM=="pwm", ACTION=="add", PROGRAM="/bin/sh -c '/bin/chmod -R ug+rw /sys/devices/platform/ocp/*.epwmss'
```

but it doesn't help when rebooted. How can I fix that permanently?

RobertCNelson, 2017-09-27:
This patch:

https://github.com/RobertCNelson/bb-kernel/blob/am33x-v4.11/patches/drivers/pwm/0001-pwm-Create-device-class-for-pwm-channels.patch

or v4.11.x-bone+ kernel or later:

Regards,

ulitiy, 2017-09-27:
@RobertCNelson can you tell me where can I find a usable kernel-bonescript pair, please?
I've installed "testing" kernel, 4.13.4-bone1, and apparently pwm paths are not compatible with this kernel.
As for the patch http://elinux.org/BeagleBoardLinuxKernel looks a bit complicated, I'd better go for an older version with root-env but the one that works. `Debian 8.7 2017-03-19 4GB SD LXQT` can you tell me please if this version is reliable?

RobertCNelson, 2017-09-27:
@ulitiy and you see why i haven't backported that patch to v4.4.x/v4.9.x..

bonescript needs to be patched. (and also the adafruit library), you'll notice in the pwm patch it's only 3 more characters.  creating issue here:

https://github.com/jadonk/bonescript/issues/156

Regards,

closed 2018-07-04 by RobertCNelson