RobertCNelson/omap-image-builder #110: IP Address change in /etc/network/interfaces is added as secondary IP

wdurocher33, 2017-10-23:
Because of compatibility with other devices, we are trying to configure a BBBW through the /etc/network/interfaces. The issue we are encountering is that the IP address we configure is used as a 'secondary' IP when the interface is restarted.

Example: Changing the wlan0 interface static IP Initially we have one IP on wlan0

output from "ip a s":

> 4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
> link/ether 50:65:83:dc:43:cc brd ff:ff:ff:ff:ff:ff
> inet 192.168.0.90/24 brd 192.168.0.255 scope global wlan0
>    valid_lft forever preferred_lft forever
> inet6 fe80::5265:83ff:fedc:43cc/64 scope link 
>    valid_lft forever preferred_lft forever
> 

ifconfig shows:

> wlan0     Link encap:Ethernet  HWaddr 50:65:83:dc:43:cc  
>       inet addr:192.168.0.90  Bcast:192.168.0.255  Mask:255.255.255.0

We edit /etc/network/interfaces with :

> iface wlan0 inet static
> address 192.168.0.91
> netmask 255.255.255.0
> gateway 192.168.0.1
> wpa-ssid SOME_SSID
> wpa-psk SOME_KEY
> 

Then we issue :

sudo ifdown wlan0
sudo ifup wlan0
sudo service networking restart

And we see:

ip a s :

> 4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
> link/ether 50:65:83:dc:43:cc brd ff:ff:ff:ff:ff:ff
> inet 192.168.0.90/24 brd 192.168.0.255 scope global wlan0
>    valid_lft forever preferred_lft forever
> inet 192.168.0.91/24 brd 192.168.0.255 scope global secondary wlan0
>    valid_lft forever preferred_lft forever
> inet6 fe80::5265:83ff:fedc:43cc/64 scope link 
>    valid_lft forever preferred_lft forever

ifconfig:

> wlan0     Link encap:Ethernet  HWaddr 50:65:83:dc:43:cc  
>       inet addr:192.168.0.90  Bcast:192.168.0.255  Mask:255.255.255.0

As you can see the new IP address is taken as a secondary IP. Ifconfig only shows the primary IP and not the secondary IP. We can connect to the device with both address but the presence of the old IP address is not acceptable. We also see this happen on the wired  ethernet interface we added through a USB-Ethernet dongle.

Note that on a reboot the correct IP address will be used.

What is causing this behaviour ?
How can we make it behave as expected. We expect that the IP in the interfaces file is the only one used !

wdurocher33, 2017-10-23:
Also note that the device is accessible through the new IP provided. But we really do not want the old IP to remain. It will be assigned to a different device.

wdurocher33, 2017-10-23:
Forgot to mention that is behaviour is present on both the Debian and the Ubuntu 16.04.3 images.

RobertCNelson, 2017-10-23:
My first guess, your probably fighing connman, so make sure that's removed.

Regards,

wdurocher33, 2017-10-23:
We have already disabled connman on the Debian distribution. On the Ubuntu 16.04 distribution connman does not appear to be available.

wdurocher33, 2017-10-23:
Though as mentioned, I have seen this on both the Debian and the Ubuntu 16.04.3 image I will concentrate only on the Ubuntu image. So as to reduce confusion.

wdurocher33, 2017-10-23:
It appears that if the IP address is removed for the IP configuration via the command:

ip address del xx.yy.zz.aa/mm dev mydevice

The network restart forgets the IP address in question. The issue still remains that this is not the expected behaviour. Is there a script of file that would keep track of the IP addresses currently configured ?

wdurocher33, 2018-03-20:
The issue is caused by BBB scripts that add the current IP address on a network restart.
My solution was to switch to networkd service.

closed 2018-03-20 by wdurocher33