RobertCNelson/omap-image-builder #113: PocketBeagle  takes 2 Minutes to start

patrickpoirier51, 2018-01-04:
Pocket Beagle with sensor cape and WIFI on a TP-LINK 722N
Boot sequence :
0-15 seconds Uboot 
15 - 60 secs  gets in a WAIT STATE  === Does it perform a File System repair ? = same delay with or without a shutdown
60-90 Goes to login
90-120 Seconds Wireless is detected 
 
Does 2 minutes for complete boot sequence is normal ?
In the type of application we are looking for (Flight Controler) this is a concern.

=========================================
git:/opt/scripts/:[ef2c5982e3788e07d9ec4a3d23089dc5e5a3f9cd]
eeprom:[A335PBGL00A21741GPB43449]
model:[TI_AM335x_PocketBeagle]
dogtag:[BeagleBoard.org Debian Image 2017-10-10]
bootloader:[microSD]:[/dev/mmcblk0]:[U-Boot 2017.09-00002-g0f3f1c7907]
kernel:[4.14.5-bone-rt-r8]
nodejs:[v6.12.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20171207.0-0rcnee1~stretch+20171207]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee1~stretch+20170829]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]

===============================
Dmesg (beginning)
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.5-bone-rt-r8 (root@b9-am57xx-beagle-x15-2gb) (gcc version 6.3.0 20170516 (Debian 6.3.0-18)) #1 PREEMPT RT Tue Dec 12 22:12:53 UTC 2017
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=50c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: TI AM335x PocketBeagle
[    0.000000] Memory policy: Data cache writeback
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 48 MiB at 0x9d000000
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat c0fd6a84, node_mem_map dcb61000
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 449220K/524288K available (8192K kernel code, 967K rwdata, 3916K rodata, 1024K init, 684K bss, 25916K reserved, 49152K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                   lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf800000 - 0xbfe00000   (   6 MB)
                     .text : 0xc0008000 - 0xc0900000   (9184 kB)
                     .init : 0xc0e00000 - 0xc0f00000   (1024 kB)
                     .data : 0xc0f00000 - 0xc0ff1cac   ( 968 kB)
                      .bss : 0xc0ffaf20 - 0xc10a6124   ( 685 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 38864 entries in 76 pages
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000]  No expedited grace period (rcu_normal_after_boot).
[    0.000000]  Tasks RCU enabled.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000015] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000028] clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000035] OMAP clocksource: timer1 at 24000000 Hz
[    0.001012] timer_probe: no matching timers found
[    0.001202] Console: colour dummy device 80x30
[    0.001243] WARNING: Your 'console=ttyO0' has been replaced by 'ttyS0'
[    0.001245] This ensures that you still see kernel messages. Please
[    0.001247] update your kernel commandline.
[    0.001264] Calibrating delay loop... 995.32 BogoMIPS (lpj=1990656)
[    0.018217] pid_max: default: 32768 minimum: 301
[    0.018448] Security Framework initialized
[    0.018457] Yama: becoming mindful.
[    0.018528] AppArmor: AppArmor initialized
[    0.018722] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.018730] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.019567] CPU: Testing write buffer coherency: ok
[    0.020294] Setting up static identity map for 0x80100000 - 0x80100054
[    0.020508] Hierarchical SRCU implementation.
[    0.023330] EFI services will not be available.
[    0.024718] devtmpfs: initialized
[    0.048757] VFP support v0.3: implementor 41 architecture 3 part 30 variant c rev 3
[    0.049147] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.049166] futex hash table entries: 256 (order: 1, 8192 bytes)
[    0.052762] xor: measuring software checksum speed
[    0.090274]    arm4regs  :  1222.000 MB/sec
[    0.130219]    8regs     :  1095.000 MB/sec
[    0.170212]    32regs    :  1093.000 MB/sec
[    0.170218] xor: using function: arm4regs (1222.000 MB/sec)
[    0.170234] pinctrl core: initialized pinctrl subsystem
[    0.170858] DMI not present or invalid.
[    0.171281] NET: Registered protocol family 16
[    0.174734] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.212676] omap_hwmod: debugss: _wait_target_disable failed
[    0.269585] OMAP GPIO hardware version 0.1
[    0.287542] No ATAGs?

RobertCNelson, 2018-01-04:
can you post your full kernel log, i know wifi can take a 'bit' to connect, but with v4.14.x i've been trying to cut things way down.. < 1 second boot, < 10 seconds login...

```
journalctl -all > file.log
```

RobertCNelson, 2018-01-04:
PS, how do you guys feel about a "special" console build...  i'm thinking a "really" cut down kernel, with "just" enough ardupilot stuff, maybe btrfs with snapshot support enabled?

patrickpoirier51, 2018-01-04:
cat file.log
-- Logs begin at Thu 2016-11-03 17:16:43 UTC, end at Wed 2018-01-03 22:46:05 UTC. --
Nov 03 17:16:43 beaglebone kernel: Booting Linux on physical CPU 0x0
Nov 03 17:16:43 beaglebone kernel: Linux version 4.14.5-bone-rt-r8 (root@b9-am57xx-beagle-x15-2gb) (gcc version 6.3.0 20170516 (Debian 6.3.0-18)) #1 PREEMPT RT Tue Dec 12 22:12:53 UTC 2017
Nov 03 17:16:43 beaglebone kernel: CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=50c5387d
Nov 03 17:16:43 beaglebone kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Nov 03 17:16:43 beaglebone kernel: OF: fdt: Machine model: TI AM335x PocketBeagle
Nov 03 17:16:43 beaglebone kernel: Memory policy: Data cache writeback
Nov 03 17:16:43 beaglebone kernel: efi: Getting EFI parameters from FDT:
Nov 03 17:16:43 beaglebone kernel: efi: UEFI not found.
Nov 03 17:16:43 beaglebone kernel: cma: Reserved 48 MiB at 0x9d000000
Nov 03 17:16:43 beaglebone kernel: On node 0 totalpages: 131072
Nov 03 17:16:43 beaglebone kernel: free_area_init_node: node 0, pgdat c0fd6a84, node_mem_map dcb61000
Nov 03 17:16:43 beaglebone kernel:   Normal zone: 1152 pages used for memmap
Nov 03 17:16:43 beaglebone kernel:   Normal zone: 0 pages reserved
Nov 03 17:16:43 beaglebone kernel:   Normal zone: 131072 pages, LIFO batch:31
Nov 03 17:16:43 beaglebone kernel: CPU: All CPU(s) started in SVC mode.
Nov 03 17:16:43 beaglebone kernel: AM335X ES2.1 (sgx neon)
Nov 03 17:16:43 beaglebone kernel: pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Nov 03 17:16:43 beaglebone kernel: pcpu-alloc: [0] 0
Nov 03 17:16:43 beaglebone kernel: Built 1 zonelists, mobility grouping on.  Total pages: 129920
Nov 03 17:16:43 beaglebone kernel: Kernel command line: console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
Nov 03 17:16:43 beaglebone kernel: PID hash table entries: 2048 (order: 1, 8192 bytes)
Nov 03 17:16:43 beaglebone kernel: Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Nov 03 17:16:43 beaglebone kernel: Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Nov 03 17:16:43 beaglebone kernel: Memory: 449220K/524288K available (8192K kernel code, 967K rwdata, 3916K rodata, 1024K init, 684K bss, 25916K reserved, 49152K cma-reserved, 0K highmem)
Nov 03 17:16:43 beaglebone kernel: Virtual kernel memory layout:
                                       vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                       fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                       vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
                                       lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
                                       pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                                       modules : 0xbf800000 - 0xbfe00000   (   6 MB)
                                         .text : 0xc0008000 - 0xc0900000   (9184 kB)
                                         .init : 0xc0e00000 - 0xc0f00000   (1024 kB)
                                         .data : 0xc0f00000 - 0xc0ff1cac   ( 968 kB)
                                          .bss : 0xc0ffaf20 - 0xc10a6124   ( 685 kB)
Nov 03 17:16:43 beaglebone kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Nov 03 17:16:43 beaglebone kernel: ftrace: allocating 38864 entries in 76 pages
Nov 03 17:16:43 beaglebone kernel: Preemptible hierarchical RCU implementation.
Nov 03 17:16:43 beaglebone kernel:      No expedited grace period (rcu_normal_after_boot).
Nov 03 17:16:43 beaglebone kernel:      Tasks RCU enabled.
Nov 03 17:16:43 beaglebone kernel: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Nov 03 17:16:43 beaglebone kernel: IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
Nov 03 17:16:43 beaglebone kernel: OMAP clockevent source: timer2 at 24000000 Hz
Nov 03 17:16:43 beaglebone kernel: sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
Nov 03 17:16:43 beaglebone kernel: clocksource: timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
Nov 03 17:16:43 beaglebone kernel: OMAP clocksource: timer1 at 24000000 Hz
Nov 03 17:16:43 beaglebone kernel: timer_probe: no matching timers found
Nov 03 17:16:43 beaglebone kernel: Console: colour dummy device 80x30
Nov 03 17:16:43 beaglebone kernel: WARNING: Your 'console=ttyO0' has been replaced by 'ttyS0'
Nov 03 17:16:43 beaglebone kernel: This ensures that you still see kernel messages. Please
Nov 03 17:16:43 beaglebone kernel: update your kernel commandline.
Nov 03 17:16:43 beaglebone kernel: Calibrating delay loop... 995.32 BogoMIPS (lpj=1990656)
Nov 03 17:16:43 beaglebone kernel: pid_max: default: 32768 minimum: 301
Nov 03 17:16:43 beaglebone kernel: Security Framework initialized
Nov 03 17:16:43 beaglebone kernel: Yama: becoming mindful.
Nov 03 17:16:43 beaglebone kernel: AppArmor: AppArmor initialized
Nov 03 17:16:43 beaglebone kernel: Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Nov 03 17:16:43 beaglebone kernel: Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Nov 03 17:16:43 beaglebone kernel: CPU: Testing write buffer coherency: ok
Nov 03 17:16:43 beaglebone kernel: Setting up static identity map for 0x80100000 - 0x80100054
Nov 03 17:16:43 beaglebone kernel: Hierarchical SRCU implementation.
Nov 03 17:16:43 beaglebone kernel: EFI services will not be available.
[... cut ...]
Jan 03 22:45:40 beaglebone systemd[1695]: Reached target Basic System.
Jan 03 22:45:40 beaglebone systemd[1695]: Reached target Default.
Jan 03 22:45:40 beaglebone systemd[1695]: Startup finished in 261ms.
Jan 03 22:45:40 beaglebone systemd[1]: Started User Manager for UID 1000.
Jan 03 22:46:05 beaglebone sudo[1713]:   debian : TTY=pts/0 ; PWD=/home/debian ; USER=root ; COMMAND=/bin/journalctl -all
Jan 03 22:46:05 beaglebone sudo[1713]: pam_unix(sudo:session): session opened for user root by debian(uid=0)
debian@beaglebone:~$

RobertCNelson, 2018-01-04:
you should update the kernel:
```
sudo /opt/scripts/tools/update_kernel.sh
```
I see a few things i've moved to built-in..

These are wasting a few clock cycles for you on bootup..
```
sudo apt remove haveged bonescript c9-core-installer roboticscape bb-node-red-installer --purge
```

Regards,

patrickpoirier51, 2018-01-04:
OK sir will do... its getting a little late here toight , will report back tomorrow
Thanks for your work and support 👍

patrickpoirier51, 2018-01-04:
Did Update & removed stuff, but still taking 2 minutes

patrickpoirier51, 2018-01-04:
I removed board from system and
sudo systemctl disable dnsmasq 
sudo systemctl disable hostapd
sudo systemctl disable arducopter
sudo systemctl disable  apache2

Connected to ttyS0 and looked at boot sequence
Its taking 10 secs to get to this:
[    1.031050] omap_voltage_late_init: Voltage driver support not added

and then it wait (just the blue heartbeat) until exactly 60 seconds and it continues loading 
during 10 secs until login:

[    1.030727] omap_voltage_late_init: Voltage driver support not added
rootfs: clean, 82502/943488 files, 515309/3817472 block
Debian GNU/Linux 9 beaglebone ttyS0
BeagleBoard.org Debian Image 2017-10-10
Support/FAQ: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian
default username:password is [debian:temppwd]
beaglebone login:

patrickpoirier51, 2018-01-18:
UPDATE:
I have been able to reduce to 1:06  minute using  this image: 
https://rcn-ee.net/rootfs/bb.org/testing/2018-01-07/stretch-console/

git:/opt/scripts/:[b372df3221bb67a9a09c6b89df5110ca32a9e482]
eeprom:[A335PBGL00A21736GPB40029]
model:[TI_AM335x_PocketBeagle]
dogtag:[BeagleBoard.org Debian Image 2018-01-07]
bootloader:[microSD]:[/dev/mmcblk0]:[U-Boot 2018.01-rc3-00002-gb33224]
kernel:[4.14.13-bone-rt-r12]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180114.0-0rcnee1~stretch+20180115]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
dmesg | grep pinctrl-single
[    0.862970] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
END

**It is waiting 30 seconds for networks  to come up** 
Jan 18 22:57:11 beaglebone kernel: ieee80211 phy0: Atheros AR9271 Rev:1
Jan 18 22:57:41 beaglebone systemd[1]: Listening on Load/Save RF Kill Switch Status /dev/rfkill Watch.

Question: Is ther a way to reduce this network waiting ?

RobertCNelson, 2018-01-21:
I wonder if we start building in the wireless stack..

I'll run some tests with that idea. ;)

Regards,

RobertCNelson, 2018-01-23:
Okay some numbers (on v4.14.14-ti-r26)
```
-CONFIG_CFG80211=m
+CONFIG_CFG80211=y
-CONFIG_MAC80211=m
+CONFIG_MAC80211=y
```
Gives us about a 1.5 second speed up..

Then with: (random usb wifi adapter)

```
-CONFIG_MT7601U=m
+CONFIG_MT7601U=y
```
Gives me about 4 second speed up..

Ps, in the stretch-console, install "connman" then comment out eth0 in /etc/network/intefaces

```
# The primary network interface
#auto eth0
#iface eth0 inet dhcp
```
Regards,

RobertCNelson, 2018-01-23:
Here's the config change, i'll push out a kernel build shortly to make it easy for you to tset:

https://github.com/RobertCNelson/ti-linux-kernel-dev/commit/9ec77983c27c79604fe798f3945be9713614dd36

Regards,

patrickpoirier51, 2018-01-23:
OK I will give it a try, Thanks !

Please note I am testing  both with connman and hostapd with a TP-LINK TL-WN722N
https://github.com/PocketPilot/PocketPilot/blob/master/Pocket%20Instructions.txt

With the actual PocketPilot prototype, i need to add allow-hotplug in order to get the wlan started:
allow-hotplug wlan0
auto wlan0

RobertCNelson, 2018-01-23:
Looking at https://github.com/PocketPilot/PocketPilot/blob/master/Pocket%20Instructions.txt

I see you swapping connman for hostapd.

Connman actually has a built-in "tether" option, that's a little faster then hostapd. (although it doesn't support as much)..

```
connmanctl tether wifi on ${wifi_ssid} ${wifi_password}
```
and then you don't need dnmasq

Regards,

patrickpoirier51, 2018-01-23:
> master/Pocket%20Instructions.txt

Robert,
connman makes things so much easier,  the  problem with the pocket  is
twofold:
A) tether needs an ethernet port to attach
B) we operate a drone, so there is no physical ethernet connexion available

If you know how to ''trick'' conman to make it teher to a virtual
connection and assign a specific dhcp range I would be please to implement

Best Regards

RobertCNelson, 2018-01-23:
i wonder if something has changed in connman, i've used it before as a tether, when no eth0 was present (or populated)... i'll test that again shortly, need to solder a new usb-header on a pocketbeagle..

patrickpoirier51, 2018-01-23:
Well, actually it starts but there is no ipv4 on the wifi, so i deducted i
it could not associate a dhcp because of the non existing theter... I
stopped my investigation there.

RobertCNelson, 2018-01-23:
i think it's because i kill:

```
systemctl disable connman-wait-online.service || true
```
Regards,

RobertCNelson, 2018-01-23:
Otherwise, my connman main.conf has:

```
[General]
PreferredTechnologies=ethernet,wifi
SingleConnectedTechnology=false
AllowHostnameUpdates=false
PersistentTetheringMode=true
NetworkInterfaceBlacklist=SoftAp0,usb0
```
and settings has:

```
[global]
OfflineMode=false

[Wired]
Enable=true
Tethering=false

[WiFi]
Enable=true
Tethering=true
Tethering.Identifier=BeagleBone
Tethering.Passphrase=BeagleBone

[Gadget]
Enable=false
Tethering=false

[P2P]
Enable=false
Tethering=false

[Bluetooth]
Enable=true
Tethering=false
```

patrickpoirier51, 2018-01-23:
Interesting, will give it another try, that really makes a the project easier to implement :-)

btw, I damaged USB1 on the pocket while doing tests because I have not installed ESD protecton on my header. I know that Digi-Key has this product available: 
https://www.digikey.com/product-detail/en/wurth-electronics-inc/8492121/732-5676-1-ND/4947875

I wonder if you had a chance to test it or better, a microUSB size with embedded ESP ?

RobertCNelson, 2018-01-23:
Sure, i'll pull that one tomorrow and give it a test.  It's too late in the day here to get it from downstairs. ;)

Regards,

patrickpoirier51, 2018-01-23:
I diasbled  hostapd & dnsmasq , removeved wlan0 from /etc/network/interfaces and enabled connman
I used your config 7 main file and systemctl disable connman-wait-online.service || true

Same issue = no IPV4:

wlan0: flags=-28605<UP,BROADCAST,RUNNING,MULTICAST,DYNAMIC>  mtu 1500
        inet6 fe80::f6f2:6dff:fe15:e79b  prefixlen 64  scopeid 0x20<link>
        ether f4:f2:6d:15:e7:9b  txqueuelen 1000  (Ethernet)
        RX packets 32  bytes 6273 (6.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 55  bytes 9167 (8.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

patrickpoirier51, 2018-01-24:
Just read that I could probably set it manually:
https://stackoverflow.com/questions/31943490/how-to-set-static-ip-address-using-connman-for-wifi-network

 I managed to get a static IP address by manually creating a file named "settings" inside the directory:  /var/lib/connman/wifi_<HASH>_managed_psk/ with the following content:[wifi_<HASH>_managed_psk]  Name=<SSID>                       ←Name of the network  SSID=544f52414445585f4252         ←Name of the network in hexadecimal format     Favorite=true IPv4.method=manual                ←Method to be used (in this case manual IP) IPv4.netmask_prefixlen=24 IPv4.local_address=192.168.0.133  ←Desired IP address IPv4.gateway=192.168.0.1 Passphrase=<PASS>                 ←Wifi network password AutoConnect=true
-- | --

patrickpoirier51, 2018-01-25:
After more reading and test I think that we cannot use tether as a single interface because its bridged to the ethernet (physical or  rndis). Unless we can creata a virtual device wit a fixed address on wich the connman dhcp service can assign ipv4, I dont see how it could work. 
So for the moment I will leave my instructions untouched.

patrickpoirier51, 2018-01-26:
I am going to close this issue because the startup delay is acceptable now.
I would appreciate your comment on the ESD USB connector.
Hope you dont mind I name you in this blog ?
https://discuss.ardupilot.org/t/pocketpilot-a-new-linux-fc-based-on-the-25-pocketbeagle/25253

RobertCNelson, 2018-01-26:
i think the ESD connector will work.. just don't bug wire it, i was getting cable issues, but finally smoked the usb port..
Your post looks awesome, can't wait for the pcb's.. ;)

closed 2018-01-26 by RobertCNelson