RobertCNelson/omap-image-builder #116: Squashfs on Beagle Bone Black with u-boot 2016.11

mrudulachougule, 2018-07-04:
Hi,
We have beagle bone black with ubuntu 16.04 filesystem up and running. Filesystem is built using omap-image-builder and now want to use squash file system.

Please share your views/ inputs on booting beagle bone black with squashfs.

RobertCNelson, 2018-07-04:
Sorry, no experience with squashfs.

Feel free to submit a pull request to support such feature..

Regards,

closed 2018-07-04 by RobertCNelson

mrudulachougule, 2018-07-05:
Sure. Will definitely submit ..
Currently i am planing to boot beagle bone with initrd.
Do you have initrd image for beagle bone black?

Please do share the image if you have or share the details to create the
initrd for beagle bone black.

Thanks,
Mrudula

mrudulachougule, 2018-07-05:
Hi Robert,

I can mount ext4 filesystem as a read-only.
Please share any other ways where in filesystem can be mounted as a
read-only.

Thanks,
Mrudula

On Thu, Jul 5, 2018 at 12:00 PM, Mrudula Chougule <chougulemrudula@gmail.com

mrudulachougule, 2018-07-05:
Hi Robert,

Can I mount ext4 filesystem as a read-only?
Please share any other ways where in filesystem can be mounted as a
read-only.

Thanks,
Mrudula

RobertCNelson, 2018-07-05:
@mrudulachougule that's been the default for a long time.

Patches welcome

Regards,