RobertCNelson/omap-image-builder #117: wget: missing URL for osd335x-blank-eeprom

JacksonBanbury, 2018-07-15:
I am trying to create an sd image for an osd335x-based board with a blank eeprom, but running the setup_sdcard.sh fails as follows:

```
Debug: ARM rootfs: armhf-rootfs-debian-stretch.tar

Downloading Device's Bootloader
-----------------------------
2018-07-15 09:25:00 URL:https://rcn-ee.com/repos/bootloader/latest/bootloader-ng [10391/10391] -> "/tmp/tmp.JPdwCxGK39/dl/bootloader-ng" [1]
wget: missing URL
Usage: wget [OPTION]... [URL]...

Try `wget --help' for more options.
```

Looking at https://rcn-ee.com/repos/bootloader/latest/bootloader-ng there does not seem to be a link for "am335x_blank_bbbw", the conf_board listed in the osd335x-blank-eeprom.conf file.

The script works as expected for other board configs, like plain "beaglebone"

RobertCNelson, 2018-07-15:
@JacksonBanbury yeah osd335x-blank-eeprom was a test *.conf at first it looked like we would need a custom memory init, but it turned out to be 100% compable with bbb in u-boot... So just use "bbb-blank-eeprom"..

Regards,

closed 2018-07-15 by RobertCNelson

JacksonBanbury, 2018-07-15:
Great, will do. Thanks for the quick update Robert.