RobertCNelson/omap-image-builder #123: Question about background operations

bdube, 2019-04-03:
https://github.com/RobertCNelson/omap-image-builder/blob/a032246f888850b628ddb96ca4754fc38e11150b/beagleboard.org_stretch_image.sh#L39
https://github.com/RobertCNelson/omap-image-builder/blob/a032246f888850b628ddb96ca4754fc38e11150b/beagleboard.org_stretch_image.sh#L66

The two linked lines above run the compression in the background. Can I ask about your execution context or workflow? When I use the script, the compression continues running after the parent script exits but still expects to output progress to the terminal. It's working for me, but that progress output from a background process is awkward. Curious if that's the intended workflow.

Thanks!

RobertCNelson, 2019-04-03:
@bdube , true we could drop the "-v" in those images..

Truefly i never run the base ./*.sh scripts:

```code
beagleboard.org_jessie_image.sh
beagleboard.org_stretch_image.sh
seeed_jessie_gcp-iot_image.sh
seeed_jessie_image.sh
```

They are only there for new users, who want to generate a base image..  For a long time they were way out of date.

I only run the ./publish/rcn_ee*.sh images

and i've merged most of rcn-ee_bb.org-stable.sh into those base *.sh

Regards,

bdube, 2019-04-03:
Ok, thanks for the clarification.

In our fork, I think I'm going to leave the `-v` and let it run in the foreground. That way I don't need to do any chasing to find out when it's complete.

closed 2019-04-03 by bdube