RobertCNelson/omap-image-builder #130: BeagleBoneBlack turn into a brick after `apt upgrade` on freshly installed image

szobov, 2020-05-25:
Hi there!
I have working BeagleBone Black with installed `Ubuntu 18.04`. After I've executed `apt update && apt upgrade` and rebooted, but my BeagleBone didn't wake up. It didn't start to answer on serial port, even on my laptop I didn't see anything in `dmesg` when I connect it.

I've decided to unfold new fresh [version](https://rcn-ee.com/rootfs/2020-03-12/flasher/bone-eMMC-flasher-ubuntu-18.04.4-console-armhf-2020-03-12-2gb.img.xz), download tar, dd'ed it and flashed.
OS has been installed and I've set up the wifi and tried `apt update && apt upgrade` again. The beaglebone again turn into the brick.

Here is a log of last commands I executed on BB:
```
ubuntu@beaglebone:~$ sudo apt update
Get:1 http://ports.ubuntu.com bionic InRelease [242 kB]
Get:2 http://ports.ubuntu.com bionic-updates InRelease [88.7 kB]
Get:3 http://repos.rcn-ee.com/ubuntu bionic InRelease [3060 B]            
Get:4 http://ports.ubuntu.com bionic/multiverse armhf Packages [129 kB]   
Get:5 http://ports.ubuntu.com bionic/universe armhf Packages [8269 kB]   
Get:6 http://repos.rcn-ee.com/ubuntu bionic/main armhf Packages [1009 kB]      
Get:7 http://ports.ubuntu.com bionic/main armhf Packages [968 kB]                                                                                                                                                                     
Get:8 http://ports.ubuntu.com bionic-updates/universe armhf Packages [898 kB]                                                                                                                                                         
Get:9 http://ports.ubuntu.com bionic-updates/multiverse armhf Packages [4496 B]                                                                                                                                                       
Get:10 http://ports.ubuntu.com bionic-updates/main armhf Packages [618 kB]                                                                                                                                                            
Fetched 12.2 MB in 24s (503 kB/s)                                                                                                                                                                                                     
Reading package lists... 91%
Reading package lists... 93%

Reading package lists... Done
Building dependency tree       
Reading state information... Done
97 packages can be upgraded. Run 'apt list --upgradable' to see them.
ubuntu@beaglebone:~$ 
ubuntu@beaglebone:~$ 
ubuntu@beaglebone:~$ 
ubuntu@beaglebone:~$ 
ubuntu@beaglebone:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following NEW packages will be installed:
  libnetplan0
The following packages will be upgraded:
  apt apt-utils bb-bbai-firmware bb-cape-overlays bb-customizations bb-usb-gadgets bb-wl18xx-firmware bind9-host binutils binutils-arm-linux-gnueabihf binutils-common bluetooth bluez bsdutils device-tree-compiler distro-info-data
  dmsetup fdisk file gcc-8-base git git-man grub-common grub-efi-arm grub-efi-arm-bin grub2-common iproute2 ipumm-dra7xx-installer kmod libapt-inst2.0 libapt-pkg5.0 libasound2 libasound2-data libatomic1 libbind9-160 libbinutils
  libblkid1 libcc1-0 libdevmapper1.02.1 libdns-export1100 libdns1100 libfdisk1 libgcc1 libgd3 libglib2.0-0 libglib2.0-data libgomp1 libicu60 libisc-export169 libisc169 libisccc160 libisccfg160 libjson-c3 libkmod2 libldap-2.4-2
  libldap-common liblwres160 libmagic-mgc libmagic1 libmount1 libnss-systemd libpam-systemd libpython2.7 libpython2.7-dev libpython2.7-minimal libpython2.7-stdlib libpython3.6 libpython3.6-dev libpython3.6-minimal
  libpython3.6-stdlib libsmartcols1 libstdc++6 libsystemd0 libudev1 libuuid1 linux-base linux-firmware linux-libc-dev mount netplan.io nplan python2.7 python2.7-dev python2.7-minimal python3.6 python3.6-dev python3.6-minimal
  rfkill systemd systemd-sysv tzdata udev util-linux vim-common vim-tiny vpdma-dra7xx-installer xxd
97 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 192 MB of archives.
After this operation, 358 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://ports.ubuntu.com bionic-updates/main armhf bsdutils armhf 1:2.31.1-0.4ubuntu3.6 [55.4 kB]
Get:2 http://ports.ubuntu.com bionic-updates/main armhf libuuid1 armhf 2.31.1-0.4ubuntu3.6 [19.4 kB]
Get:3 http://ports.ubuntu.com bionic-updates/main armhf libblkid1 armhf 2.31.1-0.4ubuntu3.6 [112 kB]
Get:4 http://ports.ubuntu.com bionic-updates/main armhf libfdisk1 armhf 2.31.1-0.4ubuntu3.6 [151 kB]
Get:5 http://ports.ubuntu.com bionic-updates/main armhf libmount1 armhf 2.31.1-0.4ubuntu3.6 [122 kB]
Get:6 http://ports.ubuntu.com bionic-updates/main armhf libsmartcols1 armhf 2.31.1-0.4ubuntu3.6 [74.6 kB]
Get:7 http://ports.ubuntu.com bionic-updates/main armhf fdisk armhf 2.31.1-0.4ubuntu3.6 [97.4 kB]
Get:8 http://ports.ubuntu.com bionic-updates/main armhf util-linux armhf 2.31.1-0.4ubuntu3.6 [845 kB]
Get:9 http://repos.rcn-ee.com/ubuntu bionic/main armhf bb-usb-gadgets all 1.20200504.0-0~bionic+20200504 [4480 B]
Get:10 http://repos.rcn-ee.com/ubuntu bionic/main armhf bb-bbai-firmware all 1.20200404.1-0~bionic+20200424 [280 kB]
Get:11 http://ports.ubuntu.com bionic-updates/main armhf libgomp1 armhf 8.4.0-1ubuntu1~18.04 [66.6 kB]
Get:12 http://ports.ubuntu.com bionic-updates/main armhf gcc-8-base armhf 8.4.0-1ubuntu1~18.04 [18.8 kB]
Get:13 http://ports.ubuntu.com bionic-updates/main armhf libgcc1 armhf 1:8.4.0-1ubuntu1~18.04 [37.1 kB]
Get:14 http://ports.ubuntu.com bionic-updates/main armhf libcc1-0 armhf 8.4.0-1ubuntu1~18.04 [32.6 kB]
Get:15 http://ports.ubuntu.com bionic-updates/main armhf libatomic1 armhf 8.4.0-1ubuntu1~18.04 [7060 B]
Get:16 http://ports.ubuntu.com bionic-updates/main armhf libstdc++6 armhf 8.4.0-1ubuntu1~18.04 [350 kB]
Get:17 http://ports.ubuntu.com bionic-updates/main armhf libnss-systemd armhf 237-3ubuntu10.41 [92.9 kB]
Get:18 http://ports.ubuntu.com bionic-updates/main armhf bluez armhf 5.48-0ubuntu3.4 [789 kB]
Get:19 http://repos.rcn-ee.com/ubuntu bionic/main armhf device-tree-compiler armhf 1.4.7-3ubuntu2rcnee0~bionic+20200430 [240 kB]
Get:20 http://ports.ubuntu.com bionic-updates/main armhf libsystemd0 armhf 237-3ubuntu10.41 [189 kB]
Get:21 http://ports.ubuntu.com bionic-updates/main armhf libpam-systemd armhf 237-3ubuntu10.41 [95.1 kB]
[... 442 lines cut ...]
done
Setting up libisccfg160:armhf (1:9.11.3+dfsg-1ubuntu1.12) ...
Setting up python3.6-dev (3.6.9-1~18.04ubuntu1) ...
Setting up libbind9-160:armhf (1:9.11.3+dfsg-1ubuntu1.12) ...
Setting up bind9-host (1:9.11.3+dfsg-1ubuntu1.12) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Processing triggers for systemd (237-3ubuntu10.41) ...
Processing triggers for dbus (1.12.2-1ubuntu1.1) ...
Processing triggers for mime-support (3.60ubuntu1) ...
Processing triggers for initramfs-tools (0.130ubuntu3.9) ...
update-initramfs: Generating /boot/initrd.img-5.4.24-armv7-x20
WARNING: missing /lib/modules/5.4.24-armv7-x20
Ensure all necessary drivers are built into the linux image!
depmod: ERROR: could not open directory /lib/modules/5.4.24-armv7-x20: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
depmod: WARNING: could not open /var/tmp/mkinitramfs_oXlErk/lib/modules/5.4.24-armv7-x20/modules.order: No such file or directory
depmod: WARNING: could not open /var/tmp/mkinitramfs_oXlErk/lib/modules/5.4.24-armv7-x20/modules.builtin: No such file or directory
ubuntu@beaglebone:~$ sudo reboot 
[sudo] password for ubuntu: 
Connection to 192.168.0.107 closed by remote host.
Connection to 192.168.0.107 closed.
```

I hope someone could explain, what I did wrong.
Thanks!

RobertCNelson, 2020-05-25:
```
Preparing to unpack .../kmod_24-1ubuntu3.4_armhf.deb ...
Unpacking kmod (24-1ubuntu3.4) over (24-1ubuntu3.2rcnee0~bionic+20190208) ...
Preparing to unpack .../libkmod2_24-1ubuntu3.4_armhf.deb ...
Unpacking libkmod2:armhf (24-1ubuntu3.4) over (24-1ubuntu3.2rcnee0~bionic+20190208) ...
Setting up libkmod2:armhf (24-1ubuntu3.4) ...
```

This broke the kernel modules..

Building right now:

https://github.com/rcn-ee/repos/commit/31b7df6c7b33e4df638f2f6e2748f47902b4f287

Regards,

RobertCNelson, 2020-05-25:
Update pushed to apt repo, background...

To save space, our kernel config compress all kernel modules as *.ko.xz kmod is perfectly happy loading compressed modules as long as it's "built" with the library...

https://github.com/rcn-ee/repos/blob/master/ubuntu-1804-kmod/suite/bionic/debian/rules#L21

At this point, ubuntu 18.04's kmod is the "last" hold out to not have this enabled by default. So every time ubuntu 18.04 pushed out a pointless kmod change i need to rebuild it..  My plan was to transition the ubuntu page to 20.04, i'm pretty close to that now.  I just got really busy with day job over this last month, so i haven't had free time to finish the 20.04 push out..

One quick fix to get everything working again. (Login over serial, as network is probally broken..)

```
cd /lib/modules/`uname -r`/kernel/
sudo unxz `find . | grep xz`
```
Then reboot and ubuntu's default kmod should be happy..

Regards,

RobertCNelson, 2020-05-25:
For reference:

```
voodoo@x1-am57xx-beagle-x15-2gb:/lib/modules$ du -sh 4.19.73-ti-r29/
68M	4.19.73-ti-r29/
voodoo@x1-am57xx-beagle-x15-2gb:/lib/modules$ du -sh 4.19.79-ti-r30/
26M	4.19.79-ti-r30/
```
4.19.73-ti-r29 = uncompressed
4.19.79-ti-r30 = xz compressed

szobov, 2020-05-25:
You're the best, man, and you're doing amazing work.
But I still don't understand why my already working OS just broke, but, yes, OK.
Thanks for this quick reply.

RobertCNelson, 2020-05-25:
On bootup, the kernel (thru kmod) will try to load up everything it needs.. Most of which we have selected as built-in, but there are a few sub-systems that can't handle that very well. For example, libcomposite and all the usb gadet modules, which we use to create the usb-network/usb-serial and usb-flash default interfaces.  So in ubuntu 18.04 with kmod not built to handle compressed kernel *.ko.xz files it fails..  So it looks like everything broke.

By pushing the kmod*.deb update to our repo, like i just did about 15minutes ago, if you were to start over where you were this morning, apt update/upgrade would pick our newer kmod*.deb and you'd have no issues..

Regards,

szobov, 2020-05-25:
Yes, now I see.
Thank you a lot again.

Btw, may be an option of have image, that include everything to work. Because personally me, and I suppose also some people could accept that trade-off. For me larger image is better than broken system.

szobov, 2020-05-26:
Hello again @RobertCNelson 
I've just roll out freshly flashed image and tried to update packages again.
The error still presence.
```
Ubuntu 18.04.4 LTS

rcn-ee.net Ubuntu Console Image 2020-03-12

Support/FAQ: http://elinux.org/BeagleBoardUbuntu

default username:password is [ubuntu:temppwd]

ubuntu@192.168.0.107's password: 
Last login: Fri Mar 13 05:05:22 2020 from 192.168.7.1
To run a command as administrator (user "root"), use "sudo <command>".
See "man sudo_root" for details.

ubuntu@beaglebone:~$ sudo apt update
[sudo] password for ubuntu: 
Get:1 http://ports.ubuntu.com bionic InRelease [242 kB]
Get:2 http://ports.ubuntu.com bionic-updates InRelease [88.7 kB]
Get:3 http://repos.rcn-ee.com/ubuntu bionic InRelease [3060 B]            
Get:4 http://ports.ubuntu.com bionic/universe armhf Packages [8269 kB]    
Get:5 http://ports.ubuntu.com bionic/multiverse armhf Packages [129 kB]                             
Get:6 http://ports.ubuntu.com bionic/main armhf Packages [968 kB]                                                                                                                                                                     
Get:7 http://repos.rcn-ee.com/ubuntu bionic/main armhf Packages [1009 kB]                                                                                                                                                             
Get:8 http://ports.ubuntu.com bionic-updates/multiverse armhf Packages [4496 B]                                                                                                                                                       
Get:9 http://ports.ubuntu.com bionic-updates/main armhf Packages [618 kB]                                                                                                                                                             
Get:10 http://ports.ubuntu.com bionic-updates/universe armhf Packages [898 kB]                                                                                                                                                        
Fetched 12.2 MB in 20s (608 kB/s)                                                                                                                                                                                                     
Reading package lists... Done
Building dependency tree       
Reading state information... Done
97 packages can be upgraded. Run 'apt list --upgradable' to see them.
ubuntu@beaglebone:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following NEW packages will be installed:
  libnetplan0
The following packages will be upgraded:
  apt apt-utils bb-bbai-firmware bb-cape-overlays bb-customizations bb-usb-gadgets bb-wl18xx-firmware bind9-host binutils binutils-arm-linux-gnueabihf binutils-common bluetooth bluez bsdutils device-tree-compiler distro-info-data
  dmsetup fdisk file gcc-8-base git git-man grub-common grub-efi-arm grub-efi-arm-bin grub2-common iproute2 ipumm-dra7xx-installer kmod libapt-inst2.0 libapt-pkg5.0 libasound2 libasound2-data libatomic1 libbind9-160 libbinutils
  libblkid1 libcc1-0 libdevmapper1.02.1 libdns-export1100 libdns1100 libfdisk1 libgcc1 libgd3 libglib2.0-0 libglib2.0-data libgomp1 libicu60 libisc-export169 libisc169 libisccc160 libisccfg160 libjson-c3 libkmod2 libldap-2.4-2
  libldap-common liblwres160 libmagic-mgc libmagic1 libmount1 libnss-systemd libpam-systemd libpython2.7 libpython2.7-dev libpython2.7-minimal libpython2.7-stdlib libpython3.6 libpython3.6-dev libpython3.6-minimal
  libpython3.6-stdlib libsmartcols1 libstdc++6 libsystemd0 libudev1 libuuid1 linux-base linux-firmware linux-libc-dev mount netplan.io nplan python2.7 python2.7-dev python2.7-minimal python3.6 python3.6-dev python3.6-minimal
  rfkill systemd systemd-sysv tzdata udev util-linux vim-common vim-tiny vpdma-dra7xx-installer xxd
97 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 192 MB of archives.
After this operation, 362 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://ports.ubuntu.com bionic-updates/main armhf bsdutils armhf 1:2.31.1-0.4ubuntu3.6 [55.4 kB]
Get:2 http://ports.ubuntu.com bionic-updates/main armhf libuuid1 armhf 2.31.1-0.4ubuntu3.6 [19.4 kB]
Get:3 http://ports.ubuntu.com bionic-updates/main armhf libblkid1 armhf 2.31.1-0.4ubuntu3.6 [112 kB]
Get:4 http://ports.ubuntu.com bionic-updates/main armhf libfdisk1 armhf 2.31.1-0.4ubuntu3.6 [151 kB]
Get:5 http://repos.rcn-ee.com/ubuntu bionic/main armhf kmod armhf 24-1ubuntu3.4rcnee0~bionic+20200525 [80.5 kB]
Get:6 http://ports.ubuntu.com bionic-updates/main armhf libmount1 armhf 2.31.1-0.4ubuntu3.6 [122 kB]
Get:7 http://ports.ubuntu.com bionic-updates/main armhf libsmartcols1 armhf 2.31.1-0.4ubuntu3.6 [74.6 kB]
Get:8 http://ports.ubuntu.com bionic-updates/main armhf fdisk armhf 2.31.1-0.4ubuntu3.6 [97.4 kB]
Get:9 http://ports.ubuntu.com bionic-updates/main armhf util-linux armhf 2.31.1-0.4ubuntu3.6 [845 kB]
Get:10 http://ports.ubuntu.com bionic-updates/main armhf libgomp1 armhf 8.4.0-1ubuntu1~18.04 [66.6 kB]
Get:11 http://ports.ubuntu.com bionic-updates/main armhf gcc-8-base armhf 8.4.0-1ubuntu1~18.04 [18.8 kB]
Get:12 http://ports.ubuntu.com bionic-updates/main armhf libgcc1 armhf 1:8.4.0-1ubuntu1~18.04 [37.1 kB]
[... cut ...]
depmod: FATAL: could not search modules: No such file or directory
depmod: WARNING: could not open /var/tmp/mkinitramfs_lpn8JN/lib/modules/5.4.24-armv7-x20/modules.order: No such file or directory
depmod: WARNING: could not open /var/tmp/mkinitramfs_lpn8JN/lib/modules/5.4.24-armv7-x20/modules.builtin: No such file or directory
ubuntu@beaglebone:~$ 
```
But now BeagleBone can be restarted.
So, I'll close this issue, because the problem now isn't so harsh, but I feel that we need more options to prevent this issue to be open again.
Thanks.

closed 2020-05-26 by szobov

RobertCNelson, 2020-05-26:
```
Found linux image: /boot/vmlinuz-5.4.24-armv7-x20
Found initrd image: /boot/initrd.img-5.4.24-armv7-x20
Found linux image: /boot/vmlinuz-4.19.94-ti-r36
Found initrd image: /boot/initrd.img-4.19.94-ti-r36
```

5.4.24-armv7-x20 => BeagleBoard/BeagleBoard-xM
4.19.94-ti-r36 => BeagleBone (all am335x varients) and BeagleBoard-x15/BeagleBone-AI