RobertCNelson/omap-image-builder #131: Debian/Beaglebone clock changing to wrong date (conflict between coonmand and systemd-timesyncd))

Goostavo, 2020-05-27:
# The Error

Tested on multiple beaglebones green and beaglebone green wireless with Debian 9.9 IoT and Debian 10 IoT clean on Brazil (UTC -3).

All of them are having the same error. Something reseting the systemtime. 

The correct time in my TZ is `sex fev 21 12:55:52 -03 2020` and in UTC `Fri Feb 21 15:56:06 UTC 2020`. But it's showing UTC as 12:56.

Running multiple `timedatectl` returns:
```
debian@beaglebone:~$ timedatectl 
      Local time: Fri 2020-02-21 12:53:54 UTC
  Universal time: Fri 2020-02-21 12:53:54 UTC
        RTC time: Fri 2020-02-21 12:53:53
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
debian@beaglebone:~$ timedatectl 
      Local time: Fri 2020-02-21 15:54:02 UTC
  Universal time: Fri 2020-02-21 15:54:02 UTC
        RTC time: Fri 2020-02-21 12:53:55
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: no
 RTC in local TZ: no
debian@beaglebone:~$ timedatectl 
      Local time: Fri 2020-02-21 12:53:56 UTC
  Universal time: Fri 2020-02-21 12:53:56 UTC
        RTC time: Fri 2020-02-21 12:53:55
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
```

hwclock --debug output:

```
hwclock from util-linux 2.29.2
Using the /dev interface to the clock.
Last drift adjustment done at 1582299529 seconds after 1969
Last calibration done at 1582299529 seconds after 1969
Hardware clock is on UTC time
Assuming hardware clock is kept in UTC time.
Waiting for clock tick...
...got clock tick
Time read from Hardware Clock: 2020/02/21 12:56:50
Hw clock time : 2020/02/21 12:56:50 = 1582289810 seconds since 1969
Time since last adjustment is -9719 seconds
Calculated Hardware Clock drift is 0.000000 seconds
2020-02-21 12:56:49.029564+0000
```
Multiple hwclock calls:

```
debian@beaglebone:~$ sudo hwclock 
2020-02-21 13:00:52.113529+0000
debian@beaglebone:~$ sudo hwclock 
2020-02-21 10:00:46.983797+0000
debian@beaglebone:~$ sudo hwclock 
2020-02-21 10:00:47.934507+0000
debian@beaglebone:~$ sudo hwclock 
2020-02-21 13:01:00.852430+0000
debian@beaglebone:~$ sudo hwclock 
2020-02-21 13:01:01.721401+0000
```
Timezone file:

```
debian@beaglebone:~$ cat /etc/timezone 
Etc/UTC
```

I've also tryied changing hwclock with --utc and --localtime.

---

More tests. I've disable ntp: ` sudo timedatectl set-ntp 0`, after i've set clock with `sudo timedatectl set-time '2019-10-31 16:24'`, And then i've renabled ntp sync (timesyncd).

The is the results. The hwclock was kept on the wrong time. Durint the set-ntp 0 the clock was kept on 3 hours less.

```
debian@beaglebone:~/DebianDogePack/dogepack_installer/etc/guidedoge$ timedatectl 
      Local time: Fri 2020-02-21 17:27:14 UTC
  Universal time: Fri 2020-02-21 17:27:14 UTC
        RTC time: Thu 2019-10-31 16:25:39
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: no
 RTC in local TZ: no
debian@beaglebone:~/DebianDogePack/dogepack_installer/etc/guidedoge$ timedatectl 
      Local time: Fri 2020-02-21 14:27:08 UTC
  Universal time: Fri 2020-02-21 14:27:08 UTC
        RTC time: Thu 2019-10-31 16:25:39
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
debian@beaglebone:~/DebianDogePack/dogepack_installer/etc/guidedoge$ timedatectl 
      Local time: Fri 2020-02-21 17:27:15 UTC
  Universal time: Fri 2020-02-21 17:27:15 UTC
        RTC time: Thu 2019-10-31 16:25:40
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: no
 RTC in local TZ: no
debian@beaglebone:~/DebianDogePack/dogepack_installer/etc/guidedoge$ timedatectl 
      Local time: Fri 2020-02-21 14:27:10 UTC
  Universal time: Fri 2020-02-21 14:27:10 UTC
        RTC time: Thu 2019-10-31 16:25:41
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
debian@beaglebone:~/DebianDogePack/dogepack_installer/etc/guidedoge$ timedatectl 
      Local time: Fri 2020-02-21 14:27:11 UTC
  Universal time: Fri 2020-02-21 14:27:11 UTC
        RTC time: Thu 2019-10-31 16:25:42
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no
debian@beaglebone:~/DebianDogePack/dogepack_installer/etc/guidedoge$ timedatectl 
      Local time: Fri 2020-02-21 17:27:19 UTC
  Universal time: Fri 2020-02-21 17:27:19 UTC
        RTC time: Thu 2019-10-31 16:25:43
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: no
 RTC in local TZ: no
```

---

I've tryied changing to my timezone. The localtime hour error was even greated (3h error + 3h timezone).

# The solution

There is a conflict between coonmand and systemd-timesyncd.

Debian for Beaglebone uses `connman`to control wifi connection. When connecting to a wifi it uses `agent on` enabling `connmand`, that also have a NTP server.

`connmand` try to set timezone from the network, and try to set time using it's internal NTP server.

The solution is disabling `connmand` NTP leaving it just for `systemd-timesyncd`. To do that edit `connman` config file:

```bash
sudo nano  /var/lib/connman/settings
```

Edit global config to be:
```
[global]
OfflineMode=false
TimeUpdates=manual
TimezoneUpdates=manual
```

RobertCNelson, 2020-05-27:
Thanks @Goostavo sounds very sane, let me configure this by default. ;)

RobertCNelson, 2020-05-27:
OKay looks good:

```
Log: (chroot): /var/lib/connman/settings
[global]
OfflineMode=false

[Wired]
Enable=true
Tethering=false

[WiFi]
Enable=true
Tethering=false

[Gadget]
Enable=false
Tethering=false

[P2P]
Enable=false
Tethering=false

[Bluetooth]
Enable=true
Tethering=false
Log: (chroot): Patched: /var/lib/connman/settings
[global]
OfflineMode=false
TimeUpdates=manual
TimezoneUpdates=manual

[Wired]
Enable=true
Tethering=false

[WiFi]
Enable=true
Tethering=false

[Gadget]
Enable=false
Tethering=false

[P2P]
Enable=false
Tethering=false

[Bluetooth]
Enable=true
Tethering=false
Log: (chroot): set_locale
```

https://github.com/RobertCNelson/omap-image-builder/commit/f0cabb066ad626527f2a0b9071598ae3724f65b2
and
https://github.com/RobertCNelson/omap-image-builder/commit/2e3566e26646a90b3589ef04f8458a97de6122c8