RobertCNelson/omap-image-builder #134: seeed-modules package is not available for 5.x kernels

ferrarimarco, 2020-09-09:
Hi! When building an image with a 5.x kernel, I see this in the output:

```shell
E: Unable to locate package seeed-modules-5.4.52-ti-r17
E: Couldn't find any package by glob 'seeed-modules-5.4.52-ti-r17'
E: Couldn't find any package by regex 'seeed-modules-5.4.52-ti-r17'
```

After checking http://repos.rcn-ee.com/ubuntu/dists/focal/main/binary-armhf/Packages (and packages for other distros and releases on `repos.rcn-ee.com`), I see no `seeed-modules` for kernels newer than `4.19.98-bone45`.

Can this be safely ignored? Thanks!

ferrarimarco, 2020-09-09:
Example configuration:

```
##
release="20.04.1"
image_type="console"
##
##Debootstrap: https://wiki.debian.org/Debootstrap
##
##debootstrap --arch=${deb_arch} --include=${deb_include} --exclude=${deb_exclude} \
##--components=${deb_components} --foreign ${deb_codename} /tmp/tmp.dir/ http://${deb_mirror}
##
deb_distribution="ubuntu"
deb_codename="focal"
deb_arch="armhf"
#
deb_include=" \
    acpid \
    alsa-utils \
    avahi-utils \
    bash-completion \
    bc \
    bluetooth \
    bsdmainutils \
    build-essential \
    ca-certificates \
    can-utils \
    cloud-guest-utils \
    connman \
    cpufrequtils \
    crda \
    device-tree-compiler \
    dnsmasq \
    dosfstools \
    fake-hwclock \
    fbset \
    file \
    git \
    gnupg \
    haveged \
    hexedit \
    hostapd \
    htop \
    i2c-tools \
    initramfs-tools \
    iw \
    linux-base \
    linux-firmware \
    locales \
    lsb-release \
    make \
    nano-tiny \
    nethogs \
    net-tools \
    nginx \
    openssh-server \
    pastebinit \
    patch \
    pkg-config \
    ppp \
    pps-tools \
    read-edid \
    rfkill \
    rsync \
    sudo \
    systemd \
    systemd-sysv \
    tio \
    u-boot-tools \
    usb-modeswitch \
    usbutils \
    wget \
    wireless-regdb \
    wireless-tools \
    wpasupplicant \
"

#
deb_exclude=""
#
deb_components="main universe multiverse"
deb_mirror=""
##
##Some packages fail to install via debootstrap: deb_additional_pkgs="<comma|space>"
##
deb_additional_pkgs=" \
    apt-transport-https \
    btrfs-progs \
    curl \
    gnupg-agent \
    grub-efi-arm \
    ifupdown \
    libnss-systemd \
    libpam-systemd \
    python3 \
    python3-dev \
    python3-setuptools \
    python3-smbus \
    software-properties-common \
"

##
rfs_username="ubuntu"
rfs_fullname="Demo User"
rfs_password="temppwd"
rfs_hostname="arm"
rfs_root_password="root"
rfs_startup_scripts="enable"
rfs_opt_scripts="https://github.com/RobertCNelson/boot-scripts"
rfs_default_desktop=""
rfs_desktop_background=""
rfs_default_locale="en_US.UTF-8"
rfs_etc_dogtag="ferrari.how Ubuntu 20.04.1 Console Image"
rfs_console_banner="Support: ferrari.marco@gmail.com"
rfs_console_user_pass="enable"
rfs_ssh_banner="Support: ferrari.marco@gmail.com"
rfs_ssh_user_pass="enable"
##
##enable kernel repo: http://repos.rcn-ee.com/(debian|ubuntu)
repo_rcnee="enable"
repo_rcnee_pkg_list=" \
    bb-bbai-firmware \
    bb-cape-overlays \
    bb-customizations \
    bb-usb-gadgets \
    bb-wl18xx-firmware \
    gpiod \
    ipumm-dra7xx-installer \
    linux-image-4.19.94-ti-r50 \
    linux-image-5.4.52-ti-r17 \
    moreutils \
    vpdma-dra7xx-installer \
    xauth \
"

repo_rcnee_depmod0="5.4.52-ti-r17"
repo_rcnee_depmod1="4.19.94-ti-r50"
repo_rcnee_pkg_version="5.4.52-ti-r17"
repo_rcnee_cmem_version="4.16.00.00"
include_firmware="enable"
#
chroot_COPY_SETUP_SDCARD="enable"
chroot_before_hook=""
chroot_after_hook=""
chroot_script="console_ubuntu_focal_armhf.sh"
chroot_post_uenv_txt=""
chroot_tarball="enable"
```

RobertCNelson, 2020-09-09:
@ferrarimarco correct, this is safe to ignore, the seeed-modules's have not been updated to v5.4.x yet.. (just v4.14.x/v4.19.x) kernel's..

I know this can be confusing, as it looks like any error, all those modules are not always needed..

https://github.com/RobertCNelson/omap-image-builder/blob/master/scripts/chroot.sh#L728-L732

Regards,

ferrarimarco, 2020-09-10:
Thanks @RobertCNelson ! This can be closed, or left open as a reminder :)