RobertCNelson/omap-image-builder #135: Beaglebone Ai

owlman2222, 2020-10-23:
Hi
unable to get --img-16gb working  for use with the AI board,
when using --dtb am57xx-beagle-x15

what am I doing wrong

thanks

RobertCNelson, 2020-10-23:
@owlman2222 , that's not a valid option:

https://github.com/RobertCNelson/omap-image-builder/blob/master/tools/setup_sdcard.sh#L1866-L1896

Why are you trying to create a 16gb img file?

The idea behind the "*-img" flag was to minimize the *.img file for network sharing, it was expected to resize the image on the final target later..

Regards,

owlman2222, 2020-10-23:
i was trying to create it the same size as the on board emmc so there was no need to resize, as completely forgot about being able resizing it.

closed 2020-10-23 by owlman2222

RobertCNelson, 2020-10-23:
@owlman2222 ps we just need to set this up to make it even easier ;)

https://github.com/beagleboard/Latest-Images/issues/30

Regards,