RobertCNelson/omap-image-builder #136: install_dependencies.sh will not function if the host system is not Debian-based

willeccles, 2021-03-15:
I use a Void Linux host system for development and have had to work around how this script works. It seems to me that it would be far simpler change how the function works. Rather than relying on `dpkg`'s output (which fails on Void and non-Debian systems), one could simply change the `debootstrap_is_installed()` function to check `if command -v debootstrap >/dev/null`. Alternatively, you could check for the presence of `dpkg` in the first place, then if it's not found fall back to checking the output of `command -v`. I would be happy to open a PR for this if necessary.

willeccles, 2021-03-15:
My current build script for this system checks if `dpkg` exists, and then if it doesn't, it simply removes everything from that script (my setup script installs the required dependencies anyway, so I don't really need this one to run in the first place). This isn't really relevant to the issue, I just thought it might be entertaining how aggressive my workaround for this is :)

RobertCNelson, 2021-03-15:
@willeccles sure no problem. ;)

closed 2021-03-15 by RobertCNelson