RobertCNelson/omap-image-builder #138: AM335x Debian Bullseye image boot is extremely slow compared to Stretch

willeccles, 2021-03-18:
(Apologies if this is not the proper place for this issue, but I am not really sure where it belongs.)

Previously, we were using the pre-built eewiki Debian Stretch image (I believe 9.8 or so). After modifying the eewiki Debian Buster build config to support Bullseye, I am getting the following boot times (abbreviated):

```
1min 10.947s ifupdown-pre.service                
     57.756s dev-mmcblk0p1.device                
     16.160s generic-board-startup.service       
      4.910s nginx.service                       
      4.534s user@1000.service                   
      3.955s dnsmasq.service                     
      2.672s systemd-udev-trigger.service        
.....
```

From what I can tell, it looks like `ifupdown-pre.service` is waiting for can0 to come online, which takes a very long time (not sure if related, but it is significantly longer than on Stretch). Also of note: haveged fails to start properly, as it apparently uses the uname(2) syscall, which is filtered by its systemd unit's seccomp filter. For the time being I have removed the filter, as in the future it's likely I will remove haveged unless there is a good reason to keep it around (I see no difference in the system with or without it). Also, this boot time does not depend on the SD card used. Whether I boot from MMC or eMMC it takes just the same amount of time.

Again, I'm not sure exactly where the right place for this issue would be, but unfortunately I need to update this system to Bullseye and am unable to make it boot in an acceptable amount of time. Any thoughts?

### Additional Details
The system is based on an OSD3558 on custom hardware. The kernel is 4.19.180-bone-rt (built with bb-kernel). We are using the same u-boot image and kernel image as before (when using Stretch), which is version 2019.04 plus various patches (our own + eewiki).

willeccles, 2021-03-19:
I'm going to try upgrading to kernel 5.10-bone-rt, as it seems upgrading the kernel has helped such situations in the past (https://github.com/RobertCNelson/linux-dev/issues/45). I know this is rather bleeding edge for such a system, so I was expecting some growing pains, but we'd really like to be able to ship this system with Bullseye on it, so I will keep trying.

willeccles, 2021-03-19:
I have just gotten 5.10.21-bone-rt-r28 built and flashed. It appears that one GPIO chip was not successfully added, so it's likely our device tree requires updates from kernel 4.19 to 5.10 (I already had to change one thing to get it to successfully build the dtb, so it's not out of the question). For the moment, our image-builder project configuration is almost entirely identical to the eewiki-minfs-buster one, except for one additional package.

Also of note, regarding the original post: haveged seems to work out of the box on kernel 5.10, so it's likely that the seccomp issue I encountered with it previously had to do with using an older kernel than it was built for.

willeccles, 2021-03-19:
Additionally, our boot is still extremely slow. It appears also that ifup@eth0 is failing: `RTNETLINK answers: file exists`. This might have something to do with connmand. It also appears to be waiting quite some time for serial-getty@ttyGS0, since /dev/ttyGS0 does not exist.

RobertCNelson, 2021-03-19:
yeah, libcomposit is slow on loading.. back in 4.19.x i had tried building it as built-in, to try and speed that up, but it was even more broken..

willeccles, 2021-03-19:
Alright, on 5.10 I am unable to get GPIOs 0-31 to register. The kernel repeatedly posts the same log message `PM: Cannot get wkup_m3_ipc handle`. This happens a few times, and then after the GPIOs have failed and many other devices have been brought up, it then says `wkup_m3_ipc 44e11324.wkup_m3_ipc: CM3 Firmware Version = 0x192`. I've included an image showing the relevant sections of the boot log. Any idea what could be causing this? I am not sure if these two things are related, but they seem quite coincidental if not.
[image]

willeccles, 2021-03-19:
Ah, never mind. That's `-EPROBE_DEFER`, and you can see it is deferring until later and successfully setting up. Very strange that our hardware isn't working then.

willeccles, 2021-03-30:
Looks like disabling the `repo_rcnee` option fixed this, so I am assuming the rules are from `bb-customizations`, though I did not check with apt to figure it out, so YMMV.

willeccles, 2021-04-01:
Okay, so to summarize my findings for the future if people are looking to use Bullseye:

1. At the moment, kernel 5.10 is not supported by TI, so if you have any issues, do not expect them to help. When asking about the GPIO chip issue, I received a response that effectively said that they don't support anything there is no TI SDK for and if I couldn't reproduce the issue on 4.19 they would not help me. Upgrade with caution, this is unsupported territory for TI EVMs, at least.
2. The udev rules added by `bb-customizations` are abysmal for boot time and system performance as a whole, so it's likely that they are somehow incompatible with a newer udev. When I enabled debug logging in /etd/udev/udev.conf, it was overflowing my kernel log every *2 seconds* with how often it was calling chown, ghgrp, etc.
3. Bullseye with 5.10 is functional on an OSD3358 in general. Note that some things in configurations will have to change: between 4.19 and 5.10, the default device trees for the AM33xx and relevant boards were completely overhauled, so if you include them you will likely have to adapt your device tree a tad. I believe the way the GPIO is defined in `arch/arm/boot/dts/am33xx-l4.dtsi` is responsible for making the GPIO come online incorrectly and be improperly ordered, though I have not tried modifying this file to fix it, so that may be wrong.

Otherwise, I have had no issues and have simply had the usual growing pains that come with upgrading Linux/Debian versions. The build scripts still work well, though there is one issue, which I will open a separate issue about.

closed 2021-04-01 by willeccles