RobertCNelson/omap-image-builder #140: chroot.sh uses the wrong initial timestamp file for Buster

jtetrault, 2021-04-21:
Hi there, 

I am using the `image-builder` repository to build a custom Debian Buster flasher image. I noticed that upon first boot of the system, that the start time of services was much further in the past than expected (2 years 2 months), and that I was getting some odd behaviour when trying to view service-specific logs with `journalctl`

```
~$ sudo systemctl status syslog
● rsyslog.service - System Logging Service
   Loaded: loaded (/lib/systemd/system/rsyslog.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-02-14 10:12:05 UTC; 2 years 2 months ago
     Docs: man:rsyslogd(8)
           https://www.rsyslog.com/doc/
 Main PID: 1338 (rsyslogd)
    Tasks: 4 (limit: 2242)
   Memory: 3.2M
   CGroup: /system.slice/rsyslog.service
           └─1338 /usr/sbin/rsyslogd -n -iNONE
```

I looked at `scripts/chroot.sh` and saw the `touch /var/lib/systemd/clock` command around line 1117 that is intended to set the initial timestamp that the system uses. It appears that at some point, `systemd-timesyncd` switched to using `/var/lib/systemd/timesync/clock` instead.

I don't know if the correct file is different between the versions available to Stretch and Buster.

I manually mounted the sd card, ran `touch -m /var/lib/systemd/timesync/clock`, changed the ownership to `systemd-timesync`, and the expected intial timestamp was correct after flashing the BeagleBone Black again and booting it up.

RobertCNelson, 2021-04-21:
awesome, thanks @jtetrault looks like:

https://github.com/systemd/systemd/commit/53d133ea1bb4c4ed44c4b6aae42f1feb33d9cb78

So with: v235

stretch: v232
buster: v241
bullseye: v247.3

RobertCNelson, 2021-04-21:
Thanks @jtetrault , i pushed out a fix, once we drop stretch, i'll remove that location, so for now, it's touching both locations.

closed 2021-04-21 by RobertCNelson

jtetrault, 2021-04-21:
Awesome, thanks! No harm in just touching both.

jtetrault, 2021-04-21:
Hey, it appears that the `/var/lib/systemd/timesync` directory doesn't exist yet, so the new `touch` cmd is failing. I think a `mkdir -p /var/lib/systemd/timesync` just needs to be inserted before the `touch`.

Please let me know if you need a new issue created!

RobertCNelson, 2021-04-21:
@jtetrault and merged. ;) after my apm mustang went down, i haven't really had ci on this repo lately, i need to fix. that..