RobertCNelson/omap-image-builder #141: systemd-journald race condition due to /etc/machine-id regeneration

jtetrault, 2021-04-22:
Hey, I think this is the appropriate repo for this issue (at least, it can be resolved in this repo).

There seems to be a minor race condition on first boot, that causes `journalctl` to just spit out "No journal files were found" until the service / system is rebooted. I traced this down to `/etc/machine-id` being deleted and regenerated by the `generic-startup.sh` script in the `boot-scripts` repo.

Basically, this is what happens:
- For whatever reason, `/etc/machine-id` is present in the final generated flasher image
- After flashing, on first boot, `journald-systemd` grabs the machine-id found at `/etc/machine-id`, and starts logging to `/var/log/journal/<machine-id>`
- `generic-startup.sh` runs, sees the `/etc/ssh/ssh.regenerate` file exists, and removes and regenerates the machine-id
- `journald-systemd` is now logging to the "wrong" spot for the new machine-id, so `journalctl` is unable to find the log files

I deleted `/etc/machine-id` from a flasher sd card, used it to flash a BBE, and did not see the above problem on the first boot. So I _think_ all that's required is to add a `rm -f /etc/machine-id` somewhere in `chroot.sh`

RobertCNelson, 2021-04-22:
Thanks @jtetrault i just added to the same location as teh ssh trigger file..

RobertCNelson, 2021-04-22:
also fixed the flashing script to do the sam.e..

closed 2021-04-22 by RobertCNelson