RobertCNelson/omap-image-builder #150: On the bb.org-debian-bullseye-console-v5.4 configuration, USB1 seems missing.

drmpeg, 2021-08-24:
/var/log/syslog gets spammed continuously. There is no /sys/class/net/usb1 file. This is on a Beagleboard-X15.

```
Aug 24 08:10:00 BeagleBone sh[5581]: autoconfigure_usb1.sh: g_multi: waiting for /sys/class/net/usb1/
Aug 24 08:10:01 BeagleBone sh[5581]: autoconfigure_usb1.sh: g_multi: waiting for /sys/class/net/usb1/
Aug 24 08:10:02 BeagleBone sh[5581]: autoconfigure_usb1.sh: g_multi: waiting for /sys/class/net/usb1/
Aug 24 08:10:03 BeagleBone sh[5581]: autoconfigure_usb1.sh: g_multi: waiting for /sys/class/net/usb1/
Aug 24 08:10:04 BeagleBone sh[5581]: autoconfigure_usb1.sh: g_multi: waiting for /sys/class/net/usb1/
Aug 24 08:10:05 BeagleBone sh[5581]: autoconfigure_usb1.sh: g_multi: waiting for /sys/class/net/usb1/
Aug 24 08:10:06 BeagleBone sh[5581]: autoconfigure_usb1.sh: g_multi: waiting for /sys/class/net/usb1/
Aug 24 08:10:07 BeagleBone sh[5581]: autoconfigure_usb1.sh: g_multi: waiting for /sys/class/net/usb1/
Aug 24 08:10:08 BeagleBone sh[5581]: autoconfigure_usb1.sh: g_multi: waiting for /sys/class/net/usb1/
Aug 24 08:10:09 BeagleBone sh[5581]: autoconfigure_usb1.sh: g_multi: waiting for /sys/class/net/usb1/
```

RobertCNelson, 2021-08-24:
Thanks @drmpeg , we aren't ready for usb0/usb1 yet, let's drop it: 

https://github.com/RobertCNelson/omap-image-builder/commit/3e248fd0196386e661f7ab5735821060f02408c3

RobertCNelson, 2021-08-24:
ps, i really want to use 5.10, i've added another config, which is a copy of 5.4.x:

bb.org-debian-bullseye-console-v5.10-ti-armhf.conf

drmpeg, 2021-08-24:
Thanks, the fix works good.

A minor issue for your consideration. On the 5.10 config, some files are missing.
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package rtl8723bu-modules-5.10.41-ti-r13
E: Couldn't find any package by glob 'rtl8723bu-modules-5.10.41-ti-r13'
E: Couldn't find any package by regex 'rtl8723bu-modules-5.10.41-ti-r13'
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package rtl8821cu-modules-5.10.41-ti-r13
E: Couldn't find any package by glob 'rtl8821cu-modules-5.10.41-ti-r13'
E: Couldn't find any package by regex 'rtl8821cu-modules-5.10.41-ti-r13'
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package seeed-modules-5.10.41-ti-r13
E: Couldn't find any package by glob 'seeed-modules-5.10.41-ti-r13'
E: Couldn't find any package by regex 'seeed-modules-5.10.41-ti-r13'
```

closed 2021-08-24 by drmpeg

RobertCNelson, 2021-08-24:
@drmpeg yeah those are safe...the two rtl* drivers aren't reallly maintained...