RobertCNelson/omap-image-builder #180: Dependency failed for Serial Getty on ttyS0 with Ubuntu Focal Build

MMetelko, 2021-09-28:
I have been trying to create a ubuntu focal console image and setting up the networking using the netplan approach with systemd-networkd.service.  It looks like the repo is being worked on for Debian Bullseye and Buster, but possibly not tested lately for focal.  I have seen that you are also restructuring some of the rcn-ee/repos for BBB specific setup.  I have looked at where repo changes were made and tried to add patches to be similar to the steps taken for the versions being worked (i.e. adding eth0.network).  I also have a patch that I add for our systems (included below).

 But, at this point, the serial port output stops at failure for Serial Getty on ttyS0 and at the moment I am not able to ssh into the BBB.   I did get an image version to boot (somewhere around 9/21) so that I could ssh into the BBB, but the serial port was still showing this issue.  

A slightly different question, but related ... I have also seen that the repos/bb-usb-gadgets requires the dnsmasq package, but that conflicts with systemd-resolved.service (sharing port 53).  So, I assume I need to disable systemd-resolve.service, but I wonder if the dnsmasq functionality is still needed when using systemd-networkd.service approach.  Since this is in a referenced repo, it seems harder to configure.

 I would appreciate  any suggestions on where to look for the serial port issue and any suggestions on possible conflicting packages for the networking setup.  Let me know if you need any other information.

Using repo commit (9/21): 2d57ac44b83e28ff377bdfebc557ea33c14f9826

Here is the serial output:
======================
U-Boot SPL 2019.04-00002-gbb4af0f50f (Jul 08 2019 - 11:44:39 -0500)
Trying to boot from MMC2
Loading Environment from EXT4... ** File not found /boot/uboot.env **

** Unable to read "/boot/uboot.env" from mmc0:1 **

U-Boot 2019.04-00002-gbb4af0f50f (Jul 08 2019 - 11:44:39 -0500), Build: jenkins-github_Bootloader-Builder-128

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Power-on reset has occurred.
RTC 32KCLK Source: External.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from EXT4... ** File not found /boot/uboot.env **

** Unable to read "/boot/uboot.env" from mmc0:1 **
Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   eth0: MII MODE
cpsw, usb_ether
Press SPACE to abort autoboot in 2 seconds
board_name=[A335BNLT] ...
board_rev=[00C0] ...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
starting USB...
USB0:   Port not available.
gpio: pin 54 (gpio 54) value is 1
Checking for: /boot/uEnv.txt ...
switch to partitions #0, OK
mmc0 is current device
gpio: pin 56 (gpio 56) value is 0
gpio: pin 55 (gpio 55) value is 0
gpio: pin 54 (gpio 54) value is 0
gpio: pin 53 (gpio 53) value is 1
switch to partitions #0, OK
mmc0 is current device
gpio: pin 54 (gpio 54) value is 1
Checking for: /uEnv.txt ...
Checking for: /boot.scr ...
Checking for: /boot/boot.scr ...
Checking for: /boot/uEnv.txt ...
gpio: pin 55 (gpio 55) value is 1
2138 bytes read in 39 ms (52.7 KiB/s)
[... 219 lines cut ...]
rfs_use_systemdnetworkd="enable"
##
##enable kernel repo: http://repos.rcn-ee.com/(debian|ubuntu)
##Beaglebone Black uses am335x, while Beagleboard-x15 and Beaglebone-AI use am57xx
##configuring for Beaglebone Black only at this time, so removing bb-u-boot-am57xx-evm
repo_rcnee="enable"
repo_rcnee_arch="armhf"
repo_rcnee_pkg_list="	\
	bb-bbai-firmware	\
	bb-cape-overlays	\
	bb-customizations	\
	bb-usb-gadgets	\
	bb-u-boot-am335x-evm	\
	bb-wl18xx-firmware	\
	gpiod	\
	ipumm-dra7xx-installer	\
	linux-image-5.4.106-ti-rt-r35	\
	moreutils
	vpdma-dra7xx-installer	\
	xauth	\
"

repo_rcnee_pkg_version="5.4.106-ti-rt-r35"
repo_rcnee_cmem_version="4.16.00.00"
include_firmware="enable"

RobertCNelson, 2021-09-28:
@MMetelko yeah sorry!  I'm pretty happy with how Bullseye is being configured now, i'll be working on backporting this to Buster and Ubuntu LTS..

correct, systemd-network removes our need for dnsmasq!!! 

the udev rules for wifi really cleanup the out of box tether stack..

We just need gadget support in systemd:

```
voodoo@x6-am57xx-beagle-x15-2gb:/lib/systemd$ find . | grep gadget
./system/usb-gadget.target
```

MMetelko, 2021-10-07:
@RobertCNelson 
Given your constant activity in updating this and the rcn-ee/repos/, I can tell you probably have a substantial list of priorities you are working through daily.  Is it possible to provide a possible time frame when you might get to a stable point where backporting makes sense?  I do understand that things happen and desired deadlines can even get delayed, so I am looking for a general timeframe so that I can set expectations on the project I am using this repo to create BBB images.  

It is far down the road, can you help me understand where to find the setup for the `usb-gadget.target`?  Doing the find does show that it is not setup in the builds I have created.

RobertCNelson, 2021-10-08:
Nah, wpa is still blocking... as long as you have a wifi adapter.. it's fast..

MMetelko, 2021-10-11:
I am still missing something.  Here is what I am running after ./RootStock-NG:

```
IMG=$(\ls deploy)
cd deploy/$IMG
sudo ./setup_sdcard.sh --img-4gb $IMG --dtb beaglebone --distro-bootloader
```

Here is what I am seeing:
```
Log: copying setup_sdcard.sh related files
Log: packaging rootfs: [armhf-rootfs-ubuntu-focal.tar]
-rw-r--r-- 1 root root 964M Oct  8 22:53 /var/lib/jenkins/workspace/ld_dev-5.4-rt-kernel-image-build/omap-image-builder/deploy/ubuntu-20.04.3-console-armhf-2021-10-08/armhf-rootfs-ubuntu-focal.tar
Log: USER:jenkins
syncing media... 
0+0 records in
0+0 records out
0 bytes copied, 0.000293332 s, 0.0 kB/s
-----------------------------
U-Boot Overlays Enabled...
Debug: ARM rootfs: armhf-rootfs-ubuntu-focal.tar

Using Distro Bootloader
-----------------------------
cp: cannot stat './u-boot/bb-u-boot-am335x-evm/MLO': No such file or directory
```

MMetelko, 2021-10-12:
Yes, I followed the pattern of rcn-ee.net-console-ubuntu-focal-v5.10-ti-armhf.conf since I want to use the netplan systemd approach to networking.  

Here are the packages I included in the `repo_rcnee_pkg_list`:
```
	bb-bbai-firmware	\
	bb-beagle-config	\
	bb-beagle-version	\
	bb-cape-overlays	\
	bb-usb-gadgets	\
	bb-u-boot-am335x-evm	\
	bb-wl18xx-firmware	\
	bb-wlan0-defaults	\
	generic-sys-mods	\
	gt	\
	libusbgx	\
	linux-image-5.4.106-ti-rt-r35	\
	overlayroot	\
```

I also included `u-boot-tools` in the `deb_include` list