RobertCNelson/omap-image-builder #198: Local bootloader not found during image creation

me21, 2022-06-27:
I'd like to supply my local U-Boot build for the Beaglebone Black. I'm running the script with the following parameters:

```
./setup_sdcard.sh --img-1gb my_image                 --dtb beaglebone --spl <my_SPL_path> --bootloader <my_bootloader_path> --rootfs_label rootfs --hostname beaglebone --enable-cape-universal --enable-uboot-pru-rproc-419ti
```

It fails on https://github.com/RobertCNelson/omap-image-builder/blob/74c3f405f730a4a2741e1eb4fc8338cd28eb52fa/tools/setup_sdcard.sh#L1512-L1513

with

```
Backup version of u-boot (MLO): /opt/backup/uboot/
cp: failed to execute stat for './u-boot/bb-u-boot-am335x-evm/MLO': No such file or directory.
```

Adding extra check so that the if statement looks like `if [ "${conf_bl_distro_SPL}" ] && [ ! "${USE_LOCAL_BOOT}" ] ; then` fixes the error.

A similar edit is applied to the subsequent if at https://github.com/RobertCNelson/omap-image-builder/blob/74c3f405f730a4a2741e1eb4fc8338cd28eb52fa/tools/setup_sdcard.sh#L1522.

Apparently the cause is the script doesn't properly check `--spl` and `--bootloader` options when creating backup version of u-boot, so that `conf_bl_distro_SPL` and `conf_bl_distro_UBOOT` from configuration files take precedence.

RobertCNelson, 2022-06-28:
idk....

```
voodoo@hestia:~/Downloads/debian-11.3-iot-armhf-2022-06-28$ sudo ./setup_sdcard.sh --img-8gb test --dtb beaglebone --spl ./u-boot/bb-u-boot-am335x-evm/MLO --bootloader ./u-boot/bb-u-boot-am335x-evm/u-boot-dtb.img 
syncing media... 
0+0 records in
0+0 records out
0 bytes copied, 4.5531e-05 s, 0.0 kB/s
-----------------------------
U-Boot Overlays Enabled...
Debug: ARM rootfs: armhf-rootfs-debian-bullseye.tar

Using Locally Stored Device Bootloader
-----------------------------
SPL Bootloader: MLO
UBOOT Bootloader: u-boot-dtb.img

Using dd to place bootloader on drive
-----------------------------
MLO: [dd if=../dl/MLO of=/home/voodoo/Downloads/debian-11.3-iot-armhf-2022-06-28/test-8gb.img count=2 seek=1 conv=notrunc bs=128k]
-----------------------------
0+1 records in
0+1 records out
106028 bytes (106 kB, 104 KiB) copied, 0.000133203 s, 796 MB/s
-----------------------------
u-boot.img: [dd if=..dl/u-boot-dtb.img of=/home/voodoo/Downloads/debian-11.3-iot-armhf-2022-06-28/test-8gb.img count=4 seek=1 conv=notrunc bs=384k]
-----------------------------
2+1 records in
2+1 records out
987236 bytes (987 kB, 964 KiB) copied, 0.000483909 s, 2.0 GB/s
-----------------------------
log: sfdisk: 2.26.x or greater detected
sfdisk: [sfdisk from util-linux 2.36.1]
sfdisk: [--force  /home/voodoo/Downloads/debian-11.3-iot-armhf-2022-06-28/test-8gb.img]
sfdisk: [4M,,L,*]
Checking that no-one is using this disk right now ... OK

Disk /home/voodoo/Downloads/debian-11.3-iot-armhf-2022-06-28/test-8gb.img: 7.03 GiB, 7549747200 bytes, 14745600 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

>>> Created a new DOS disklabel with disk identifier 0xede59f83.
/home/voodoo/Downloads/debian-11.3-iot-armhf-2022-06-28/test-8gb.img1: Created a new partition 1 of type 'Linux' and of size 7 GiB.
/home/voodoo/Downloads/debian-11.3-iot-armhf-2022-06-28/test-8gb.img2: Done.

New situation:
Disklabel type: dos
Disk identifier: 0xede59f83

Device                                                                Boot Start      End  Sectors Size Id Type
/home/voodoo/Downloads/debian-11.3-iot-armhf-2022-06-28/test-8gb.img1 *     8192 14745599 14737408   7G 83 Linux

The partition table has been altered.
Syncing disks.
Partition Setup:
-----------------------------
Disk /home/voodoo/Downloads/debian-11.3-iot-armhf-2022-06-28/test-8gb.img: 7.03 GiB, 7549747200 bytes, 14745600 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xede59f83

Device                                                                Boot Start      End  Sectors Size Id Type
/home/voodoo/Downloads/debian-11.3-iot-armhf-2022-06-28/test-8gb.img1 *     8192 14745599 14737408   7G 83 Linux
-----------------------------
add map loop1p1 (254:1): 0 14737408 linear 7:1 8192
Formating with: [mkfs.ext4  /dev/mapper/loop1p1 -L rootfs]
-----------------------------
[... 347 lines cut ...]
│           ├── BB-W1-P9.12-00A0.dtbo
│           ├── BONE-ADC.dtbo
│           ├── M-BB-BBG-00A0.dtbo
│           ├── M-BB-BBGG-00A0.dtbo
│           ├── PB-HACKADAY-2021.dtbo
│           ├── PB-MIKROBUS-0.dtbo
│           └── PB-MIKROBUS-1.dtbo
├── initrd.img-5.10.109-ti-r45
├── SOC.sh
├── System.map-5.10.109-ti-r45
├── uboot
├── uEnv.txt
└── vmlinuz-5.10.109-ti-r45

4 directories, 99 files
Finished populating rootfs Partition
-----------------------------
setup_sdcard.sh script complete
-----------------------------
The default user:password for this image:
debian:temppwd
-----------------------------
Image file: test-8gb.img
-----------------------------
```

me21, 2022-06-29:
Here's my build log:
[build.log](https://github.com/RobertCNelson/omap-image-builder/files/9010865/build.log)

RobertCNelson, 2022-06-29:
very very weird...

```
-----------------------------
Backup version of u-boot (MLO): /opt/backup/uboot/
'./u-boot/bb-u-boot-am335x-evm/MLO' -> '/tmp/tmp.tysVP6PFzx/disk/opt/backup/uboot/MLO'
Backup version of u-boot (u-boot.img): /opt/backup/uboot/
'./u-boot/bb-u-boot-am335x-evm/u-boot.img' -> '/tmp/tmp.tysVP6PFzx/disk/opt/backup/uboot/u-boot.img'
making ping/ping6 setuid root
```
vs:
```
-----------------------------
Backup version of u-boot (MLO): /opt/backup/uboot/
cp: не удалось выполнить stat для './u-boot/bb-u-boot-am335x-evm/MLO': Нет такого файла или каталога
```

looks like it didn't pass the path...

me21, 2022-06-29:
```
azarubkin@ubuntu-2004-vm:~/beaglebone/image-builder$ git branch -v
* master 74c3f405 add EW 2022 demo for BeagleBone AI-64
```