RobertCNelson/omap-image-builder #203: Will not run from Debian

jeanparpaillon, 2022-07-25:
When running from Debian `RootStock-NG.sh` fails installing dependancies.

In `scripts/install_dependencies.sh`, it looks for `qemu` package, which is a Ubuntu metapackage and does not exist on Debian.

Depending on `qemu-system-arm` solves the issues on Debian. The package also exists on Ubuntu.

RobertCNelson, 2022-07-25:
They must have dropped this for Bookworm/Sid.. ;)

https://packages.debian.org/bullseye/qemu

Thanks

closed 2022-07-25 by RobertCNelson

jeanparpaillon, 2022-07-26:
@RobertCNelson actually, they have made it a virtual package (provided by 'qemu-system-*') so it won't be listed with `dpkg -l` neither can be installed through `apt install qemu`

jeanparpaillon, 2022-07-26:
As far as I know, there is no easy way to check if a virtual package is actually 'installed' :/
https://askubuntu.com/questions/679137/how-to-check-if-a-virtual-package-is-installed

jeanparpaillon, 2022-07-26:
My bad, this is not even a virtual package, but a dummy one. I have no idea how to deal with that :)

RobertCNelson, 2022-07-26:
The binary's we really want is..

```
/usr/bin/qemu-arm-static
/usr/bin/qemu-aarch64-static
/usr/bin/qemu-riscv64-static
```

qemu-user-static, should be safe for all 'lts' versions of ubuntu and debian now right?

https://packages.debian.org/unstable/qemu-user-static

Regards,

jeanparpaillon, 2022-07-26:
tbh, I don't really know the difference between qemu-user-static and other qemu-*-static
From the package name point of view, however, qemu-user-static seems a good fit

PR #204 (2022-07-25): Depend on qemu-system-arm instead of qemu
qemu-system-arm is availale on both Ubuntu and Debian, while qemu does
not exist on Debian.

FIxes #203