RobertCNelson/omap-image-builder #210: Trouble running the image-builder

orgua, 2022-10-01:
Is there more info about running the scripts to build your own image for the BBB?

As the rcn-ee apt repo seems to support ubuntu 22.04 / jammy, I wanted to create a test-image but already failed at the default ubuntu image instructions given in the readme: 

```Shell
git clone https://github.com/RobertCNelson/omap-image-builder
cd ./omap-image-builder
./RootStock-NG.sh -c rcn-ee.net-console-ubuntu-focal-v5.10-ti-armhf
```

As debian-based linux as a host seems tested, I tried to run it on ubuntu 20.04 and and 22.04 without success: 

```
Traceback (most recent call last):
  File "/usr/lib/cnf-update-db", line 3, in <module>
    import apt_pkg
ModuleNotFoundError: No module named 'apt_pkg'
Reading package lists... Done
E: Problem executing scripts APT::Update::Post-Invoke-Success 'if /usr/bin/test -w /var/lib/command-not-found/ -a -e /usr/lib/cnf-update-db; then /usr/lib/cnf-update-db > /dev/null; fi'
E: Sub-process returned an error code
```

I found one undocumented prereq I had to install: 
- m4 (via apt) 

is there more? I looked through all directories, saw the jenkins-files and the ./docs, but could not find helpful information.

RobertCNelson, 2022-10-01:
Hi @orgua sorry, at one time we did have a dependency check, but debootstrap is actually common on non-debian systems, so it was decided to not try to do every combination..

now days i'm having too much fun on other arch's, i don't think i'll actually finish the 22.04/jammy bits..

RobertCNelson, 2023-03-10:
are you running in WSL (1) or WSL (2) mode, WSL 2 provides a better linux internal vm..

ps, sorry i haven't tried this script personally on wsl..  running the debootstrap step should be the easiest for wsl to do, afterwards i'd fully expect to barf on the bind mount and chroot into the  qemu-<arch> enviromrment for final install.

orgua, 2023-03-10:
Yes it was indeed WSL1, but after updating to WSL 2 it still fails :) 
But using a VM did the trick - the script ran almost through ('tree' was not installed). Thanks so much for your work!

One more question if I may:

Now that there is an riscv64-jammy on the image-server and the [rcn-ee apt repo](http://repos.rcn-ee.com/ubuntu/) also has jammy-support -> Would it be feasible to compile a working BB Black image by just adapting the focal-config?

RobertCNelson, 2023-03-10:
yeah, i think that would work,

just 
copy the focal one directly rcn-ee.net-console-ubuntu-focal-v5.10-ti-armhf, and then tweak the new stuff defined in rcn-ee.net-console-ubuntu-2304-riscv64 

the biggest issue, these might now all be built yet for jammy..

https://github.com/RobertCNelson/omap-image-builder/blob/master/configs/rcn-ee.net-console-ubuntu-2304-riscv64.conf#L131-L143

orgua, 2023-03-10:
awesome spidersenses! Thats exactly where its failing. some of these packages you pointed out are in the repo, but most of them aren't. 

```
Hit:1 http://ports.ubuntu.com jammy InRelease
Hit:2 http://repos.rcn-ee.com/ubuntu jammy InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package bb-beagle-config
E: Unable to locate package bb-beagle-version
E: Unable to locate package bb-cape-overlays
E: Unable to locate package bb-usb-gadgets
E: Unable to locate package generic-sys-mods
E: Unable to locate package gt
E: Unable to locate package libusbgx
Log: umount: [/home/service/omap-image-builder/ignore/tmp.tGPfyN1lqQ/dev/pts]
```

Some fallback to focal would suffice. I did an ubuntu-upgrade once - it took forever on the BB - but the new software-versions brought a significant and much needed speed improvement for us.

BTW: is [that line](https://github.com/RobertCNelson/omap-image-builder/blob/master/configs/rcn-ee.net-console-ubuntu-2304-riscv64.conf#L130) a possible bug? I tried adding it to jammy, but there are only debian-source in it.

RobertCNelson, 2023-03-11:
Oh that line is only for arm64/riscv64 debian.beagleboard.org is not a full mirror of rcn-ee.net (the armhf has a large history, no reason to mirror that when we enabled the BeagleV project and then the TI bbai64 project)...

Regards,

closed 2023-12-03 by orgua