RobertCNelson/omap-image-builder #212: binary-armhf/Packages no longer exist at http://us.archive.ubuntu.com/ubuntu/dists/xenial/main.

davenmccalla-diagraph, 2023-10-09:
I followed a "Getting Started with the Beaglebone Black" tutorial back in 2016 that had a "bare" file system option. Now I want to upgrade to bullseye and the updated tutorial no longer has the "bare" file system option. After poking around inside the tar file for the "bare" file system, and reading the image-builder project file in there, it's obvious that the scripts in this repository were used to create that "bare" file system. So I'd like to try and edit the "bare" image-builder project file from jessie and use it to create a "bare" bullseye file system. Will this work?

I cloned this repository and tried building the bullseye "minfs" file system first:
./RootStock-NG.sh -c eewiki_minfs_debian_bullseye_armhf

It couldn't clone the linux-firmare repository, and I had to change the "git_clone_address" to:
git_clone_address="git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git"

After that I'm getting the following error:
E: Failed to fetch http://us.archive.ubuntu.com/ubuntu/dists/xenial/main/binary-armhf/Packages  404  Not Found

Can you help me?

RobertCNelson, 2023-10-09:
```E: Failed to fetch http://us.archive.ubuntu.com/ubuntu/dists/xenial/main/binary-armhf/Packages 404 Not Found```

why are you loading xenial when building "bullseye"?

anywho, ubuntu does not store 'armhf' on us.archive.ubuntu.com  non-x86 as always been on their ports server..`ports.ubuntu.com`

davenmccalla-diagraph, 2023-10-09:
I wasn't trying to load xenial, but something in RootStock-NG.sh (or supporting scripts) is trying to load xenial. I thought you may know why this is happening. I'll keep investigating to see if I can figure it out.

The reason I'm doing this is because I need an image smaller than 'minfs' but more than 'bare'. I've been using the 'bare' jessie image and installing additional packages. Once I'm able to get one of your standard .conf files working, I'm going to try to create a custom .conf that will build a small file system that has everything I need.

davenmccalla-diagraph, 2023-10-09:
I think I see what happening. The install_dependencies.sh is trying to install debootstrap on my Linux VM when it gets the load xenial error.

davenmccalla-diagraph, 2023-10-09:
I tried on a different Linux VM and it built eewiki_minfs_debian_bullseye_armhf image without any issues. When I convert the 'bare' jessie .conf into a 'bare' bullseye .conf to build a 'bare' bullseye image, I get the following error:

E: Unable to locate package rcn-ee-archive-keyring

Is this something you can help with? Sorry, I know you don't support the 'bare' file system anymore. It will help ALOT if I can get this to work.

RobertCNelson, 2023-10-10:
That package was later  re-named as...  `bbb.io-keyring`..

So the Bare image was a always a little more complex.

As my original Goal with it was Debian in a 256MB NAND package on a sama5 mpu..

At the time, NAND was pretty expensive and Docker image sizes wasn't a big deal.

Now days, it's cheaper to just throw a bigger NAND.  As most of complexity of how Bare Image was built, stemmed from  a lot of testing, verification, repeat, etc..

For example, in this commit, you can see where i had verified 6 different broken BusyBox application: https://github.com/RobertCNelson/omap-image-builder/commit/bfa0218e7391f1c3b5937b69fdccc3a6e536c4a6

Side note, i like what that author did, it makes it much easier to choose what BusyBox works as..

So it was nice at the time, but memory is cheap, and debian has done a better job making minimal images.

If you look at Docker, the debian-slim is pretty impresive..

This team has done a lot of the work now: https://github.com/debuerreotype

Regards,

davenmccalla-diagraph, 2023-10-11:
Changing rcn-ee-archive-keyring to bbb.io-keyring in the .conf file worked. I also had to remove a couple items from chroot_manual_deborphan_list and add rfs_root_password.

Thanks for your help!

closed 2023-10-11 by davenmccalla-diagraph