RobertCNelson/omap-image-builder #215: Ubuntu 24.04 missing packages

tmarkettos, 2024-07-03:
I tried to build the rcn-ee.net-console-ubuntu-2404-v5.10-ti-armhf-am335x image via `ci/daily-ubuntu-armhf-2404-minimal-v5.10-ti-am335x.sh`.  The build proceeds but then fails when trying to add some packages from the rcn-ee.com repo:

```
Log: (chroot) Installing (repo_rcnee_pkg_list): bbb.io-kernel-5.10-ti bbb.io-kernel-5.10-ti-am335x bbb.io-keyring bb-beagle-flasher bb-customizations bb-usb-gadgets bb-wlan0-defaults generic-sys-mods libusbgx2 overlayroot systemd-timesyncd
Hit:1 http://ports.ubuntu.com/ noble InRelease
Hit:2 http://ports.ubuntu.com/ noble-security InRelease
Hit:3 http://ports.ubuntu.com/ noble-updates InRelease
Hit:4 https://repos.rcn-ee.com/ubuntu noble InRelease
Reading package lists...
Log: (chroot) Installing (repo_rcnee_pkg_list): bbb.io-kernel-5.10-ti bbb.io-kernel-5.10-ti-am335x bbb.io-keyring bb-bbai-firmware bb-beagle-flasher bb-beagle-version bb-customizations bb-usb-gadgets bb-u-boot-am335x-evm bb-wl18xx-firmware bb-wlan0-defaults generic-sys-mods gt libusbgx overlayroot systemd-timesyncd sancloud-firmware
Hit:1 http://ports.ubuntu.com/ noble InRelease
Hit:2 http://ports.ubuntu.com/ noble-security InRelease
Hit:3 http://ports.ubuntu.com/ noble-updates InRelease
Hit:4 https://repos.rcn-ee.com/ubuntu noble InRelease
Reading package lists...
Log: (chroot): [apt-get install -yq bbb.io-kernel-5.10-ti bbb.io-kernel-5.10-ti-am335x bbb.io-keyring bb-bbai-firmware bb-beagle-flasher bb-beagle-version bb-customizations bb-usb-gadgets bb-u-boot-am335x-evm bb-wl18xx-firmware bb-wlan0-defaults generic-sys-mods gt libusbgx overlayroot systemd-timesyncd sancloud-firmware]
Reading package lists...
Building dependency tree...
Reading state information...
Package gt is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libusbgx is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package bb-bbai-firmware
E: Unable to locate package bb-beagle-version
E: Unable to locate package bb-wl18xx-firmware
E: Package 'gt' has no installation candidate
E: Package 'libusbgx' has no installation candidate
E: Unable to locate package sancloud-firmware
```

I tried to adapt by removing packages not in the rcn-ee 'noble' repo, but eventually ran into problems:
```
Log: (chroot): [apt-get install -yq bbb.io-kernel-5.10-ti bbb.io-kernel-5.10-ti-am335x bbb.io-keyring bb-beagle-flasher bb-customizations bb-usb-gadgets bb-wlan0-defaults generic-sys-mods libusbgx2 overlayroot systemd-timesyncd]
Reading package lists...
Building dependency tree...
Reading state information...
overlayroot is already the newest version (0.48).
systemd-timesyncd is already the newest version (255.4-1ubuntu8.1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 bb-usb-gadgets : PreDepends: gt but it is not installable
                  PreDepends: libusbgx but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Is something out of step here, or should there be more packages in the repo?

RobertCNelson, 2024-07-03:
No @tmarkettos this is me, i really need to finish porting arm32/arm64 package set to noble (from jammy).. Been to focused on getting the BeagleY-AI launched. But that's out the door now too..

(PS, i'm tacking the same build failure on my daily ci.. 
[image]
)

tmarkettos, 2024-07-03:
Ah, thanks.  Not to worry, for now I'll carry on using 22.04 until 24.04 is stable - have no hard requirement for 24.04 at the moment.

RobertCNelson, 2024-07-03:
@tmarkettos and it's up.. https://rcn-ee.net/rootfs/ubuntu-armhf-24.04-console-v5.10-ti/

i built the missing packages and relaxed the need for qt/libusbgx

tmarkettos, 2024-07-17:
Great, thanks that's working nicely now. Ubuntu 24.04 deployed and in production :-)

closed 2024-07-17 by tmarkettos