RobertCNelson/omap-image-builder #216: Hardcoded path for USB mass storage gadget

tmarkettos, 2024-07-17:
/usr/bin/bb-start-acm-ncm-rndis-old-gadget has a hardcoded path for the USB storage device:

```
                if [ ! "x${USB_IMAGE_FILE_DISABLED}" = "xyes" ]; then
                        if [ -f /etc/bbb.io/templates/beagle-getting-started.img ] ; then
                                echo "${log} enable USB mass_storage beagle-getting-started.img"
                                mkdir -p functions/mass_storage.usb0
                                echo ${usb_ms_stall} > functions/mass_storage.usb0/stall
                                echo ${usb_ms_cdrom} > functions/mass_storage.usb0/lun.0/cdrom
                                echo ${usb_ms_nofua} > functions/mass_storage.usb0/lun.0/nofua
                                echo ${usb_ms_removable} > functions/mass_storage.usb0/lun.0/removable
                                echo ${usb_ms_ro} > functions/mass_storage.usb0/lun.0/ro
                                echo /etc/bbb.io/templates/beagle-getting-started.img > functions/mass_storage.usb0/lun.0/file

                                ln -s functions/mass_storage.usb0 configs/c.1/
                        fi
                fi
```

I think it goes without saying that /etc/bbb.io/templates/beagle-getting-started.img is not a location everyone wants to put their disc images.  I know this is only set up for a demo, but we use this functionality in production and it works fine, apart from the very odd path.

I modified the script to use a different path, but that was wiped out when the `bb-usb-gadgets` package was updated.  My current workaround is:

```
sudo ln -s /var/lib/cfumass.img /etc/bbb.io/templates/beagle-getting-started.img
```

which works but is ugly.  Since the bb-start-acm-ncm-rndis-old-gadget script sources `/etc/default/usb-oem` I could put a setting to override a variable like `usb_ms_imagefile=/var/lib/cfumass.img` in there if the script used it.  Any chance the path could go via a variable so it can be overriden?

(I would make a PR, but I don't know how the bb-usb-gadgets package is generated or where its repo is)

RobertCNelson, 2024-07-17:
Hi @tmarkettos my ci hosting provider is down, so i can't push any changes to our apt repo at the moment, how does `USB_IMAGE_FILE` override sound? we can shove it in `/etc/default/usb-oem` so it'll easily override the default.. 

https://github.com/rcn-ee/repos/blob/master/bb-usb-gadgets/suite/bookworm/debian/bb-start-acm-ncm-rndis-old-gadget#L159

```
USB_IMAGE_FILE=/etc/bbb.io/templates/beagle-getting-started.img
```

Regards,

closed 2024-07-17 by RobertCNelson

RobertCNelson, 2024-07-17:
This is now pushed out to all active distros (armhf, arm64, riscv64)..