RobertCNelson/omap-image-builder #220: Question about .conf file.

davenmccalla-diagraph, 2025-03-27:
I'm migrating from Debian 8.5 (Jessie) to Debian 11.8 (Bullseye) and I'm trying to maintain the smallest filesystem as possible. I was using debian-8.5-bare-armhf-2016-08-16.tar.xz that was available at that time. The "bare" filesystem for Debian 11.8 is not available, so I'm using the omap-image-builder to make one by modifying the image-builder.project file I found inside debian-8.5-bare-armhf-2016-08-16.tar.xz.

Here's the image-builder.project from the Debian 8.5 "bare" filesystem:
```
tempdir="/var/lib/jenkins/workspace/eewiki_image_snapshots/ignore/tmp.KCbOr9tBNF"
time="2016-08-16"
export_filename="debian-8.5-bare-armhf-2016-08-16"
#
##
release="8.5"
image_type="bare"
##
##Debootstrap: https://wiki.debian.org/Debootstrap
##
##debootstrap --arch=${deb_arch} --include=${deb_include} --exclude=${deb_exclude} \
##--components=${deb_components} --foreign ${deb_codename} /tmp/tmp.dir/ http://${deb_mirror}
##
deb_distribution="debian"
deb_codename="jessie"
deb_arch="armhf"
deb_include="busybox"
deb_exclude="	\
	aptitude	\
	aptitude-common	\
	dmidecode	\
	groff-base	\
	info	\
	install-info	\
	man-db	\
	manpages	\
	tasksel	\
	tasksel-data	\
	vim-common	\
	vim-tiny	\
	wget	\
	whiptail	\
"

deb_components="main contrib non-free"
deb_mirror=""
##
##Some packages fail to install via debootstrap: deb_additional_pkgs="<comma|space>"
##
deb_additional_pkgs=""
##
rfs_username="debian"
rfs_fullname="Demo User"
rfs_password="temppwd"
rfs_hostname="arm"
rfs_startup_scripts=""
rfs_opt_scripts=""
rfs_console_banner=""
rfs_console_user_pass="enable"
rfs_ssh_banner=""
rfs_ssh_user_pass=""
##
##enable kernel repo: http://repos.rcn-ee.com/(debian|ubuntu)
repo_rcnee="enable"
repo_rcnee_pkg_list="rcn-ee-archive-keyring"
##
chroot_not_reliable_deborphan=""
chroot_manual_deborphan_list="	\
	libboost-iostreams1.55.0	\
	libgdbm3	\
	libnewt0.52	\
	libpipeline1	\
	libsigc++-2.0-0c2a	\
	libssl1.0.0	\
"

chroot_very_small_image="enable"
chroot_tarball="enable"

```
The "deb_include" section is straight forward, but it's not clear what the "deb_exclude" section does. Does it remove it from a "base" list not defined in this file? Also, it appears that the "chroot_manual_deborphan_list" removes packages, but how did these packages get installed when they weren't part of the "deb_include"? Are these packages part of a "base" install not defined in this file?

RobertCNelson, 2025-03-27:
deb_exclude list will get passed to debootstrap:

```
       --exclude=alpha,beta
              Comma separated list of packages which will be removed from download and extract lists.  WARNING: you can and probably will exclude essential packages, be careful using this option.
```

at the time i generated these bare images, i was targeting a device with 128MB of nand... since then most devices have GB's of space.. for really small OS's buildroot/yocto is better..

Sorry haven't really played around with these bare really small debian images anymore..

Regards,

davenmccalla-diagraph, 2025-04-06:
I added deb_variant="minbase" to my .conf file:
```
##
release="11.8"
image_type="bare"
##
##Debootstrap: https://wiki.debian.org/Debootstrap
##
##debootstrap --arch=${deb_arch} --include=${deb_include} --exclude=${deb_exclude} \
##--components=${deb_components} --foreign ${deb_codename} /tmp/tmp.dir/ http://${deb_mirror}
##
deb_distribution="debian"
deb_codename="bullseye"
deb_arch="armhf"
deb_include="	\
	busybox	\
	hostapd	\
	dnsmasq	\
	wpasupplicant	\
	cifs-utils	\
	samba-common	\
	dbus	\
	libftdi-dev	\
	rsync	\
	inotify-tools	\
	telnetd	\
	uftp	\
	libusb-1.0-0	\
	python2.7	\
	usbutils	\
	gpiod	\
	firmware-realtek	\
	systemd-timesyncd	\
	sudo	\
	passwd	\
	ca-certificates	\
"
deb_exclude="	\
	aptitude	\
	aptitude-common	\
	dmidecode	\
	groff-base	\
	info	\
	install-info	\
	man-db	\
	manpages	\
	tasksel	\
	tasksel-data	\
	vim-common	\
	vim-tiny	\
	wget	\
	whiptail	\
"

deb_components="main contrib non-free"
deb_variant="minbase"
deb_mirror=""
##
##Some packages fail to install via debootstrap: deb_additional_pkgs="<comma|space>"
##
deb_additional_pkgs=""
##
rfs_username="debian"
rfs_fullname="Demo User"
rfs_password="temppwd"
rfs_hostname="arm"
rfs_startup_scripts=""
rfs_opt_scripts=""
rfs_console_banner=""
rfs_console_user_pass="enable"
rfs_ssh_banner=""
rfs_ssh_user_pass=""
##
##enable kernel repo: http://repos.rcn-ee.com/(debian|ubuntu)
repo_rcnee="enable"
repo_rcnee_arch="armhf"
repo_rcnee_pkg_list="	\
	generic-sys-mods	\
"
##
chroot_not_reliable_deborphan=""
chroot_manual_deborphan_list="	\
	libnewt0.52	\
	libpipeline1	\
	libsigc++-2.0-0c2a	\
	ca-certificates	\
"

chroot_very_small_image="enable"
chroot_tarball="enable"
```

It shaved about 30MB off the size of my filesystem, which is helpful. However, now it doesn't run my startup script, so I'm sure I need to add more packages. The last thing I see in the boot process is:
```
[    5.340340] mmc0: new high speed SDHC card at address aaaa
[    5.560210] mmcblk0: mmc0:aaaa SL16G 14.8 GiB
[    5.788675]  mmcblk0: p1 p2 p3
[    5.995935] of_cfs_init
[    5.996044] of_cfs_init: OK
[    6.032612] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    6.032757] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[    6.045349] devtmpfs: mounted
[    6.047372] Freeing unused kernel memory: 1024K
[    6.051606] Run /sbin/init as init process
[    6.053438] Run /etc/init as init process
[    6.055013] Run /bin/init as init process
[    6.056414] Run /bin/sh as init process

```

I'm just now noticing there is a "rfs_startup_scripts=" in the config file. I've been placing a .service file in the \etc\systemd\system\multi-user.target.wants folder to run my startup script.

It's probably a small research project for me to figure what packages I need to add back so it runs my startup script. If you happen to know what I'm missing, please share.